zwave-js/zwave-js #3296: S2 inclusion times out and fails

tsoslow, 2021-09-07:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

YES, BUT a ZWaveJS2MQTT developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I have two Aeotec Range Extender 7 devices. They are the same version and have the same firmware. One was included with S2 security with no problem. The other one seems to timeout after entering the DSK and then it is added with security "None".

Steps to reproduce the behavior:
  1. Put z-wave js into inclusion mode with defaults
  2. Click button on range extender to start inclusion
  3. Click next for default security settings (S2 Authenticated & S2 Unauthenticated)
  4. Enter DSK key

Expect to inclusion to complete successfully with security. Instead it pauses for about 2 minutes and then says that it was added with security "None".

### Device information

Manufacturer: Aeotec Ltd.
Model name: ZW189-C15
Node ID in your network: 16

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Yes. It worked with another Range Extender with identical version information.

### Attach Driver Logfile

[zwavejs_2021-09-07.log](https://github.com/zwave-js/node-zwave-js/files/7124575/zwavejs_2021-09-07.log)

AlCalzone, 2021-09-08:
Part of the S2 bootstrapping process requires the nodes to send commands to which Z-Wave JS must react. Most of the time when S2 bootstrapping fails, these commands are never received - like in this case, which is exactly as I described there: https://github.com/zwave-js/node-zwave-js/issues/3239#issuecomment-906963206

Duplicate of #3239

AlCalzone, 2021-09-08:
Duplicate of #3239

closed 2021-09-08 by AlCalzone