zwave-js/zwave-js #3297: Allegion BE469 lock repeatedly does not answer commands (Failed to send the command after 1 attempts. Transmission queue full)

saucepwn, 2021-09-07:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Attempts to send commands to a specific BE469 lock fail with ""

What do you observe?
ZWave JS marks the node as "dead"

What did you expect to happen?
The lock responds to the command.

The attached log output is the result of me sending a "Lock" command to the lock. This lock has full batteries and has been working well for ~2 years with this controller. There have been no recent changes to my Z-Wave network, other than me keeping up with Z-Wave JS releases.

### Device information

Manufacturer: Allegion
Model name: BE469
Node ID in your network: 008

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: Driver Version: 8.2.1 / Server Version: 1.10.3
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

This used to work ~99% of the the until late August (I do not know exactly when the issue started as I was away for several weeks).

Now, issuing commands to this lock only succeeds about ~20% of the time.

Troubleshooting steps I've taken:
* Changed the batteries in the lock
* Factory reset the lock
* Un-enrolled and re-enrolled the lock onto the network (it took several tries to successfully un-enroll, and several more to successfully re-enroll)

### Attach Driver Logfile

[zwave_failed_lock_command.log](https://github.com/zwave-js/node-zwave-js/files/7124762/zwave_failed_lock_command.log)

Ikcelaks, 2021-09-08:
I have the same door lock and am running the same versions as you without any issues.

The logs show communication issues.
Is it possible to move the controller close to the lock for testing?
Have any powered nodes been moved or gone dead?
Are there any new electronics that might be causing interference?

saucepwn, 2021-09-08:
My controller is rack mounted, and the lock is affixed to the door, so it wouldn't be easy to bring the two closer together. The problem lock is approx ~35 feet from the controller, with only a cabinet between them. I have an identical lock that is ~5 feet from the controller which is still working. I will place an order for a USB extension cable so I can get the Z-Wave stick closer to the lock, will update in several days when the cable arrives.

All of my powered nodes are alive and can communicate with the controller. None have been moved recently.

I am starting to suspect the lock itself (2 years old) may be failing. But I will try the USB extension cable and report back.

AlCalzone, 2021-09-08:
> My controller is rack mounted

That's probably your problem. The short log you posted shows both a situation where the controller can't send anything (might be due to interference) and one where the node didn't acknowledge that it received a command (or it didn't receive it).

Z-Wave sticks are very sensitive to intereference and USB ports in server racks are close to the worst thing you can do for them. In these cases, a USB extension cord usually helps.

saucepwn, 2021-09-12:
Thanks, I added a 16ft USB extension cable ([this product](https://www.amazon.com/gp/product/B009GUTFX8)) and it looks to be working reliably again.

closed 2021-09-12 by saucepwn