zwave-js/zwave-js #3301: Missing "Manufacturer Specific" for Eurotronics Spirit radiator valves

lenwar, 2021-09-08:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
The Radiator Valve has a Thermostate Mode called 'Manufacturer Specific' by the vendor, which effectively allows manual control of the valve. (so, you can set it to a certain percentage open, 0 being completely closed, 100 being completely open).

Using OpenZwave the valves can be set to 'Manufacturer Specific' setting via ozw-admin or mqtt commands. Zwave-js doesn't seem to support the setting.

The setting visible in Zwave-JS is: [12-64-0-mode] Thermostat mode which has:
0. Off
1. Heat
11. Energy heat
15. Full power

If is missing the option:
31. Custom

(When setting the valve to Manufacturer Specific, this is what the value is set to)

The manual describes it at chapter 6.8 as Mode '0x1F'
https://products.z-wavealliance.org/ProductManual/File?folder=&filename=MarketCertificationFiles/2543/Spirit%20Z-Wave%20Plus_DRAFT_V4.pdf

I think it is an error in the firmware of the device (it doesn't seem to provide the setting/mode during an interview (by lack of a proper description from my side )).

I tried contacting Eurotronics about it, so maybe they could provide a fixed firmware, but they don't respond to anything.

Given that other Zwave-implementations have workarounds for this, I was hoping it could also be 'fixed' in Zwave-JS. (At least OpenZwave, Domoticz and also the Athom Homey appliance can work with it)

### Device information

Manufacturer: Eurotronics (Manufacturer ID: 0x0148)
Model name:  Spirit
Node ID in your network: 

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
I'm using the current 'Zwave JA to MQTT' addon in Home Assistant: which uses ZwaveJS2Mqtt v5.5.1

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/7128192/zwave_js.log)

AlCalzone, 2021-09-08:
I could swear there is an issue about this somewhere. The device's response to which modes it supports is incomplete, which you correctly described.
```
2021-09-08T10:37:45.291Z CNTRLR « [Node 026] received supported thermostat modes:
                                  · Off
                                  · Heat
                                  · Energy heat
                                  · Full power
```

We'd need to implement https://github.com/zwave-js/node-zwave-js/issues/1625 for this, which is a bit tricky. I don't like single-use workarounds for single devices, so this needs to be implemented in a way that we can use the workaround for all those devices that report their capabilities incorrectly.

AlCalzone, 2021-09-08:
You should be able to work around this on your end by using the Basic CC `targetValue` to control the thermostat mode:
[image]

It might be hidden, but if you change the value programmatically, the driver will send a command.

lenwar, 2021-09-08:
> 
> 
> You should be able to work around this on your end by using the Basic CC `targetValue` to control the thermostat mode:
> [image]
> 
> It might be hidden, but if you change the value programmatically, the driver will send a command.

And how would I approach this?
I tried using an MQTT-command, which can set it to the other four settings, but not the 'manufacturer specific' one. I couldn't figure it out trying it another way.

Is there some example somewhere, which shows how I can send a Basic CC in that way?

AlCalzone, 2021-09-08:
@robertsLando can you help?

robertsLando, 2021-09-09:
> How can you set Basic CC targetValue via mqtt?

If the value reports a Basic CC `targetValue` (you can see the value in control panel) you can use the usual set value: https://zwave-js.github.io/zwavejs2mqtt/#/guide/mqtt?id=set-values

Otherwise you can use the `sendCOmmand` api: https://zwave-js.github.io/zwavejs2mqtt/#/guide/mqtt?id=send-command

The topic depends on your settings, the payload I ask to you @AlCalzone to edit the one I report here: 

Topic: zwavejs/_CLIENTS/ZWAVE_GATEWAY-<yourName>/api/sendCommand/set

Payload:

{ "args": [
  {
    "nodeId": <nodeId>,
    "commandClass": 32,
    "endpoint": 0,
  },
  "set",
  [<your value>] // this are the args
  ]
}

AlCalzone, 2021-09-09:
This should also work with a driver function (not sure if you can do that via mqtt though):
```
await driver.controller.nodes.get( yourNodeId ).commandClasses.Basic.set( value )
```
@robertsLando correct me if I forgot something

robertsLando, 2021-09-10:
Yeah you could firstly tyy to send that function from Control Panel, if it works you can then call it with that payload using MQTT

lenwar, 2021-09-11:
Hi, sorry for the late reply.
I'm unable to send it via the Control Panel, as the options isn't available. (which is the base of the issue :) ). It only has a drop-down with the other functions. Option 31 (which is the 'manufacturer specific' isn't available)

I did, however, manage to send it via an MQTT-command now. In the past that also didn't work, so at least I have a working workaround for this :)

AlCalzone, 2021-09-12:
> I'm unable to send it via the Control Panel, as the options isn't available. (which is the base of the issue :) ). It only has a drop-down with the other functions. Option 31 (which is the 'manufacturer specific' isn't available)

I think you misunderstood what @robertsLando meant. zwavejs2mqtt lets you execute custom scripts like the one I posted above, either via the Control Panel or MQTT (I think).

Also, to be sure: There is no `Basic CC` `targetValue` for this device exposed in the control panel, correct?

lenwar, 2021-09-12:
I wasn't aware of the Control Panel being able to do that. Thanks for that. With that I finally managed to provide the following Service Command in HA:

```
service: zwave_js.set_value
data:
  value: 31
  command_class: '64'
  property: mode
target:
  entity_id: climate.radiatorkraan_gang
```

This is an even better workaround for me! Thanks loads for your time!!

closed 2021-09-13 by robertsLando

lenwar, 2021-09-13:
For reference to anyone needing this. I wrote a small how-to on the HA community forum describing how I use the above in a solution:
https://community.home-assistant.io/t/eurotronic-spirit-radiator-valves-with-external-temperature-sensors-in-zwavejs/288720

jip67, 2021-12-29:
Thanks a lot. The service call works for me, but after a minute or so, the setting is changed automaticly bij 'supervisor'. 
[image]
Does anyone know why this happens and more importantly how to disable this?
Greetz

lenwar, 2021-12-29:
If 'supervisor' performs actions it's usually something 'timed' or maybe another addon like node-red.

(Maybe you have an automaton running 'every minute' that touches your radiator valve?)

jip67, 2021-12-29:
I feel stupid!!!!

It was a node red flow.

Thanks!