zwave-js/zwave-js #3304: Cannot add secure lock after upgrade to Z-Wave JS

the-gray-ghost, 2021-09-08:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
After upgrade, I lost my network key and had to exclude and re-include my smart locks.  However, they are unable to be included securely.  I have excluded it, did a factory reset, restarted HA and the Z-Wave server, then stated the secure inclusion process 6" from the controller.

What do you observe?
Lock is added to HA, but not securely and I am unable to use lock function.  All non-secure devices work properly.

### Device information

Manufacturer: Kwikset
Model name: 910
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 0.1.38
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Upgraded to Z-Wave JS

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Worked properly in deprecated Z-Wave integration

### Attach Driver Logfile

[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/7130630/zwave_js.log)

AlCalzone, 2021-09-08:
Which driver version is that? I can't see it in your log.
It also doesn't look like secure inclusion was chosen.

> YES, BUT a Home Assistant developer has told me to come here

Do you have some context for this? An issue in the HA repo or a link to Discord?

the-gray-ghost, 2021-09-09:
Not sure where I find the driver version.  I am running server version 0.1.38.

I am toggling the "use secure inclusion" switch, the button changes to "Start Secure Inclusion".

I was in the HA community forum, was referred over here.  Not sure if it was a developer, didn't check his credentials.  Is there a better place for this to go?

kpine, 2021-09-09:
The driver version is 8.2.1.

@the-gray-ghost Are you running HA version 2021.9.4 or later? There was an inclusion bug in 2021.9.0 - 3.

Do you have a link to the community forum for context? Usually, problems in HA are reported to its Github project first for triage.

the-gray-ghost, 2021-09-09:
I was running 2021.9.3, upgraded to 9.5 last night and retried, didn't work.  However, I tested the lock in place (about 20 ft from controller), I will retest tonight next to the controller.  One notable change, the device never fully added to the integration like it has before (even from this distance).  The log showed it was constantly being queried and timing out.  I didn't capture the log as I figured it might have been related to the proximity.

Here is the forum thread:
https://community.home-assistant.io/t/kwikset-910-no-longer-working-after-z-wave-js-upgrade/336249

I thought this was the proper project page, my apologies if it is not.

AlCalzone, 2021-09-09:
In your log in the forum thread I see this:
```
2021-09-06T20:34:24.280Z CNTRLR » [Node 024] Querying securely supported commands (S0)…
2021-09-06T20:34:34.381Z CNTRLR [Node 024] Timed out while waiting for a response from the node
```
The node doesn't respond when asked for its secure commands, so the driver continues to talk to it without encryption, to which the node doesn't respond either. My guess: wrong key or some other kind of usage/application error.

> I thought this was the proper project page, my apologies if it is not.

And that's why the first question the issue template asks if a developer sent you here :)

the-gray-ghost, 2021-09-09:
I will be sure to check the poster's credentials next time!  lol.

I actually have two of these locks, they both have the same problem.  Even after a factory reset, neither one will connect, so I think it is safe to assume it's not a device hardware issue.  I have checked my integration and I see a proper network key.  I just can't figure out what is wrong!

I guess the only thing I have left to do is a full uninstall of the Z-Wave JS server and integration and hope that a reinstall will fix the issue.  Will I have to start the network from scratch?  If not, should I?  It would be a royal pain to have to redo everything, but I'm tired of going in circles with these locks.

I do greatly appreciate the help and support you have given me here!

AlCalzone, 2021-09-09:
> Will I have to start the network from scratch? If not, should I? It would be a royal pain to have to redo everything, but I'm tired of going in circles with these locks.

No need.

FYI, I double checked with latest zwavejs2mqtt:
[image]

I don't have a lock though, so I chose the S0 inclusion strategy to force S0.

Can you try zwavejs2mqtt - that should let us figure out if it is a problem in the HA integration.

the-gray-ghost, 2021-09-09:
I can certainly give it a shot, however, I have not used the MQTT route before, so it will be a learning experience.  Do I need to remove my current setup or will it run parallel with it?

AlCalzone, 2021-09-09:
AFAIK it is as easy as: shut down the official integration, start zwavejs2mqtt and open its UI. They can be installed in parallel, they just can't run at the same time.

the-gray-ghost, 2021-09-12:
Updated the server and core to the latest builds and both locks added to network without issue.  I appreciate the help guys!

closed 2021-09-12 by AlCalzone