zwave-js/zwave-js #3315: ZSO7300 Device ID is wrong

briis, 2021-09-10:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
The device driver in https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0234/zso7300.json has the wrong productType and productId.

What do you observe?
When adding the device to Z-Wave it is not found and added as a generic switch.

What did you expect to happen?
I expected it to be reported as a Logic Group ZSO7300 Switch

The device is just acquired, and I only have 1, so I don't know if the current ProductType and ProductID worker for older version. I have added the correct Type and ID below. I did manually edit the file and with these ID's all is working as expected

### Device information

Manufacturer:  Logic Group (0x0234)
Model name: ZSO7300
ProductType should be: 0x0004 (is currently 0x0003)
ProductID should be 0x002a (is currently 0x012a)

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 8.3.2
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

There is no need for a log file. If the ProductType and ProductID are changed to what I wrote above ot works. I have attached a working zso7300.json file.
[zso7300.json.zip](https://github.com/zwave-js/node-zwave-js/files/7142335/zso7300.json.zip)

AlCalzone, 2021-09-10:
It's possible that there are two revisions of the device with different IDs. We generally don't remove ids unless they are proven wrong.
Could you provide a pull request adding your ids to the existing file?

briis, 2021-09-10:
Yes, there might be more than 1 revision. I only have 1 device and that is brand new. I have created a PR, where I added the ProductType and ProductID. Thanks

closed 2021-09-10 by briis