zwave-js/zwave-js #3327: Docs and issue template: Mention typical troubleshooting steps

AlCalzone, 2021-09-13:
Like:
* controller on extension cord
* update to 7.15 when healing fails
* ...

@blhoward2 got any additions to the list?

kpine, 2021-09-13:
- Re-interview nodes when configuration parameters don't match the device DB
- Hard pull and re-insertion of USB controller when it stops responding

blhoward2, 2021-09-15:
For the docs, should these live still Zjs2mqtt? I worry most users don't know to look here.

blhoward2, 2021-09-15:
More in no specific order:
- How to decide if the issue lies in HA or zwavejs
- Aeotec Stick 5 requires a powered usb hub if not the + model
- For the docs: I'm pushing the scene button and nothing is happening (platform specific how to get to the scene events)
- The state doesn't update when I control locally (needs polling)
- Cannot open device file error message (zwave stick is in use)
- The container starts but I can't connect to it via HA (stick not configured)
- Whatever the error says when a stick is not configured in zjs2mqtt and you start it
- My mesh is slow or buggy (heal network)
- One device works more slowly than others or not at all (heal node, and likely intermediate hop nodes)
- My lock won't pair (range issues?)
- My battery devices are stuck interviewing (multiple wake ups needed)

AlCalzone, 2021-09-16:
>     * The state doesn't update when I control locally (needs polling)

Or the associations need to be changed. Maybe the device just needs a re-interview if updating from an older driver that behaved differently. Maybe the device file needs a compat flag to hide/preserve an endpoint.

AlCalzone, 2021-09-17:
* Apparently some S2-capable battery-powered devices give you a grand total of 20 seconds from starting the inclusion to entering the PIN (ZCOMBO being one example)

mhutch, 2021-10-09:
> * Apparently some S2-capable battery-powered devices give you a grand total of 20 seconds from starting the inclusion to entering the PIN (ZCOMBO being one example). According to the specs this should be 8 minutes (4 for granting keys, 4 for entering PIN).

@AlCalzone I suspect something like this is happening with my Ring gen 2 motion sensor, given that the manual for the device describes entering the PIN into the controller _before_ activating "classic inclusion" on the device.

AlCalzone, 2021-10-15:
I've started working on this in https://github.com/zwave-js/node-zwave-js/pull/3530
If anyone wants to help out or chime in, please go ahead. It is very much appreciated.