zwave-js/zwave-js #3345: Missing device configuration: MCOHome A8-9 (different product ID)

icecreamfarmer, 2021-09-17:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

MCOHome A8-9

### Manufacturer ID

0x015f

### Product Type

0xa803

### Product ID

0x1352

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/3995?selectedFrequencyId=-1

### Do you have a manual?

Similar product is listed here: https://devices.zwave-js.io/?jumpTo=0x015f:0xa803:0x135a:0.0
But the device ID seemed to have changed from 0x135a to 0x1352
In Zwave JS to MQTT it is now listed as Device ID 351-4946-43011 (0x015f-0xa803-0x1352).
But no config is available.

AlCalzone, 2021-09-17:
Let me get this straight: Your device (which is a MCOHome A8-9) has product ID `0x1352`, we have a file for `0x135a`?

icecreamfarmer, 2021-09-17:
Yes exactly that. Maybe it is a revision of the same product. Manual and product name is the same.

AlCalzone, 2021-09-18:
@zwave-js-bot add fingerprint 0xa803:0x1352 to 0x015f:0xa803:0x135a:0.0

PR #3356 (2021-09-18): fix(config): add fingerprint `0xa803:0x1352` to McoHome A8-9
fixes: #3345
closed 2021-09-18 by AlCalzone