zwave-js/zwave-js #3346: [question] Aeotech Gen 2 DSB28

haavardNO, 2021-09-17:
I have a Aeotech Gen DSB28. I have been using this device for several years and it has been working fine.
Used it with different controllers and it has worked fine. It also worked in the beginning with zwave2mqtt. But at some point it stopped to work. Due to several reasons I have not been able to report it until now.

I am able to add the device, and it looks ok. But I am not able to get it to auto report. I have tried to use the UI to set it up. And also the aeotech excel sheet and do it manually by setting parameter 101-103 and 111-113. 
When I click refresh, I am able to poll new values.
Any ideas?
[image]

[image]

[node_106.json.zip](https://github.com/zwave-js/zwavejs2mqtt/files/7184734/node_106.json.zip)

AlCalzone, 2021-09-17:
Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs), loglevel `debug` or `silly` and attach it here as a file.
Also, seeing the configured associations would be good.

haavardNO, 2021-09-17:
No problem. 
See attached.
There seams to be no groups set up. Could that be the cause?
[image]

[zwavejs2mqtt-store-8.zip](https://github.com/zwave-js/node-zwave-js/files/7185764/zwavejs2mqtt-store-8.zip)

haavardNO, 2021-09-17:
That fixed the issue.
I have no idea when that might have happened, since I dont play with groups at all. But when I add the following it works.
[image]

Thanks for your help @AlCalzone

AlCalzone, 2021-09-17:
Did you recently do a network heal? Maybe the route was removed but not re-added due to a communication error in the process.

haavardNO, 2021-09-17:
Hmm did a lot of faultfinding. Including a exclude reset and re include in to network...

AlCalzone, 2021-09-17:
Ok, so no clear cause then. At least we found it.

closed 2021-09-17 by AlCalzone

haavardNO, 2021-09-17:
Thanks again 😊

robertsLando, 2021-09-17:
Weird that in the node export the lifeline association was there 🤷🏼‍♂️

kpine, 2021-09-17:
> 
> 
> Weird that in the node export the lifeline association was there 🤷🏼‍♂️

Are you talking about the `"groups"` field in the node JSON? If so, I've noticed that it just shows all the groups the the device supports, not the ones that are currently defined.

If I understand correctly, you are using [`driver.controller.getAllAssociationGroups(nodeId)`](https://github.com/zwave-js/zwavejs2mqtt/blob/713815da3711179e218171922f7adc84df9b77c0/lib/ZwaveClient.ts#L763) to export the groups, and that returns all supported groups. If your goal is to show the currently defined associations, I think you would want to use [`driver.controller.getAllAssociations(nodeId)`](https://zwave-js.github.io/node-zwave-js/#/api/controller?id=managing-associations) instead.

robertsLando, 2021-09-17:
Sorry, you are right, those are available groups, I fetch groups only when user ask for them in the page. Stupid me 😆