zwave-js/zwave-js #3349: [bug] Everything stopped responding

thecobra666, 2021-09-17:
### Version

**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [x] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - yarn install - yarn run build )

zwavejs2mqtt: 5.6.0
zwave-js: 8.3.1
home id: 3439378454
home hex: 0xcd00c016

### Describe the bug

All nodes stop responding to switch on/off requests. Not via the control panel, not via home assistant. Sensor data still gets send but, nothing else. No commands. Everything worked fine until I tried to add a new Qubino Mini Dimmer with S2 mode. The inclusion window appeared but nothing came asking for a S2 key. I then reset the device and tried to add it again via S0 mode, but the same happend. 

Yet when restarting the docker container, the node showed up but as unknown. it also stayed like that. Since then nothing is working correctly. The "death" nodes have been removed.

I'm unsure what is going on.

Unplugged the aotec stick, restarted the docker container. Switched the breaker off and on again, doesn't help. I can only assume this a bug somewhere..
[zwavelog.txt](https://github.com/zwave-js/zwavejs2mqtt/files/7185550/zwavelog.txt)

I did notice that NODE 30 times out, yet when switched on via a physical switch it responds back. Then minutes later a light suddenly goes on (that I requested minutes ago). What is going on? 

```
2021-09-17T12:40:09.561Z SERIAL « 0x01090004001e032503ff36                                            (11 bytes)
2021-09-17T12:40:09.563Z CNTRLR   [Node 030] [~] [Binary Switch] currentValue: false => true        [Endpoint 0]
2021-09-17T12:40:09.566Z SERIAL » [ACK]                                                                   (0x06)
2021-09-17T12:40:09.568Z DRIVER « [Node 030] [REQ] [ApplicationCommand]
                                  └─[BinarySwitchCCReport]
                                      current value: true
2021-09-17T12:40:10.563Z SERIAL « 0x01100004001e0a32022134000000360000ec                              (18 bytes)
2021-09-17T12:40:10.564Z CNTRLR   [Node 030] [~] [Meter] value[66049]: 1.5 => 5.4                   [Endpoint 0]
2021-09-17T12:40:10.568Z SERIAL » [ACK]                                                                   (0x06)
2021-09-17T12:40:10.570Z DRIVER « [Node 030] [REQ] [ApplicationCommand]
                                  └─[MeterCCReport]
                                      type:       Electric
                                      scale:      W
                                      rate type:  Consumed
                                      value:      5.4
                                      time delta: 0 seconds
2021-09-17T12:40:13.139Z SERIAL » 0x010900131e02260225f008                                            (11 bytes)
2021-09-17T12:40:13.140Z DRIVER » [Node 030] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      240
                                  └─[MultilevelSwitchCCGet]
2021-09-17T12:40:13.143Z CNTRLR   [Node 030] Timed out while waiting for a response from the node
2021-09-17T12:40:13.145Z SERIAL « [ACK]                                                                   (0x06)
2021-09-17T12:40:13.149Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-09-17T12:40:13.150Z SERIAL » [ACK]                                                                   (0x06)
2021-09-17T12:40:13.151Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-09-17T12:40:13.251Z SERIAL « 0x01070013f000000b10                                                 (9 bytes)
2021-09-17T12:40:13.252Z SERIAL » [ACK]                                                                   (0x06)
2021-09-17T12:40:13.253Z DRIVER « [REQ] [SendData]
                                    callback id:     240
                                    transmit status: OK
2021-09-17T12:40:23.261Z SERIAL » 0x010900131e02260225f109                                            (11 bytes)
2021-09-17T12:40:23.262Z DRIVER » [Node 030] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      241
                                  └─[MultilevelSwitchCCGet]
2021-09-17T12:40:23.264Z CNTRLR   [Node 030] Timed out while waiting for a response from the node
2021-09-17T12:40:23.268Z SERIAL « [ACK]                                                                   (0x06)
2021-09-17T12:40:23.272Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-09-17T12:40:23.274Z SERIAL » [ACK]                                                                   (0x06)
2021-09-17T12:40:23.275Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-09-17T12:40:23.314Z SERIAL « 0x01070013f10000051f                                                 (9 bytes)
2021-09-17T12:40:23.315Z SERIAL » [ACK]                                                                   (0x06)
2021-09-17T12:40:23.316Z DRIVER « [REQ] [SendData]
                                    callback id:     241
                                    transmit status: OK
```

This file is a json so rename the extension. [nodes_dump.txt](https://github.com/zwave-js/zwavejs2mqtt/files/7185720/nodes_dump.txt)

//edit: disabling the ws socket to home assistant and subsquent reboot of the zwavejs driver, resulted in everything working after a re-interview of some nodes. Now i'm baffled as to why this is happend. 

//edit2: after connecting back the WS socket option and reboot of the driver, I've now lost the dimming entities for the qubino dimmer (node 30)... is there something corrupted? I have already re-interviewd it. A lot of the parameters are mentioned as "0 custom"? 

[image]

AlCalzone, 2021-09-17:
Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs) which captures the problem, loglevel `debug` or `silly` and attach it here as a file.

thecobra666, 2021-09-17:
> Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs) which captures the problem, loglevel `debug` or `silly` and attach it here as a file.
[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/7188085/zwave_js.log)

Attached is the log when the re-interview has taken place.

The multilevel switch is gone.
[image]

Below the same device but how it should be (and how it was).

[image]

AlCalzone, 2021-09-17:
> The multilevel switch is gone.

That is because the device says so:
```
2021-09-17T19:31:11.277Z DRIVER « [Node 030] [REQ] [ApplicationCommand]
                                  └─[VersionCCCommandClassReport]
                                      CC:      Multilevel Switch
                                      version: 0
2021-09-17T19:31:11.280Z CNTRLR   [Node 030]   does NOT support CC Multilevel Switch (0x26)
```

According to the device manual, the CC is only present in dimmer mode:
https://qubino.com/wp-content/uploads/2019/09/Qubino_Mini-Dimmer-PLUS-extended-manual_eng_3.4_09092019.pdf

[image]

Is it possible that you changed that parameter at some point and did not exclude and re-include the device?

thecobra666, 2021-09-19:
> That is because the device says so:
>
> ```
> 2021-09-17T19:31:11.277Z DRIVER « [Node 030] [REQ] [ApplicationCommand] [...]

That will be the reason. I have excluded and included the device back and now it´s back. 

I also think I know why everything stopped working: when installing the new dimmer I disabled the breaker (for obvious reasons), in the background HA was still trying to poll one dimmer every second. After time all those messages became back logged. When the power came back on, HA or zwavejs2mqtt tried to send them out. 

Now everything works fine.

closed 2021-09-19 by thecobra666