zwave-js/zwave-js #3350: feat(cc): add compatibility flag(s) to Scene Controller Configuration CC

Ikcelaks, 2021-09-17:
I'm opening this issue to consolidate planning for Compat flag strategies for the Scene Controller Configuration CC.

zwave-js/zwavejs2mqtt#1640 exposed two ways that devices fail to follow the specs:

1. Some devices support more `groupId`s in the `SceneControllerConfigurationCC` than in `AssociationCC`.

I propose this be handled with a `compat` flag that directly specifies the number of groups the `SceneControllerConfigurationCC` supports. Along with this, a value to track this separately will be added to the `valueDB`. For devices that follow the specs, this value will just be set equal to the group count reported by the `AssociationCC`.

2. Some devices are write-only and don't respond to `SceneControllerConfigurationCC.Get` commands.

No "compat" flag required, imo. We should just switch to the `UserCode` interview style of setting MetaData and not even trying `Get` commands. Most users won't even care about the existing values, and, if they do, they can refresh values and hope their device responds.

AlCalzone, 2021-09-17:
> Along with this, a value to track this separately will be added to the `valueDB`

Not necessary IMO. When the value is needed, use `node.deviceConfig?.compat?.newFlag ?? <no of association groups>` (don't know the exact syntax without looking at the code.

The rest I agree with.

Ikcelaks, 2021-09-18:
Didn't have much time today, but I have a PR mostly written.

One question, the "interview" now no longer actually interviews the node, except that we still need require the `Association CC` interview and set the MetaData. So, eslint is complaining that I don't have an `await` in the `async` method. Should I just have it ignore eslint there?

PR #3354 (2021-09-19): feat(cc): add `compat` for `Scene Controller Configuration CC`
Resolves #3350 
Should also resolve zwave-js/zwavejs2mqtt#1640

- `forceSceneControllerGroupCount` compat flag sets the `groupCount` for devices that don't adhere to the 1:1 corespondence between `Scene Controller Configuration CC` groups and `Association CC` groups. This is know to be needed by the "Leviton VRCZ4-M0Z".
- The `SceneControllerConfigurationCC.interview` no longer queries the node for the scene configuration of each group. Instead, it only presets the `ValueMetaData` for each group. This speeds up the interview and av [... cut]
closed 2021-09-19 by AlCalzone