zwave-js/zwave-js #3351: Configuration Parameter 7 is missing from Philio PSM02

orandasoft, 2021-09-18:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Philio PSM02 (and similar Philio devices) use notification report rather sensor binary report for motion by default and as a result motion may not properly work,

The non-documented parameter 7 allows the sensor binary report to be enabled.

bit 1 (decimal 2) enable motion off report
bit 2 (decimal 4) enable super PIR sensitivity
bit 4 (decimal 16) 0 = notification report, 1 = sensor binary report
bit 5 (decimal 32) disable multi CC in auto report
bit 6 (decimal 64) disable report battery state when device triggered

### Device information

Manufacturer: Philio
Model name: PSM02
Node ID in your network: 4

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: Driver Version: 8.2.1, Server Version: 1.10.1
`node-zwave-js` branch:
`zwavejs2mqtt` branch: 0.24.1 

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwavejs_2021-09-18 (4).log](https://github.com/zwave-js/node-zwave-js/files/7189589/zwavejs_2021-09-18.4.log)

AlCalzone, 2021-09-18:
Many Philio config files are terrible, but we should at least add the param. Accepting PRs.

orandasoft, 2021-09-19:
I am a newbie here. I assume PR stands for Pull Request. I will read through the documentation and try to submit one.

You mentioned that the Philio config files are terrible. In what way could they be improved? Looking at the one for the PSM02, it seems to contain the information which is publicly available: https://devices.zwave-js.io/?jumpTo=0x013c:0x0002:0x0002:0.0

AlCalzone, 2021-09-19:
> I assume PR stands for Pull Request.

Bingo

> In what way could they be improved?

We have a section in the docs detailing what we consider a good config file:
https://zwave-js.github.io/node-zwave-js/#/config-files/contributing-files

The Philio files often have labels/descriptions that don't yet match our style guide. And the parameters 6/7 that some of their devices have are only half documented there (min 0, max 127, user has to figure out what they do).

Here's an example of bad one: https://devices.zwave-js.io/?jumpTo=0x013c:0x0002:0x000d:0.0
And here's a better version of a similar device, using [partial parameters](https://zwave-js.github.io/node-zwave-js/#/config-files/partial-parameters) to describe the user function settings: https://devices.zwave-js.io/?jumpTo=0x013c:0x0002:0x000e:0.0

orandasoft, 2021-09-19:
Just to confirm, when creating a pull request, I pull the whole node-zwave-js and make changes to the Philio PSM02 config file?

AlCalzone, 2021-09-19:
Yep

orandasoft, 2021-09-22:
Now that changes have been merged for PSM02, shall I make updates to PST02A and PST02B as well as they are very similar changes?

AlCalzone, 2021-09-22:
Sure, go for it :)

AlCalzone, 2021-09-22:
fixed by #3373

closed 2021-09-22 by AlCalzone