zwave-js/zwave-js #3352: Missing device configuration: GoControl WO15EMZ5-1

pdobrien3, 2021-09-18:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

GoControl Smart Wall Outlet

### Manufacturer ID

0x574f

### Product Type

0x014f

### Product ID

0x3535

### Firmware Version

1.35

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/2527

### Do you have a manual?

https://products.z-wavealliance.org/ProductManual/File?folder=&filename=MarketCertificationFiles/2527/WO15EMZ5-1%20Instructions.pdf

pdobrien3, 2021-09-18:
Also found it under the OpenHAB at https://opensmarthouse.org/zwavedatabase/1186

AlCalzone, 2021-09-18:
@zwave-js-bot import config 2527 from zwa

@pdobrien3 the bot should create a PR with a starter template. That likely needs to be cleaned up to match our style guide though.

blhoward2, 2021-09-18:
The easiest way to edit this is for you to push a PR onto the bot's branch.

PR #3357 (2021-09-19): feat(config): add Nortek WO15EMZ5
fixes: #3352

**TODO:**
- [x]	Change PR title to be more specific
closed 2021-09-19 by AlCalzone