zwave-js/zwave-js #3353: Aeotec DSB05 4 in 1 Multisensor reporting motion as "Any"

sotatech, 2021-09-18:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

This multisensor works well with 3 of it's 4 sensors, apart from the motion detection. Motion is reported as a Binary Sensor [4-48-0-Any] Any. It responds true/false to motion but HA then sees it as a generic sensor rather than a motion sensor.

I have set parameter [4-112-0-5] Motion Sensor Report Type to Send to [2] Binary Sensor CC Report as that was how it was configured the older Z-Wave integration.

### Device information

Manufacturer: Aeotec
Model name: 4 in 1 Multisensor f/w 1.18 (latest)
Node ID in your network: 4

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 8.3.1
`zwavejs2mqtt` branch: 5.6.0

### Did you change anything?

yes (please describe)

### If yes, what did you change?

I tried changing Motion Sensor Report Type to Send to Basic CC Report, but it made no difference.

### Did this work before?

Yes (please describe)

### If yes, where did it work?

I have a number of these multisensors working and reporting motion correctly under the original Z-Wave integration in Home Assistant.

### Attach Driver Logfile

As I'm just starting the migration to ZWave-JS, there are just 2 nodes and the DSB05 is node 4.

[zwavejs_2021-09-18.log](https://github.com/zwave-js/node-zwave-js/files/7189923/zwavejs_2021-09-18.log)

AlCalzone, 2021-09-18:
That is expected. The node uses version 1 of the `Binary Sensor CC` which doesn't carry any information which kind of sensor this is. According to the Z-Wave Alliance entry, this device announces itself as a `Routing Sensor Binary`, which isn't any more specific.

Since HomeAssistant seems to rely on the sensor type, I guess this is more of a HA issue to do something useful with unspecified binary sensors.

closed 2021-09-18 by AlCalzone

sotatech, 2021-09-18:
OK, thank you. I can either customize the sensor in HA or else upgrade - they are 6 or 7 years old :)

AlCalzone, 2021-09-18:
I've linked the HA devs to this issue - maybe that's something that can be improved in HA. (or it is already possible, I don't know 😅)