zwave-js/zwave-js #3355: fix(config): add `forceSceneControllerGroupCount` compat flag for Leviton VRCZ4

Ikcelaks, 2021-09-18:
This depends on #3354 
@blhoward2 reported this in zwave-js/zwavejs2mqtt#1640
https://blog.gruby.com/2019/01/04/setting-up-a-leviton-vrcz4-m0z-for-use-with-home-assistant/

The device has 4 `Association CC` groups but supports 8 `Scene Controller Configuration CC` groups. The `forceSceneControllerGroupCount` flag should be set to 8.

AlCalzone, 2021-09-19:
@blhoward2 we have two VRCZ4: [VRCZ4](https://devices.zwave-js.io/?jumpTo=0x001d:0x0701:0x0206:0.0) and [VRCZ4-MR](https://devices.zwave-js.io/?jumpTo=0x001d:0x1202:0x0243:0.0). Which one do you have? I suspect we have to add the flag to both, but I'd like to verify.

j9brown, 2021-09-19:
I think this flag is also required for VRCS4.

blhoward2, 2021-09-19:
Agreed, both. The difference is the one has a relay and the other is purely a scene controller. They work the same way.

AlCalzone, 2021-09-19:
If anyone has time, go ahead

j9brown, 2021-09-19:
Yep, there are several variants of VRCS4 and VRCZ4, all of which have 8 scene groups.

FWIW, in my house I have: VRCS4-M0Z, VRCS4-MRZ, VRCZ4-MRX, and VRCZ4-MRZ.

Thanks for taking care of this!