zwave-js/zwave-js #3359: ZWave Devices Fail to Interview, Network Stability Issues, Most devices assume within range of controller

brgaulin, 2021-09-18:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

This is really hard to describe, but I'm totally lost when it comes to stabilizing my ZWave Network. The issue I am having is that I can't reliably interview any nodes on the network, and even if I do succeed, 50-60% of the time, commands are delayed 30-60s(ex. In home assistant, click turn on light, and a minute later it turns on). Some devices in the deployment work flawless everytime, but I have probably 7-8 switches that are not stable at all. Contacting Zooz Support and they typically point the finger at ZWaveJS, so I figured I would ask here if anyone had ideas or maybe had seen this before.

Description of Deployment:
- 35 Powered Dimmers / Wall Switches as repeaters
- Home Layout: Basement + Living Floor + Bedroom Floor + Attic (4 stories total) typical US wood / drywall Colonial
- Each floor is approximately 900 sqft
- ZWave Stick is centrally located in the basement, about 1ft below the ceiling(3ft USB 3.0 Extension from UGREEN)
- There are 3 switches on the same floor approximately 4' from the stick, and another 2 at the top of the stairs to the first floor
- 2 Battery multi sensors, and 2 battery locks are present, but those seem to all work really well
- Switches are mostly a mix of Zooz Zen27, Zen 76, Zen77
- Only the Locks were added secure, the rest were added insecure

Overall: Controller placement is sub-optimal, but I would have expected the mesh nature of ZWave to make this a non-issue.

Server Environment:
- Rack mounted Ryzen system
- Ubuntu 20.04 w/ HAOS VM via KVM
- Only other connected USB device is a Cyberpower UPS which is monitored with APCUPSD in a docker container on the Ubuntu Host OS

Observations:
 - Almost every device in the home thinks it has a strong connection to the controller, very few use the switches as repeaters. Including switches that are in the Attic, they are using a direct connection to the controller and are fairly unstable
 - Polling is disabled on the entire network, and no excess traffic is observed while interviewing nodes
 - Attempted to replace ZST10 500 series with a ZST10 700 series stick, and results are actually worse
 - Network Wide Healing often fails, but I have been able to succeed in healing each device a few times and see no improvement
 - All of this behavior is consistent across the two sticks. When migrating from one to the other, I factory reset each ZWave switch and then included it again on the new stick

I guess my first question, is there any way I could designate which nodes are within range of the controller manually? Or manually exclude certain nodes, and force them to use a hop. I feel my issue is involving range, but ZWaveJS is paring their return routes directly to the controller even though they aren't able to reliably communicate that way.

### Device information

Zooz ZST10 700 series Stick

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

HA Integration Version: 0.1.40
Z-Wave JS Version:  8.3.0

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[out3.log](https://github.com/zwave-js/node-zwave-js/files/7190749/out3.log)

Note: These logs were taken after the controller was unplugged and replugged and then ZWaveJS started, but still receiving controller ack issues. Command failures, heal failures, etc.

blhoward2, 2021-09-19:
You cannot manually change the routing, as that is all handled external to us between the stick and the device. Have you tried healing individually starting with devices near the controller and working outward? This sounds very much like your routes are broken.

brgaulin, 2021-09-19:
I didn't necessarily heal them in order, but I was able to get a full network heal at one point earlier today, then I went one device at a time and healed it manually. But even after healing a device several times, rebooting it, etc. It will fail to respond during node interview

brgaulin, 2021-09-19:
Another interesting log of a node(Node 40) that is ~8' from the controller. It fails it's interview twice outright. Succeeds on the third. Then fails to heal afterwards. It responds to commands to turn on/off just fine though. Node 40 is a Zen77 Firmware 1.3 https://pastebin.com/Vxb2PKrp

blhoward2, 2021-09-19:
Have you tried a new extension cord? A different USB port?

brgaulin, 2021-09-19:
Yes, I tried the front USB 2.0 ports, rear 2.0 and 3.0 ports, a 6' amazon basics extension, and 3' UGREEN extension that was a bit beefier. Also tried a handful of positions with the extensions to see if placement would be better horizontal / vertical / open air / taped to a wall / taped to the ceiling.

blhoward2, 2021-09-19:
What else do you have operating in the 900 MHz (assuming US) band in your home or nearby? Wireless doorbells? Baby monitors? Cordless phones? Nearby pager towers?

brgaulin, 2021-09-19:
The main interference I would think is that it is under a staircase, so lots of wood framing. But the extension I've used to bring the stick out of that space and into open basement. Dryer is on the outside wall, stick is central, about 7-8' apart and is off during all these tests. Furnace is about the same distance, oil, and was also off in all these tests. And all the switches are up above, so I would hope they don't interfere much.

I think trying another zwavejs server is my next step to rule out the host.

AlCalzone, 2021-09-19:
Seeing this

```
2021-09-18T22:34:25.270Z DRIVER « [Node 045] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       1
                                      info:              of t
                                      reports to follow: 9
2021-09-18T22:34:25.626Z SERIAL « 0x011200a800012d09700d0001096f66207400a7ef                          (20 bytes)
2021-09-18T22:34:25.628Z SERIAL » [ACK]                                                                   (0x06)
2021-09-18T22:34:25.629Z DRIVER « [Node 045] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       1
                                      info:              of t
                                      reports to follow: 9
2021-09-18T22:34:25.970Z SERIAL « 0x011200a800012d09700d0001096f66207400a7ef                          (20 bytes)
2021-09-18T22:34:25.971Z SERIAL » [ACK]                                                                   (0x06)
2021-09-18T22:34:25.972Z DRIVER « [Node 045] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       1
                                      info:              of t
                                      reports to follow: 9
2021-09-18T22:34:26.857Z SERIAL « 0x011200a800012d09700d0001096f66207400b7ff                          (20 bytes)
2021-09-18T22:34:26.858Z SERIAL » [ACK]                                                                   (0x06)
2021-09-18T22:34:26.860Z DRIVER « [Node 045] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       1
                                      info:              of t
                                      reports to follow: 9
2021-09-18T22:34:26.885Z SERIAL « 0x011200a800012d09700d0001096f66207400b7ff                          (20 bytes)
2021-09-18T22:34:26.885Z SERIAL » [ACK]                                                                   (0x06)
2021-09-18T22:34:26.886Z DRIVER « [Node 045] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       1
                                      info:              of t
                                      reports to follow: 9
2021-09-18T22:34:26.953Z SERIAL « 0x011200a800012d09700d0001096f66207400b7ff                          (20 bytes)
2021-09-18T22:34:26.954Z SERIAL » [ACK]                                                                   (0x06)
2021-09-18T22:34:26.955Z DRIVER « [Node 045] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       1
                                      info:              of t
                                      reports to follow: 9
2021-09-18T22:34:27.051Z SERIAL « 0x011200a800012d09700d0001096f66207400b2fa                          (20 bytes)
2021-09-18T22:34:27.052Z SERIAL » [ACK]                                                                   (0x06)
2021-09-18T22:34:27.052Z DRIVER « [Node 045] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       1
                                      info:              of t
                                      reports to follow: 9
2021-09-18T22:34:27.155Z SERIAL « 0x011200a800012d09700d0001096f66207400b2fa                          (20 bytes)
2021-09-18T22:34:27.156Z SERIAL » [ACK]                                                                   (0x06)
2021-09-18T22:34:27.156Z DRIVER « [Node 045] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       1
                                      info:              of t
                                      reports to follow: 9
2021-09-18T22:34:27.354Z SERIAL « 0x011200a800012d09700d0001096f66207400b7ff                          (20 bytes)
2021-09-18T22:34:27.355Z SERIAL » [ACK]                                                                   (0x06)
2021-09-18T22:34:27.356Z DRIVER « [Node 045] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       1
                                      info:              of t
                                      reports to follow: 9
2021-09-18T22:34:27.385Z SERIAL « 0x011200a800012d09700d0001096f66207400b7ff                          (20 bytes)
2021-09-18T22:34:27.385Z SERIAL » [ACK]                                                                   (0x06)
2021-09-18T22:34:27.386Z DRIVER « [Node 045] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       1
[... cut ...]
I'm also not sure why the node suddenly starts to broadcast the response, but I guess it has such bad connectivity that it almost never gets any acknowledgement that the other side received its messages that goes into full-on panic mode. Side note 3: Also not a good idea to panic even harder just because one not very important command doesn't go through.

This likely also explains the other issues you're seeing. Your network is so damn flooded that the controller (and probably some of the other nodes) doesn't get enough duty cycles to actually send messages, leading to delays and failed healing attempts.

brgaulin, 2021-09-19:
AlCalzone and blhoward2, Thank you so much for helping me debug this! I know this is open source and your time is valuable, so I wanted to just give a quick shoutout!

Oh wow, I didn't notice the duplicates, for the most part I was seeing delayed messages I thought. I think there are 3 interviews during the timeframe of reports you sent, but even still, should have only sent 3 messages back, not 21. And it looks like the first message came almost 42 seconds after it was asked for. I don't think increasing the timeout in this case would be reasonable.

```
2021-09-18T22:34:13.238Z DRIVER » [Node 045] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ route:            0, 0, 0, 0
                                  │ callback id:      43
                                  └─[ConfigurationCCInfoGet]
                                      parameter #: 1
2021-09-18T22:35:19.419Z DRIVER » [Node 045] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ route:            0, 0, 0, 0
                                  │ callback id:      43
                                  └─[ConfigurationCCInfoGet]
                                      parameter #: 1
2021-09-18T22:35:52.510Z DRIVER » [Node 045] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ route:            0, 0, 0, 0
                                  │ callback id:      50
                                  └─[ConfigurationCCInfoGet]
                                      parameter #: 1
```

I agree and that this feels like a connectivity issue. Node 40 is interesting, as it is really close to the controller. I'm wondering if maybe I have a bad device? 

I'll have to poke around my network and see if I can replicate that panicking behavior. 

While I was watching the logs, it looked like some of my home assistant automations were partially to blame for my connectivity issues. Things like: Asking a light to transition while it is already at its max brightness. And using scene.create to capture the current state of 5-6 switches and then minutes later apply that scene. Moving to simpler commands seems to have cleared up many of my slowdown issues at the runtime phase, but would still love to troubleshoot why I can't interview certain nodes.

For that I think my next steps are to see what other nodes are doing during interviews. I'll report back here if any of them are not panicking, but doing something else. And otherwise I'll engage Zooz support to see if they have any tips.

brgaulin, 2021-09-19:
🤦‍♂️ Would help if I open my eyes. Would a microwave being right behind that switch pose some interference? Think of that wall being open on the backside with a counter, and a microwave pressed right up against it. They are different electrical circuits, but I'd bet it has something to do with a big metal box being right there

blhoward2, 2021-09-19:
One theory you may try as devices appear to be spamming. Kill power to as much of your network as you can. Restore power near the stick, heal, and interview. Try to do that working out waiting on interviews to finish.

AlCalzone, 2021-09-19:
Big metal boxes might still have some influence 😅

brgaulin, 2021-09-19:
Not as bad, but here were the logs after removing the microwave. It still exhibited timeouts, but did overall complete the interview: https://pastebin.com/nVdbxJL4

blhoward2, 2021-09-19:
> Big metal boxes might still have some influence 😅

Not to mention when the microwave is running. They are, after all, RF generators. I wouldn't actually be surprised if one could fry a nearby stick after repeated use. Consumer microwaves are supposed to work at 2.5 GHz according to google, but commercial ones work at 915 MHZ so it's possible.

blhoward2, 2021-09-19:
> Not as bad, but here were the logs after removing the microwave. It still exhibited timeouts, but did overall complete the interview: https://pastebin.com/nVdbxJL4

Given how small Zooz breaks down their configuration parameter packets and the sheer volume, some timeouts isn't necessarily a bad omen especially if other unseen traffic is bogging down the mesh. I have a few Zooz products that do this, especially if they're a bit removed from the stick. 

A sniffer would help pinpoint this if you have an old stick lying around.

AlCalzone, 2021-09-19:
It starts out pretty good, but gradually gets worse. First you get some corrupted frames which are also re-transmitted (notice the parameter #):
```
2021-09-19T15:17:16.196Z DRIVER « [Node 040] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       10
                                      info:              99,0
                                      reports to follow: 3
2021-09-19T15:17:16.893Z SERIAL « 0x011200a800012809700d000a0339392c3000b6bb                          (20 bytes)
2021-09-19T15:17:16.893Z SERIAL » [ACK]                                                                   (0x06)
2021-09-19T15:17:16.894Z DRIVER « [Node 040] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       10
                                      info:              99,0
                                      reports to follow: 3
2021-09-19T15:17:17.602Z SERIAL « 0x011200a800012809700d000a0339392c3000b6bb                          (20 bytes)
2021-09-19T15:17:17.602Z SERIAL » [ACK]                                                                   (0x06)
2021-09-19T15:17:17.603Z DRIVER « [Node 040] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       10
                                      info:              99,0
                                      reports to follow: 3
2021-09-19T15:17:18.412Z SERIAL « 0x011200a800012809700d090f0339392c3000babb                          (20 bytes)
2021-09-19T15:17:18.413Z SERIAL » [ACK]                                                                   (0x06)
2021-09-19T15:17:18.413Z DRIVER « [Node 040] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       2319
                                      info:              99,0
                                      reports to follow: 3
2021-09-19T15:17:18.757Z SERIAL « 0x011200a800012809700d090f0339392c3000b7b6                          (20 bytes)
2021-09-19T15:17:18.758Z SERIAL » [ACK]                                                                   (0x06)
2021-09-19T15:17:18.758Z DRIVER « [Node 040] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       2319
                                      info:              99,0
                                      reports to follow: 3
2021-09-19T15:17:19.133Z SERIAL « 0x011200a800012809700d090f0339392c3000babb                          (20 bytes)
2021-09-19T15:17:19.134Z SERIAL » [ACK]                                                                   (0x06)
2021-09-19T15:17:19.135Z DRIVER « [Node 040] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       2319
                                      info:              99,0
                                      reports to follow: 3
2021-09-19T15:17:19.244Z SERIAL « 0x011200a800012809700d090f0339392c3000a9a8                          (20 bytes)
2021-09-19T15:17:19.245Z SERIAL » [ACK]                                                                   (0x06)
2021-09-19T15:17:19.246Z DRIVER « [Node 040] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       2319
                                      info:              99,0
                                      reports to follow: 3
```

and then the flooding is back:
```
2021-09-19T15:17:19.268Z DRIVER « [Node 040] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       10
                                      info:              maxi
                                      reports to follow: 10
2021-09-19T15:17:19.598Z SERIAL « 0x011200a800012809700d000a0a6d61786900b9bc                          (20 bytes)
2021-09-19T15:17:19.599Z SERIAL » [ACK]                                                                   (0x06)
2021-09-19T15:17:19.600Z DRIVER « [Node 040] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       10
                                      info:              maxi
                                      reports to follow: 10
2021-09-19T15:17:19.926Z SERIAL « 0x011200a800012809700d000a0a6d61786900babf                          (20 bytes)
2021-09-19T15:17:19.928Z SERIAL » [ACK]                                                                   (0x06)
2021-09-19T15:17:19.929Z DRIVER « [Node 040] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       10
[... 232 lines cut ...]
[... cut ...]
2021-09-19T15:17:53.856Z DRIVER « [Node 040] [REQ] [BridgeApplicationCommand]                          [partial]
                                  └─[ConfigurationCCInfoReport]
                                      parameter #:       10
                                      info:              maxi
                                      reports to follow: 10
```

We could think about forcing V1 of the `Configuration CC` with these devices. Splitting the text into 4-byte chunks is just unnecessarily conservative. Even the smallest frames can fit a 46-byte payload - these contain 9 bytes...

brgaulin, 2021-09-19:
I have a ZST10 500 series, is it possible to install zniffer on that? I've made it most of the way through the tutorials on setting up ZWave Programmer, but am not sure what firmware I should try and flash onto it. Wondering if instead I should pick up a separate stick that is more purpose built for it. Is there a recommended stick these days? UZB?

The only zniffer hex file I have is `ZW_ZnifferPTI_7.16.2_267_ZGM130S_REGION_US.hex` from this location `C:\SiliconLabs\SimplicityStudio\v5\developer\sdks\gecko_sdk_suite\v3.2\protocol\z-wave\Apps\bin`, but I'm not sure if that'll work/

AlCalzone, 2021-09-19:
You should be able to flash the Zniffer firmware on any 500-series sticks. The one you mentioned is 700-series though. You'll need the 6.8x version of the SDK.

brgaulin, 2021-09-19:
I downloaded SDK_v6_84_00.zip and don't see it in there, should I instead install simplicity studio v4?

AlCalzone, 2021-09-19:
Send me an e-mail to the address in my profile and I'll send you the Zniffer firmware and the PC programmer. Both are hard to find.

brgaulin, 2021-09-19:
To end the tangent of Zniffer setup, you can get the Zniffer_v4_60.zip from the search box in Simplicity v4. I did a search for `zniffer` and then went to Miscellaneous, and it was listed as `Z-Wave Zniffer 4.60`. That contains the tool, programming driver, and firmware. I then used the programmer which you can still find on the silabs website (or also grabbed via the simplicity v4 search under `Z-Wave Programmer`. Search for `programmer` and look under Miscellaneous)

brgaulin, 2021-09-19:
@blhoward2 I think you are on to something with turning off nodes, healing the ones closest to the controller first, etc.

I'm seeing this in zniffer, which if I'm reading it right is rather troubling: 
[image]

Notes on placement:
Controller is in Basement
Node 40 is above it on first floor
Node 2 is in the Attic
Node 17 is on the 2nd Floor

AlCalzone, 2021-09-19:
Keep in mind that the routes may sometimes seem unintuitive because of how RF communication works, but that seems weird.

> healing the ones closest to the controller first, etc.

A full network heal does that, as long as the controller knows which nodes are its neighbors. It will heal those first, then the ones that take 1 hop, then 2 hops, then 3.

Your earlier attempts didn't really work because of node 45, so maybe start by turning that one off and trying a full heal.

brgaulin, 2021-09-19:
Totally understand the unintuitive RF comment. I know physical proximity isn't the only need. But in that case, the signal will travel right past Node 17 to get to node 2. I can't imagine an RF scenario where 2 is a stronger signal than 17, so that really should have been Node 40 -> 17 -> 1. I'm also really shocked 40 isn't direct connected or going through 10.  Will have to watch signal strength and see if I can figure out what is going on, but 3 hops for a node 6-8' from the controller seems wild. Especially with so many other switches closer to it that it could communicate with

brgaulin, 2021-09-21:
I disconnected a bunch of nodes, and was able to get a full network heal, twice. But certain nodes are still not very responsive. Question about Zniffer if anyone happens to know. "Routed Error", are those anything to worry about? or "CRC_ERROR". I was reading that CRC_ERROR might be that my zniffer is out of range, but wasn't sure what a Routed Error was. I've been seeing a lot of Routed Errors. 5-10 every time I send a command for a certain light to turn on/off.

Would it be beneficial to rebuild my network(exclude everything, reset controller, reset zwavejs, and add one node at a time watching for errors / instability?). And if so, are there any guidelines I should be following. I have a few ideas in my head, but wasn't sure if there was a doc anywhere.

AlCalzone, 2021-09-21:
CRC_Error could indicate that something corrupted the reception of the command. I have that sometimes, even if the Zniffer is immediately next to the controller.

According to https://forum.fibaro.com/topic/50752-zniffer-analysis-of-fgwp-102-mesh-update/, "routed error" means that something along the route failed, and that this message should show you which part of the route is bad.

> Would it be beneficial to rebuild my network

I don't think so if the locations don't change. If you do, it's probably best if you include from the controller outwards so the further away nodes already have a stable mesh they can hook into.

brgaulin, 2021-09-22:
I took the zwave stick and made it mobile. Had an old laptop, installed Ubuntu + Docker + ZwaveJS2MQTT and brought over my config from home assistant. I moved it to a central space on the first floor and after a few heals I'm unfortunately seeing the same behavior. I don't think it was controller position / host issues. I'll let it settle overnight and do another heal in the morning. But the heal seems to work a lot better now, overall I find the network much more stable with the more central location. But I still have 5-6 nodes that are having a lot of routing errors, something like 70-90% of the traffic going to them seems to fail

AlCalzone, 2021-09-22:
> But I still have 5-6 nodes that are having a lot of routing errors, something like 70-90% of the traffic going to them seems to fail

You could try to figure out with the Zniffer at which point it fails and zniff there if there's something weird about that location.

brgaulin, 2021-09-22:
Ok, progress? Last night I used zniffer to detect any nodes totally misbehaving, and I excluded / included those nodes. After the network heal today, all seems well. I had a couple routing errors, but the commands still executed perfectly fine, and I'm able to interview pretty much any node on the network with no issues.

Overall, it seems my controller was getting paired with nodes that it didn't have a strong signal with, which caused a lot of my initial issues. Relocating the stick to a better location a floor up seems to resolve most of the issues. I've seen a few other users do deployments of zwavejs2mqtt on a pi then linking that to home assistant, I may try and go that route, just need to find a good place to put the controller where it won't get disturbed.

I also have one node I might have fried(its connected to a fluorescent light, manufacturer says don't do that). So I also removed that for the later half of the testing and will replace it plus the light fixture this weekend. That may have been the cause of some of the interference, but it was still not resolved until re including the nodes corrupted by that interference.

@AlCalzone and @blhoward2 Thank you a ton for sticking with me to try and debug these, I'll keep you posted if anything else comes up, but I think I'm good to go.

brgaulin, 2021-09-23:
Ok, server rack under a staircase is not the place for a zwave stick lol. I put it back down there to see if it would work, and my whole network went haywire again. Rebooted the house and moved it back upstairs and all is good again. That was likely the main issue

closed 2021-09-23 by brgaulin