zwave-js/zwave-js #3360: Missing device configuration: Fibaro FGT001

klim123123, 2021-09-18:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Fibaro FGT001

### Manufacturer ID

0x010f

### Product Type

0x1301

### Product ID

0x4001

### Firmware Version

4.7

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/3636

### Do you have a manual?

Looks like for Russian market Fibaro made specific device with Product Type 0x1301 and ProductID 0x4001
Please add this ProductID for https://devices.zwave-js.io/?jumpTo=0x010f:0x1301:0x1000:0.0

[image]
[image]

blhoward2, 2021-09-19:
@zwave-js-bot add fingerprint 0x1301:0x4001 to 0x010f:0x1301:0x1000:0.0

closed 2021-09-19 by AlCalzone

klim123123, 2021-09-26:
zwavejs2mqtt: 5.6.0
zwave-js: 8.3.1

Re-paired device and it's still Unknown product 0x4001

AlCalzone, 2021-09-26:
1. No need to re-pair. A re-interview is enough.
2. https://github.com/zwave-js/node-zwave-js/releases/tag/v8.3.1 was released 12 days ago. This PR was merged 7 days ago, so the file cannot be included in 8.3.1 :)
Wait for 8.4.0 which is in the release pipeline right now.

PR #3364 (2021-09-19): fix(config): add fingerprint `0x1301:0x4001` to "Fibargroup FGT001"
fixes: #3360