zwave-js/zwave-js #3367: Extended serial API setup command bitmask isn't parsed correctly

AlCalzone, 2021-09-19:
```
2021-09-18T22:33:12.732Z DRIVER « [RES] [SerialAPISetup]
                                    command: GetSupportedCommands
                                    payload: 0xfe1601030001000000010000000000000001
2021-09-18T22:33:12.737Z CNTRLR   supported serial API setup commands:
                                  · GetSupportedCommands
                                  · SetTxStatusReport
                                  · unknown (0x03)
                                  · unknown (0x05)
                                  · unknown (0x09)
                                  · GetLRMaximumPayloadSize
                                  · unknown (0x12)
                                  · unknown (0x21)
                                  · unknown (0x41)
                                  · unknown (0x81)
```

blhoward2, 2021-09-19:
Is it a parsing problem, or that we're missing those commands?

```
export enum SerialAPISetupCommand {
	Unsupported = 0x00,
	GetSupportedCommands = 0x01,
	SetTxStatusReport = 0x02,
	SetPowerlevel = 0x04,
	GetPowerlevel = 0x08,
	GetMaximumPayloadSize = 0x10,
	GetLRMaximumPayloadSize = 0x11,
	GetRFRegion = 0x20,
	SetRFRegion = 0x40,
	SetNodeIDType = 0x80,
}
```

AlCalzone, 2021-09-19:
Those are fine but the problem is that they represent the bits' values and I'm using a bitmask parse function expecting them to be 1,2,3,...

AlCalzone, 2021-09-21:
Hmm... I compared with the specs. My parser code is fine, but the bitmask that is sent seems to be off by 1 bit...

AlCalzone, 2021-09-21:
Confirmed with another stick. Either the 7.15 firmware has a bug or the Host API specification is faulty 😑

AlCalzone, 2021-09-21:
Bug report opened with Silabs - let's see if they respond:
https://community.silabs.com/s/question/0D58Y000085ett3SAA/zwave-api-setup-supported-sub-commands-encoding-in-sdk-715-does-not-match-host-api-specification?language=en_US

PR #3383 (2021-09-21): fix: work around buggy serial API cmd bitmask encoding in firmware 7.15
fixes: #3367
closed 2021-09-21 by AlCalzone