zwave-js/zwave-js #3368: Discard duplicate segments in segmented ConfigurationCCNameReports

AlCalzone, 2021-09-19:


Ikcelaks, 2021-09-22:
I imagine this bug also lurks in `AssociationCCReport` or any other CCs that deal with segmented reports (except `TransportServiceCCSubsequentSegment`, which merges very differently).

I think the proper solution is to add a new member in `CommandClass` like:
`public addPartialTo(session: CommandClass[]): CommandClass[]`

It could default to the current `session.push(this);` but be overridden in the specific `CC` implementations to drop duplicate partials or do other error checking.

AlCalzone, 2021-09-23:
Sounds good!

Ikcelaks, 2021-09-23:
#3401 should is just a start. It should be expanded to fix the underlying issues with how partial CC sessions are handled.