zwave-js/zwave-js #3384: bug(test): `ConfigManager Tests` for bypassing `ZWAVEJS_EXTERNAL_CONFIG` are broken

Ikcelaks, 2021-09-21:
The tests "from the embedded config dir" and "from the deviceConfigPriorityDir (without ZWAVEJS_EXTERNAL_CONFIG)" are supposed to use the embedded config dir. Instead they create a new external config directory at path "./undefined".

The problem seems to be that [the test setup](https://github.com/zwave-js/node-zwave-js/blame/master/packages/config/src/ConfigManager.test.ts#L108) sets the environment variable to the string "undefined", not `undefined`.

To test this, I modified [`syncExternalConfigDir` in utils.ts](https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/src/utils.ts#L80) to check `extConfigDir == undefined || extConfigDir == "undefined"`. With this change, the tests appeared to run as expected.

AlCalzone, 2021-09-21:
The line you linked needs to be `delete process.env.ZWAVEJS_EXTERNAL_CONFIG`. I think I've fixed that in some branch but not sure where.

Ikcelaks, 2021-09-21:
Yep, I was just about to add that / do a PR.

PR #3385 (2021-09-21): fix(test): correctly delete env variable in ConfigManager Test
fixes #3384
closed 2021-09-21 by AlCalzone