zwave-js/zwave-js #3386: Missing device configuration: Aeotec Illumino Wall Switch

sdwilsh, 2021-09-22:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Aeotec Illumino Wall Switch

### Manufacturer ID

0x371

### Product Type

0x103

### Product ID

0x26

### Firmware Version

1.4

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4273

### Do you have a manual?

https://help.aeotec.com/support/solutions/articles/6000247459-illumino-wall-switch-user-guide

AlCalzone, 2021-09-22:
@zwave-js-bot import config 4273 from zwa

AlCalzone, 2021-09-22:
The bot has created a PR for this file with a starter template. This needs to be cleaned up however to match our [requirements](https://zwave-js.github.io/node-zwave-js/#/config-files/contributing-files).
If you'd like to help with that, you can create a PR onto the target branch import-config-1632291810.

closed 2021-10-10 by AlCalzone