zwave-js/zwave-js #3389: Missing device configuration: Vesternet Z-Wave 2-Wire Capable Switch

imerr, 2021-09-22:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Vesternet Z-Wave 2-Wire Capable Switch

### Manufacturer ID

0x330

### Product Type

0x200

### Product ID

0xd00f

### Firmware Version

1.6

### Do you have a manual?

https://cdn.shopify.com/s/files/1/0066/8149/3559/files/VES-ZW-SWI-002.pdf

richardob50, 2021-12-05:
Is the config for this live now?

AlCalzone, 2021-12-05:
No, no one has stepped up to do it. If you'd like to help out: https://zwave-js.github.io/node-zwave-js/#/config-files/contributing-files

richardob50, 2021-12-05:
Is there a YouTube video or some other tutorial for someone who hasn't done it before? I reckon the vesternet models are made under contract by shenzen sunricher so it should be a case of copying across from another model made by them that a config file is already there for

AlCalzone, 2021-12-05:
Have you read the extensive documentation I've linked you to? We try to make the process as easy as possible. You'll need to know how to create a pull request though, which there are plenty of tutorials for.

richardob50, 2021-12-05:
Ok will look at documentation a little more.

PR #4105 (2022-01-18): feat(config): add Vesternet devices
Added dimmer and switch from Vesternet range.

Fixes #3848 
Fixes #3389

I've also proposed to add dimmer type (leading or trailing edge) to master template, if approved, I can update all the existing instances in other config files.  A quick search shows one or other of these properties in 9x other devices and 3x manufacturer templates (used in a further 6 devices).  Which is admittedly less than I expected when I was writing it
closed 2022-01-18 by AlCalzone