zwave-js/zwave-js #3400: S2 Inclusion DSK Check Timeout

MikeAg03, 2021-09-23:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

What do you observe?

What did you expect to happen?
I expected to be able to add a wall switch using S2 inclusion.  In the prior version of zwave-js before the updates were made for supervisorCC, I was able to add these switches with S2 without an issue.  The issue then was ZWaveJS just wouldn't see updates when manually pressed/toggled without manually pulling them with a refresh.
The latest version seemed to fix that, but I have since reset my network to re-add all switches with S2 and now they don't appear to be adding anymore.
This is with Zooz Z77/Z76 devices.
This behavior seems identical to that reported in issue 3296

### Device information

Manufacturer: Zooz
Model name: Zen76

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Home Assistant zwaveJS2MQTT which uses the following versions:

zwavejs2mqtt: 5.6.0
zwave-js: 8.3.1

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

See description above.  Inclusion worked in the previous/first release with DSK.  In the latest release it seems to be timing out.

### Attach Driver Logfile

Will get log file if requested/needed.

AlCalzone, 2021-09-23:
> Will get log file if requested/needed.

Yes, please.

> In the latest release it seems to be timing out.

This sounds more like a node issue than a driver issue, but I'll wait for the log to make sure.

MikeAg03, 2021-09-23:
2021-09-23T21:50:44.413Z CNTRLR   handling add node request (status = Done)
2021-09-23T21:50:44.414Z CNTRLR   done called for 5
2021-09-23T21:50:44.415Z CNTRLR   finished adding node 5:
                                    basic device class:    Routing Slave
                                    generic device class:  Multilevel Switch
                                    specific device class: Multilevel Power Switch
                                    supported CCs: 
                                    · Basic (0x20)
                                    · Multilevel Switch (0x26)
                                    · All Switch (0x27)
                                    · Z-Wave Plus Info (0x5e)
                                    · Configuration (0x70)
                                    · Central Scene (0x5b)
                                    · Association (0x85)
                                    · Multi Channel Association (0x8e)
                                    · Association Group Information (0x59)
                                    · Transport Service (0x55)
                                    · Version (0x86)
                                    · Manufacturer Specific (0x72)
                                    · Device Reset Locally (0x5a)
                                    · Indicator (0x87)
                                    · Powerlevel (0x73)
                                    · Security 2 (0x9f)
                                    · Supervision (0x6c)
                                    · Firmware Update Meta Data (0x7a)
                                    controlled CCs: 
2021-09-23T21:50:44.416Z CNTRLR » [Node 005] Assigning SUC return route...
2021-09-23T21:50:44.430Z CNTRLR   the inclusion process was stopped
2021-09-23T21:54:54.974Z CNTRLR   [Node 005] Security S2 bootstrapping failed, the node was not granted any S2 s
                                  ecurity class
2021-09-23T21:54:54.975Z CNTRLR   [Node 005] Configuring Z-Wave+ Lifeline association...
2021-09-23T21:54:55.067Z CNTRLR » [Node 005] Assigning return route to node 1...
2021-09-23T21:54:55.102Z CNTRLR   [Node 005] Beginning interview - last completed stage: None
2021-09-23T21:54:55.103Z CNTRLR   [Node 005] new node, doing a full interview...
2021-09-23T21:54:55.103Z CNTRLR » [Node 005] querying protocol info...
2021-09-23T21:54:55.129Z CNTRLR « [Node 005] received response for protocol info:

Perhaps I'll need to try again and turn on a lower level of logging...

MikeAg03, 2021-09-23:
@AlCalzone do you have an email I can send the log to?

MikeAg03, 2021-09-23:
@AlCalzone more detailed log sent you info@zwave-js.io

AlCalzone, 2021-09-24:
Same problem as described here: https://github.com/zwave-js/node-zwave-js/issues/3239#issuecomment-906963206
The node doesn't start requesting the keys, so the process times out. You can try the things I mentioned in the following comments. If anything, this is a connectivity issue between the node and the controller.

closed 2021-09-27 by zwave-js-assistant[bot]