zwave-js/zwave-js #3403: HS-WD200+ Supervision: NoSupport when using S2

nemith, 2021-09-24:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
The supervisor workaround added for HS-WD200+ switches doesn't seem to apply when using S2 security.   

https://github.com/zwave-js/node-zwave-js/blob/1bfd96adf68db669eeb221eba32392e3b3168d7d/packages/config/config/devices/0x000c/hs-wd200.json#L107-L117

What do you observe?

When switching on/off the device from HA The status is reverted after 2 seconds and doesn't reflect the state of the device.  This does not happen if the device is included without S2.

You can see NoSupport in the logs

```
2021-09-24T01:24:11.843Z DRIVER » [Node 004] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 201
└─[Security2CCMessageEncapsulation]
│ sequence number: 153
└─[SupervisionCCGet]
│ session id: 21
│ request updates: true
└─[MultilevelSwitchCCSet]
target value: 0
duration: default
2021-09-24T01:24:11.856Z DRIVER « [RES] [SendData]
was sent: true
2021-09-24T01:24:11.873Z DRIVER « [REQ] [SendData]
callback id: 201
transmit status: OK
2021-09-24T01:24:11.906Z DRIVER « [Node 004] [REQ] [ApplicationCommand]
└─[Security2CCMessageEncapsulation]
│ sequence number: 81
└─[SupervisionCCReport]
session id: 21
more updates follow: false
status: NoSupport
duration: 0s
```

What did you expect to happen?

Home Assistant and the state of the device should match when triggering on/off for the dimmer from the HA frontend.

Steps to reproduce the behavior:

1. Include HS-WD200+ switch with S2 
2. Trigger the device to turn off via zwave
3. See the longs

### Device information

Manufacturer:  HomeSeer
Model name:  HS-WD200+
Node ID in your network: 12

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

version: 8.3.1
`node-zwave-js` branch: 8.3.1
`zwavejs2mqtt` branch: 5.6.0

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Included the devices with S2.

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Worked with no encryption but doesn't work with S2

### Attach Driver Logfile

[zwavejs_2021-09-23.log](https://github.com/zwave-js/node-zwave-js/files/7222176/zwavejs_2021-09-23.log)

AlCalzone, 2021-09-24:
We'll work around it, but I wonder if those switches should even have certification. Supporting Supervision is a hard requirement when using S2.

nemith, 2021-09-24:
Yeah looks like it has been a known issue on the Homeseer forums for awhile and I have messaged their support as well to see if they have an answer.

closed 2021-09-26 by AlCalzone

i8beef, 2021-10-16:
Did this make it into 8.5.0? I have been moving over to zwavejs2mqtt here and I'm hitting this still with the latest which says its running 8.5.0... I tried reinterviewing several times, I tried "replacing" (but not a full exclude and include). Is there something else I would need to do here?

Also, funny that several of the tickets trying to fix this here link to my HomeSeer post from last year trying to get them to own up to doing something wrong here... Still waiting on that response from support... 😄

PR #3408 (2021-09-26): fix: apply removeCC compat flag again after security interview
fixes: #3403