zwave-js/zwave-js #3410: Allegion BE469ZP is reporting the value of the last configuration parameter set every 5 seconds

mattstocum, 2021-09-24:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

Join BE469ZP to the network using S2 Access Control and set any configuration parameter.

What do you observe?

Every 5 seconds the device reports the current value of that parameter

What did you expect to happen?

Device should report the current value of the parameter once

Steps to reproduce the behavior:
  1. Join BE469ZP to the network using S2 Access Control
  2. Set a configuration parameter to any value

### Device information

Manufacturer: Schlage/Allegion
Model name: BE469ZP
Node ID in your network: 64

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 8.3.1

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Moved device from S0 to S2 Access Control

### Did this work before?

Yes (please describe)

### If yes, where did it work?

When included using S0, it worked fine.

### Attach Driver Logfile

[zwavejs_2021-09-24.log](https://github.com/zwave-js/node-zwave-js/files/7226563/zwavejs_2021-09-24.log)

mattstocum, 2021-09-24:
[zwavejs_2021-09-24-update.log](https://github.com/zwave-js/node-zwave-js/files/7226607/zwavejs_2021-09-24-update.log)
Tried changing a config parameter on my back door (node 2), which is still included with S0, and confirmed the lock does not use supervision when reporting the parameter value.

kpine, 2021-09-24:
I have a similar issue with my ZCOMBO smoke alarm. These devices perform a Supervision Get to verify that the notifications were received. There are configuration parameters to tweak the Get timeout and the number of retries, so it doesn't go forever. When I perform a smoke alarm test, it will always report 1 + N (config retry count) times. So with the default of 1, I see 2 notifications, and when I change it to 3, I see 4 total notifications, meaning the alarm is definitely acting as if it doesn't receive the Supervision reports. I notice that the Supervision reports from zwave-js are not S2 encapsulated, are they supposed to be?

[zwavejs_2021-09-24.log](https://github.com/zwave-js/node-zwave-js/files/7227524/zwavejs_2021-09-24.log)

AlCalzone, 2021-09-26:
This is a driver bug. We respond to the `SupervisionCCGet`, but without security encapsulation.

AlCalzone, 2021-09-26:
@mattstocum can you share your cache files `<homeid>.json` and `<homeid>.values.jsonl`? For some reason, the `secure` flag does not propagate properly.

mattstocum, 2021-09-26:
[eb906f1c.json.txt](https://github.com/zwave-js/node-zwave-js/files/7231505/eb906f1c.json.txt)
[eb906f1c.values.jsonl.txt](https://github.com/zwave-js/node-zwave-js/files/7231506/eb906f1c.values.jsonl.txt)
Here you go, I threw a .txt on the end of the filename to get github to allow them to upload. Also, the node ID is now 66. I had removed and re-added the lock with S0 security to get it to stop being so chatty.

AlCalzone, 2021-09-26:
Ok, I know what's happening. The `sendReport` method uses `sendMessage` internally, not `sendCommand`. But only the latter does auto-encapsulation.

AlCalzone, 2021-09-26:
Can you guys test https://github.com/zwave-js/node-zwave-js/pull/3413 (branch fix-supervision-report-security) please? I can make you a test release if necessary, but I don't think that's necessary since you're using Docker.

kpine, 2021-09-26:
Unfortunately, I am unable to build this from source. Is it possible to publish a test release? Then I can at least build the Docker image.

mattstocum, 2021-09-27:
@AlCalzone do you know if the zwavejs2mqtt Dockerfile.contrib build process is working? If so I should be able to do a custom build tomorrow and test.

kpine, 2021-09-27:
The Dockerfile.contrib file worked OK for me for amd64, however it has failed in two tries (in different ways) cross-compling (buildx) for arm64.

AlCalzone, 2021-09-27:
> Is it possible to publish a test release?

Took a few attempts, but there is one now.

mattstocum, 2021-09-27:
> Can you guys test #3413 (branch fix-supervision-report-security) please? I can make you a test release if necessary, but I don't think that's necessary since you're using Docker.

I think that did it.
[node-66.txt](https://github.com/zwave-js/node-zwave-js/files/7237125/node-66.txt)

There was no further traffic from the node after that. This was using a local build of zwavejs2mqtt v5.6.0 with the fix-supervision-report-security branch (8.4.1-0.dev-39155750 is how zwavejs2mqtt is reporting the version number).

PR #3413 (2021-09-27): fix: use sendCommand for Supervision Reports
fixes: #3410
closed 2021-09-27 by AlCalzone