zwave-js/zwave-js #3411: Ecolink DWZWAVE25 is not reporting correct cover tampering value

zappy16, 2021-09-24:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Pairing this device type with ZWaveJS

What do you observe?
That the incorrect "Cover status" is being reported as "Tampering, product cover removed" when cover is on. Removing the cover and putting cover back on does not change this value at all. I have two of these devices paired to ZWaveJS and both do the same behavior. When putting the cover back on it does not send an update that the cover is back on. 

What did you expect to happen?
That the value of "Cover Status" returns to normal value when cover is re-installed onto device. 

Steps to reproduce the behavior:
  1. Pair device
  2. Remove cover on device
  3. Re-install cover on device
  4. ZWaveJS does not update the "Cover status" value

### Device information

Manufacturer: Ecolink
Model name: DWZWAVE25
Node ID in your network: 7 & 15

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 8.3.1
`node-zwave-js` branch: latest
`zwavejs2mqtt` branch: latest

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwavejs_2021-09-24.log](https://github.com/zwave-js/node-zwave-js/files/7227436/zwavejs_2021-09-24.log)

AlCalzone, 2021-09-26:
Can you confirm that your device has these IDs?
> Product Manufacturer: 0x014a
> Product Identifiers: 0x0004:0x0002

https://devices.zwave-js.io/?jumpTo=0x014a:0x0004:0x0002:0.0

We'll have to add a compat flag because the device doesn't idle its notifications automatically.

zappy16, 2021-09-26:
I have confirmed that these devices have those IDs you have mentioned.

AlCalzone, 2021-09-27:
@zwave-js-bot add compat flag to 0x014a:0x0004:0x0002:0.0
```jsonc
forceNotificationIdleReset: true
```

AlCalzone, 2021-09-27:
Thanks, this will be included in the next release

closed 2021-09-27 by AlCalzone

zappy16, 2021-10-02:
@AlCalzone I updated my ZWaveJS2MQTT running version 5.7.2 with ZWaveJS version 8.4.1. I re-interviewed the device after the update and it still is not updating the cover status once the cover is put back on. Reporting "[3] Tampering, product cover removed" but do not see any updates once cover is re-installed. 
[zwavejs_2021-10-01.log](https://github.com/zwave-js/node-zwave-js/files/7270615/zwavejs_2021-10-01.log)

AlCalzone, 2021-10-02:
The delay after which notifications are idled is 5 minutes. I don't see any situation in your log where you waited that long before re-triggering it.

zappy16, 2021-10-02:
Ah! Thank you. I was impatient 
It is working now this morning and reporting idle

PR #3416 (2021-09-27): fix(config): auto-idle notifications for Ecolink DWZWAVE25
fixes: #3411

**TODO:**
- [x]	Change PR title to be more specific