zwave-js/zwave-js #3418: Garage door controller doesn't support controlling barrier state

edgauthier, 2021-09-28:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
I upgraded the host OS that zwavejs2mqtt docker container is running on from Ubuntu 18.04 LTS to 20.04 LTS.

What do you observe?
I can view the state of doors (open/closed) but cannot control the door (open or close it)

What did you expect to happen?
I expected to be able to open/close the door like I was able to before the upgrade.

Steps to reproduce the behavior:
  1. Go to the control panel
  2. Click on the garage door controller node
  3. Scroll down to barrier state
  4. Set target state to open or closed and send the command
  5. Observer no activity with the garage door controller

### Device information

Manufacturer: Nortek Security & Control LLC
Model name: NGD00Z-4
Node ID in your network: 18 and 23 (23 was previously 17, but has been excluded and re-added in an attempt to resolve this issue)

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 5.6.2 / 8.3.1
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Previous version of the container was working on this system prior to the host update. I don't remember the previous version numbers though. 

### Attach Driver Logfile

Logs over the past couple days, including excluding and trying to include the device (node 23) again with S2 security (default) enabled, and the option to Force S0 checked. Node 18 is the previously securely added device that hasn't been touched, but is also not working. 

[secure_logs.zip](https://github.com/zwave-js/node-zwave-js/files/7240254/secure_logs.zip)

edgauthier, 2021-09-28:
Forgot to mention in the original post. When excluding and attempting to include the garage door controller again (with default security options, including Force S0), the device was added with security "None".

AlCalzone, 2021-09-28:
> The "Force Security" option for the Default strategy does not do what you think it does. It's quite confusing really.

In https://zwave-js.github.io/node-zwave-js/#/getting-started/security-s2?id=inclusion-strategy, I called it "Prefer S0 over no encryption".
I'm not sure how to best describe the strategy in one sentence that fits in the inclusion UI, because what it does is:
Use S2 if available. If that fails -> unencrypted
If S2 is not available, S0 is avaliable and the device is one that requires security OR the "prefer S0" option is enabled, use S0. If that fails -> unencrypted
Otherwise: unencrypted.

In this case, @kpine was on the right track. The device supports S2 (but not really), so the driver attempts to include it with S2. That fails.
You'll have to use the **S0 only** option.

edgauthier, 2021-09-28:
Thanks for the help. I re-added using S0 security and now everything is working again. I didn't see the comment to re-interview, so I didn't try that. However, given that door wasn't responding to commands either, I'm guessing the security was lost on that. 

My guess is that my host OS upgrade had nothing to do with the issue, and it had been broken already and I just didn't notice. 

Is it expected that if I upgraded from a device before S2 security was introduced to a version with it, that I would have lost these security settings? I'm still curious as to what triggered the issue (though it's only an idle curiosity at this point now that I've resolved the issue). 

Thanks again!

AlCalzone, 2021-09-29:
> However, given that door wasn't responding to commands either, I'm guessing the security was lost on that.

And the re-interview would have figured that out. I guess that some upgrade broke the cache (or the security key settings)

closed 2021-09-29 by AlCalzone