zwave-js/zwave-js #3423: Inclusion reliably takes 2 attempts for a specific device, and then doesn't finish interview

decairn, 2021-09-29:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
2 bugs within same process, will describe both here but may need to bump one to another issue if they're unrelated.  I'm using zwave2mqtt UI and not sure if source of errors are in node-zwave-js or zwave2mqtt.

Try to add a node to the network, it only completes node addition after 2nd inclusion, and having been added it does not complete interview, and for 30 minutes floods the network with traffic before giving up.

What do you observe?
Part A:
To get a device included and showing up in the zwave2mqtt UI requires 2 separate inclusion attempts.
On the first attempt, device flashes LED indicating it's added but it never registers in UI, and the inclusion process times out.
On second attempt, device flashes LED indicating it's added and then confirms node as added, and shows up in UI.

Part B:
Having been added, the device never gets beyond Node Info stage.  The zwave logs spams about 25x per second with 'Node 52, Endpoint 1: Trying to access endpoint instance before Multi Channel interview' and looks to give up after 30 minutes if I let it run on.

What did you expect to happen?
Inclusion and interview process would result in a recognized new node on the network.

Steps to reproduce the behavior:
Attempt 1
ZWave2mqtt UI - start inclusion without security
On device - press 3 times to do the inclusion
Inclusion does not register and times out.  Have to close UI window manually.

Attempt 2
ZWave2mqtt UI - start inclusion without security
On device - press 3 times to do the inclusion
Inclusion registers returns with new node ID.

Watch UI for interview status - it gets to Node Info and marked as Alive but no further.

### Device information

The device here in the logs is
Manufacturer: Zooz
Model name: ZEN25
Node ID in your network: 52, 53 (same device, different attempts to include on network)

I have 3 of these devices. 2 successfully added prior to this problem, the third refuses to do so.  The 2 that did add OK both suffered from the 2-attempt to add node problem but they completed the interview OK after that.

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

Deployed to HA via frenck's zwavejs2mqtt add-on and configured to use the web socket connection.

zwavejs2mqtt = 5.6.2
node-zwave-js = 8.3.1
HA = 2021.9.7

### Did you change anything?

yes (please describe)

### If yes, what did you change?

The device was previously added / working on a different network without error. I factory reset it and attempted to add to a new network.

### Did this work before?

Yes (please describe)

### If yes, where did it work?

I rebuilt network on a new stick.  Old network was Zooz ZST 10 500, new network uses ZST 10 700 (with flashed firmware to latest, 7.16.4).

Old network allows add of nodes reliably including this one device without error.

### Attach Driver Logfile

[zwavejs_2021-09-28.log](https://github.com/zwave-js/node-zwave-js/files/7248917/zwavejs_2021-09-28.log)

AlCalzone, 2021-09-29:
[image]

Your log is **not** on loglevel debug, let's try this again :)

decairn, 2021-09-29:
Sigh I am so sorry. Meticulously read instructions, and can't see a thing!  This device has me annoyed!

Logs attached with debug. Node 54.  Of course on this time it did manage to complete the interview and it's now ready to use; all I can think of there is the 'Trying to access endpoint instance before Multi Channel interview' traffic spam was causing issue with getting the data back.

On the add node, this still took 2 attempts to add node.  First one timed out and I had to close window, 2nd attempt resulted in the add OK.

These are clean logs from a restart, let network startup finish, then add new node x2.

[zwavejs_2021-09-29.log](https://github.com/zwave-js/node-zwave-js/files/7252514/zwavejs_2021-09-29.log)
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/7252515/zwavejs2mqtt.log)

AlCalzone, 2021-09-29:
Some things that stand out:
1. After almost exactly 30 seconds, something stops the inclusion process. I'm guessing you have a command timeout configured in zwavejs2mqtt?
2. It takes a whopping 40 seconds from receiving the AddingSlave callback until ProtocolDone is received, which normally takes milliseconds. At this point the inclusion process was stopped already, so the callback is ignored and nothing gets included.

AlCalzone, 2021-09-29:
To piggyback on the meter reporting: Z-Wave is a very constrained protocol and not meant for real-time electricity measurement. These can easily bring a whole network to its knees (like here).
You can get pretty close to real-time though if you configure the devices to report changes and throw in a few timed reports (order of magnitude: minutes) for good measure.

decairn, 2021-09-29:
To the add process - the 30s timeout is done by the zwave2mqtt UI, there's no user input to set that.

To the meter reads, the defaults are report every 30s, I had set them to be much higher but a factory reset cleared those.  No idea why those are reporting more frequently, I'll raise with Zooz.

kpine, 2021-09-29:
> To the add process - the 30s timeout is done by the zwave2mqtt UI, there's no user input to set that.

FYI, you can configure the timeout inclusion timeout in the zwavejs2mqtt Z-Wave settings, option `Inclusion/Exclusion timeout`.

closed 2021-09-30 by decairn