zwave-js/zwave-js #3424: [question] Flush shutter DC, Can not set cover position.

labaland, 2021-09-29:
[image]

Hello! i have Flush shutter DC (ZMNHOD)
Lastes HA & Zwavejs2mqtt in docker on synology. Aeotec Z-stick gen5

When i try to set the cover ad a specific % nothing happens. i can only open/close or pause.

I have 2 cover motors and two Flush shutter DC (ZMNHOD). Same on both.

Here is the info from the DEBUG INFO tab:

```
{
  "id": 5,
  "name": "Rullgardin_TV",
  "loc": "Vardagsrum",
  "values": [
    {
      "id": "5-37-0-currentValue",
      "nodeId": 5,
      "commandClass": 37,
      "commandClassName": "Binary Switch",
      "endpoint": 0,
      "property": "currentValue",
      "propertyName": "currentValue",
      "type": "boolean",
      "readable": true,
      "writeable": false,
      "label": "Current value",
      "stateless": false,
      "list": false,
      "value": "unknown",
      "isCurrentValue": true,
      "targetValue": "37-0-targetValue",
      "lastUpdate": 1629586432813,
      "newValue": "unknown"
    },
    {
      "id": "5-37-0-targetValue",
      "nodeId": 5,
      "commandClass": 37,
      "commandClassName": "Binary Switch",
      "endpoint": 0,
      "property": "targetValue",
      "propertyName": "targetValue",
      "type": "boolean",
      "readable": true,
      "writeable": true,
      "label": "Target value",
      "stateless": false,
      "list": false,
      "lastUpdate": 1629586432814
    },
    {
      "id": "5-38-0-targetValue",
      "nodeId": 5,
      "commandClass": 38,
      "commandClassName": "Multilevel Switch",
      "endpoint": 0,
      "property": "targetValue",
      "propertyName": "targetValue",
      "type": "number",
      "readable": true,
      "writeable": true,
      "label": "Target value",
      "stateless": false,
      "min": 0,
      "max": 99,
      "list": false,
      "value": 0,
[... 1016 lines cut ...]
  "isControllerNode": false,
  "isListening": true,
  "isFrequentListening": false,
  "isRouting": true,
  "keepAwake": false,
  "maxDataRate": 100000,
  "deviceClass": {
    "basic": 4,
    "generic": 17,
    "specific": 7
  },
  "deviceId": "345-83-3",
  "status": "Alive",
  "interviewStage": "Complete",
  "statistics": {
    "commandsTX": 24,
    "commandsRX": 94,
    "commandsDroppedRX": 0,
    "commandsDroppedTX": 0,
    "timeoutResponse": 0
  },
  "lastActive": 1629586887784,
  "_name": "Rullgardin_TV (Vardagsrum)"
}
````

AlCalzone, 2021-09-29:
I can see in the logs that something happens:

Command is sent and received:
```
2021-09-28T21:07:38.658Z DRIVER » [Node 005] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      5
                                  └─[MultilevelSwitchCCSet]
                                      target value: 50
...
2021-09-28T21:07:38.685Z DRIVER « [REQ] [SendData]
                                    callback id:     5
                                    transmit status: OK
```

Device uses some electricity:
```
2021-09-28T21:07:41.211Z CNTRLR   [Node 005] [~] [Meter] value[66049]: 0.6 => 0.4                   [Endpoint 0]
2021-09-28T21:07:42.214Z CNTRLR   [Node 005] [~] [Meter] value[66049]: 0.4 => 0.5                   [Endpoint 0]
2021-09-28T21:07:43.218Z CNTRLR   [Node 005] [~] [Meter] value[66049]: 0.5 => 0.4                   [Endpoint 0]
```

But then it seemingly doesn't know where it is
```
2021-09-28T21:07:43.704Z DRIVER » [Node 005] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      6
                                  └─[MultilevelSwitchCCGet]
...
2021-09-28T21:07:43.767Z DRIVER « [Node 005] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCReport]
                                      current value: undefined
```

Maybe it needs to be calibrated or something, but on our end there's nothing wrong.

darkbasic, 2021-09-29:
Did you calibrate your blinds? There is a parameter for that (don't forget to switch it back once done) otherwise it won't be able to compute percentages.

labaland, 2021-09-29:
> Did you calibrate your blinds? There is a parameter for that in the flush shutter DC (don't forget to switch it back once done) otherwise it won't be able to compute percentages.

It was working fine before o switched from "openzwave/ozwdaemon"
When i first tried auto calibrate hey did not go all the way down. But i will try now and test again xD

closed 2021-09-29 by robertsLando