zwave-js/zwave-js #3467: [question] Fibaro Dimmer 2 FGD212 isn't turning the lights on

wilmardo, 2021-10-04:
Sorry if there is a better place for this issue, please let me know if this is the case!

I have a problem with my Fibaro FGD212 since I switched from Zwave2MQTT to ZwaveJS2MQTT. It just doesn't turn the lights on anymore. 
When I switch the light on I see the update in the logs:
```
2021-10-03 17:37:49.760 INFO MQTT: Message received on zwave/nodeID_2/38/0/targetValue/set: '99'
2021-10-03 17:37:49.763 INFO ZWAVE: Writing 99 to 2-38-0-targetValue
2021-10-03 17:37:49.821 INFO ZWAVE: Node 2: value updated: 38-0-currentValue 0 => 99
```
And the currentValue in the webinterface shows indeed 99 after I switch the light on. But the bulbs themselves are still off. The same with dimming, I can sent all the values I want from HA but the bulbs itself are still off.

What I tried so far:

* Re-interview of the dimmer
* Force calibration with the B-Button on the dimmer (this does turn the bulbs on and goes through the sequence but makes no difference for the problem)
* Updated to the latest 5.7.1 with Z-Wave JS 8.4.0 (which includes https://github.com/zwave-js/node-zwave-js/pull/2854 which I hoped would improve the situation)

Does anyone have some more suggestions where to look or something to try?

wilmardo, 2021-10-04:
Thanks for the prompt response!

This is the relevant excerpt from the logs:
```
2021-10-04 14:40:44.973 INFO MQTT: Message received on zwave/nodeID_2/38/0/targetValue/set: '99'
2021-10-04 14:40:44.979 INFO ZWAVE: Writing 99 to 2-38-0-targetValue
2021-10-04 14:40:45.104 DEBUG MQTT: Publishing to zwave/nodeID_2/38/0/currentValue: { time: 1633358445103, value: 99 } with options { qos: 1, retain: true }
2021-10-04 14:40:45.106 INFO ZWAVE: Node 2: value updated: 38-0-currentValue 0 => 99
2021-10-04 14:40:50.648 INFO MQTT: Message received on zwave/nodeID_2/38/0/targetValue/set: 'OFF'
2021-10-04 14:40:50.652 INFO ZWAVE: Writing 0 to 2-38-0-targetValue
2021-10-04 14:41:00.270 DEBUG MQTT: Publishing to zwave/nodeID_2/38/0/currentValue: { time: 1633358460269, value: 0 } with options { qos: 1, retain: true }
2021-10-04 14:41:00.272 INFO ZWAVE: Node 2: value updated: 38-0-currentValue 99 => 0
```

<details>
<summary>Whole log</summary>
<p>

```
2021-10-04 14:40:18.832 INFO APP: Version: 5.7.2
2021-10-04 14:40:18.857 INFO APP: Application path:/zwavejs2mqtt

 ______                       _     ___                  _   _
|___  /                      (_)   |__ \                | | | |
   / /_      ____ ___   _____ _ ___   ) |_ __ ___   __ _| |_| |_
  / /\ \ /\ / / _` \ \ / / _ \ / __| / /| '_ ` _ \ / _` | __| __|
 / /__\ V  V / (_| |\ V /  __/ \__ \/ /_| | | | | | (_| | |_| |_
/_____|\_/\_/ \__,_| \_/ \___| |___/____|_| |_| |_|\__, |\__|\__|
                            _/ |                      | |
                           |__/                       |_|

2021-10-04 14:40:19.012 WARN STORE: /data/scenes.json not found
2021-10-04 14:40:19.160 INFO APP: Listening on port 8091 host 0.0.0.0 protocol HTTP
2021-10-04 14:40:19.361 INFO MQTT: Connecting to mqtt://mosquitto:1883
2021-10-04 14:40:19.614 INFO ZWAVE: Connecting to tcp://192.168.1.23:23
2021-10-04 14:40:19.791 INFO MQTT: MQTT client connected
2021-10-04 14:40:19.813 DEBUG MQTT: Subscribing to homeassistant/status with options { qos: 1 }
2021-10-04 14:40:19.824 DEBUG MQTT: Subscribing to zwave/_CLIENTS/ZWAVE_GATEWAY-zwavejs2mqtt/broadcast/# with options { qos: 1 }
2021-10-04 14:40:19.829 DEBUG MQTT: Subscribing to zwave/_CLIENTS/ZWAVE_GATEWAY-zwavejs2mqtt/api/# with options { qos: 1 }
2021-10-04 14:40:19.835 DEBUG MQTT: Subscribing to zwave/_CLIENTS/ZWAVE_GATEWAY-zwavejs2mqtt/multicast/# with options { qos: 1 }
2021-10-04 14:40:19.849 INFO MQTT: Subscribed to homeassistant/status
2021-10-04 14:40:19.855 INFO MQTT: Subscribed to zwave/_CLIENTS/ZWAVE_GATEWAY-zwavejs2mqtt/broadcast/#
2021-10-04 14:40:19.857 INFO MQTT: Subscribed to zwave/_CLIENTS/ZWAVE_GATEWAY-zwavejs2mqtt/api/#
2021-10-04 14:40:19.860 INFO MQTT: Subscribed to zwave/_CLIENTS/ZWAVE_GATEWAY-zwavejs2mqtt/multicast/#
2021-10-04 14:40:29.963 INFO ZWAVE: Zwave driver is ready
2021-10-04 14:40:29.966 INFO ZWAVE: Controller status: Driver ready
2021-10-04 14:40:30.104 DEBUG ZWAVE: Binding to node 1 events
2021-10-04 14:40:30.116 DEBUG MQTT: Publishing to zwave/nodeID_1/status: { time: 1633358430112, value: true, status: 'Unknown', nodeId: 1 } with options { qos: 1, retain: true }
2021-10-04 14:40:30.118 DEBUG ZWAVE: Node 1 has been added to nodes array
2021-10-04 14:40:30.124 DEBUG ZWAVE: Binding to node 2 events
2021-10-04 14:40:30.129 DEBUG MQTT: Publishing to zwave/nodeID_2/status: { time: 1633358430127, value: true, status: 'Unknown', nodeId: 2 } with options { qos: 1, retain: true }
2021-10-04 14:40:30.131 DEBUG ZWAVE: Node 2 has been added to nodes array
2021-10-04 14:40:30.137 DEBUG ZWAVE: Binding to node 3 events
2021-10-04 14:40:30.141 DEBUG MQTT: Publishing to zwave/nodeID_3/status: { time: 1633358430140, value: true, status: 'Unknown', nodeId: 3 } with options { qos: 1, retain: true }
2021-10-04 14:40:30.144 DEBUG ZWAVE: Node 3 has been added to nodes array
2021-10-04 14:40:30.148 DEBUG ZWAVE: Binding to node 4 events
2021-10-04 14:40:30.152 DEBUG MQTT: Publishing to zwave/nodeID_4/status: { time: 1633358430150, value: true, status: 'Unknown', nodeId: 4 } with options { qos: 1, retain: true }
2021-10-04 14:40:30.155 DEBUG ZWAVE: Node 4 has been added to nodes array
2021-10-04 14:40:30.161 DEBUG ZWAVE: Binding to node 5 events
2021-10-04 14:40:30.165 DEBUG MQTT: Publishing to zwave/nodeID_5/status: { time: 1633358430164, value: true, status: 'Unknown', nodeId: 5 } with options { qos: 1, retain: true }
2021-10-04 14:40:30.168 DEBUG ZWAVE: Node 5 has been added to nodes array
2021-10-04 14:40:30.172 DEBUG ZWAVE: Binding to node 6 events
2021-10-04 14:40:30.176 DEBUG MQTT: Publishing to zwave/nodeID_6/status: { time: 1633358430175, value: true, status: 'Unknown', nodeId: 6 } with options { qos: 1, retain: true }
2021-10-04 14:40:30.179 DEBUG ZWAVE: Node 6 has been added to nodes array
2021-10-04 14:40:30.181 INFO ZWAVE: Scanning network with homeid: 0xeaab709e
2021-10-04 14:40:30.192 DEBUG MQTT: Publishing to zwave/nodeID_1/status: { time: 1633358430191, value: true, status: 'Alive', nodeId: 1 } with options { qos: 1, retain: true }
2021-10-04 14:40:30.194 INFO ZWAVE: Node 1 is alive
2021-10-04 14:40:30.214 DEBUG MQTT: Publishing to zwave/nodeID_1/status: { time: 1633358430213, value: true, status: 'Alive', nodeId: 1 } with options { qos: 1, retain: true }
2021-10-04 14:40:30.216 INFO ZWAVE: Node 1 ready: Z-Wave.me - RaZberry Controller 2016 ZWave+ (Z-Wave.Me Razberry daughtercard)
[... cut ...]
2021-10-04 14:42:24.033 DEBUG MQTT: Publishing to zwave/nodeID_6/50/0/value/66561: { time: 1633358544032, value: 231.69 } with options { qos: 1, retain: true }
2021-10-04 14:42:24.035 INFO ZWAVE: Node 6: value updated: 50-0-value-66561 231.51 => 231.69
2021-10-04 14:42:25.032 DEBUG MQTT: Publishing to zwave/nodeID_6/50/0/value/66817: { time: 1633358545031, value: 0 } with options { qos: 1, retain: true }
2021-10-04 14:42:25.034 INFO ZWAVE: Node 6: value updated: 50-0-value-66817 0 => 0
```

</p>
</details>  

Is this what you are looking for?

robertsLando, 2021-10-04:
Nope, I need driver logs: https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs

wilmardo, 2021-10-04:
Doh. Everything is working as it should now :bulb: 

Massive thanks for the pointer to the driver logs. That revealed:

```
2021-10-04T15:28:11.765Z DRIVER   Dropping message with invalid payload
2021-10-04T15:28:11.768Z DRIVER « [Node 002] [REQ] [ApplicationCommand]
                                  └─[SecurityCCCommandEncapsulation] [INVALID]
                                      error: Invalid auth code, won't accept security encapsulated command.
```

So then it clicked, this is my only secure connected device and I forgot to move my key over. It never occurred to me since the device is connected (and can be re-interviewed etc) and the INFO logs show nothing out of the ordinary.

It might be useful to propagate this error as an ERROR in the logs since imo this is a real error.

I feel so dumb now haha.
Thanks for your time, for being the little nudge in the correct direction and the rubber duck I needed :duck:

closed 2021-10-04 by wilmardo

AlCalzone, 2021-10-05:
> It never occurred to me since the device is connected (and can be re-interviewed

That is normal, but the interview won't query anything of interest