zwave-js/zwave-js #3468: Missing device configuration: Eurotronics Temperature & Humidity Sensor Z-Wave Plus

amrheing, 2021-10-04:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Eurotronics Temperature & Humidity Sensor Z-Wave Plus

### Manufacturer ID

do not know

### Product Type

0x0004

### Product ID

0x0001

### Firmware Version

0.1

### Do you have a manual?

[TFS-Z-Wave-Plus_BDA_web_EN.pdf](https://github.com/zwave-js/node-zwave-js/files/7278288/TFS-Z-Wave-Plus_BDA_web_EN.pdf)

Debug Info zwavejs2mqtt controller:

`{
  "id": 52,
  "name": "",
  "loc": "",
  "values": [
    {
      "id": "52-49-0-Humidity",
      "nodeId": 52,
      "commandClass": 49,
      "commandClassName": "Multilevel Sensor",
      "endpoint": 0,
      "property": "Humidity",
      "propertyName": "Humidity",
      "type": "number",
      "readable": true,
      "writeable": false,
      "label": "Humidity",
      "ccSpecific": {
        "sensorType": 5,
        "scale": 0
      },
      "stateless": false,
      "unit": "%",
      "list": false,
      "value": 50,
      "lastUpdate": 1633352725582,
      "newValue": 50
    },
    {
      "id": "52-49-0-Air temperature",
      "nodeId": 52,
      "commandClass": 49,
      "commandClassName": "Multilevel Sensor",
      "endpoint": 0,
      "property": "Air temperature",
      "propertyName": "Air temperature",
      "type": "number",
      "readable": true,
[... 296 lines cut ...]
      "endpoint": 0,
      "value": 1,
      "maxNodes": 5,
      "isLifeline": true,
      "multiChannel": false
    },
    {
      "text": "Temperature Report",
      "endpoint": 0,
      "value": 2,
      "maxNodes": 5,
      "isLifeline": false,
      "multiChannel": false
    }
  ],
  "neighbors": [],
  "ready": true,
  "available": true,
  "hassDevices": {},
  "failed": false,
  "inited": true,
  "hexId": "0x0148-0x0004-0x0001",
  "dbLink": "https://devices.zwave-js.io/?jumpTo=0x0148:0x0004:0x0001:0.1",
  "manufacturerId": 328,
  "productId": 1,`

AlCalzone, 2021-12-06:
@zwave-js-bot config help wanted

AlexCPU, 2022-01-21:
@AlCalzone this was fixed by #4099 and can be closed.

closed 2022-01-21 by AlCalzone