zwave-js/zwave-js #3470: FGBS001 (Fibaro Universal Sensor)

darkbasic, 2021-10-04:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Adding the missing lifeline association fixes the binary reports, but not the level reports.

What do you observe?
FGBS001 does not create a lifeline association and thus does not report updates whenever the binary status changes.
Manually creating the lifeline association fixes the binary reports, but not the level reports. I tried to create additional associations for Input 1 and Input 2 but they didn't help as well.

What did you expect to happen?
Get working reports for both the binary and level out of the box.

Additional info:
I noticed that zwave2mqtt (the archived openzwave one) didn't expose the "dimmer" level (while zwavejs2mqtt does, under the Basic tab) thus I can't say for sure that this used to work w/ Openzwave:
[image]

But what I've noticed is that I **did** get a level in openzwave logs:
```
2021-10-04 17:17:25.285 Detail, Node002,   Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x02, 0x07, 0x60, 0x0d, 0x01, 0x01, 0x20, 0x01, 0xff, 0x40
2021-10-04 17:17:25.285 Detail, 
2021-10-04 17:17:25.285 Info, Node002, Received a MultiChannelEncap from node 2, endpoint 1 for Command Class COMMAND_CLASS_BASIC
2021-10-04 17:17:25.285 Info, Node002, Received Basic set from node 2: level=255. Treating it as a Basic report.
2021-10-04 17:17:25.285 Detail, Node002, Queuing (Send) MultiChannel Encapsulated (instance=1): SensorBinaryCmd_Get (Node=2): 0x01, 0x0d, 0x00, 0x13, 0x02, 0x06, 0x60, 0x0d, 0x01, 0x01, 0x30, 0x02, 0x25, 0x10, 0x8f
2021-10-04 17:17:25.285 Detail, 
2021-10-04 17:17:25.286 Info, Node002, Sending (Send) message (Callback ID=0x10, Expected Reply=0x04) - MultiChannel Encapsulated (instance=1): SensorBinaryCmd_Get (Node=2): 0x01, 0x0d, 0x00, 0x13, 0x02, 0x06, 0x60, 0x0d, 0x01, 0x01, 0x30, 0x02, 0x25, 0x10, 0x8f
2021-10-04 17:17:25.286 Info, Node002, Encrypted Flag is 0
2021-10-04 17:17:25.294 Detail, Node002,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2021-10-04 17:17:25.295 Detail, Node002,   ZW_SEND_DATA delivered to Z-Wave stack
2021-10-04 17:17:25.313 Detail, Node002,   Received: 0x01, 0x07, 0x00, 0x13, 0x10, 0x00, 0x00, 0x03, 0xf8
2021-10-04 17:17:25.314 Detail, Node002,   ZW_SEND_DATA Request with callback ID 0x10 received (expected 0x10)
2021-10-04 17:17:25.314 Info, Node002, Request RTT 27 Average Request RTT 78
2021-10-04 17:17:25.314 Detail, Node002,   Expected callbackId was received
2021-10-04 17:17:25.325 Detail, Node002,   Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x02, 0x07, 0x60, 0x0d, 0x01, 0x01, 0x30, 0x03, 0xff, 0x52
2021-10-04 17:17:25.325 Detail, 
2021-10-04 17:17:25.325 Info, Node002, Response RTT 39 Average Response RTT 78
2021-10-04 17:17:25.325 Info, Node002, Received a MultiChannelEncap from node 2, endpoint 1 for Command Class COMMAND_CLASS_SENSOR_BINARY
2021-10-04 17:17:25.325 Info, Node002, Received SensorBinary report: State=On
2021-10-04 17:17:25.325 Detail, Node002, Value Updated: old value=false, new value=true, type=bool
2021-10-04 17:17:25.326 Detail, Node002, Changes to this value are not verified
2021-10-04 17:17:25.326 Detail, Node002,   Expected reply and command class was received
2021-10-04 17:17:25.326 Detail, Node002,   Message transaction complete
2021-10-04 17:17:25.326 Detail, 
2021-10-04 17:17:25.326 Detail, Node002, Removing current message
2021-10-04 17:17:25.326 Detail, Node002, Notification: ValueChanged CC: COMMAND_CLASS_SENSOR_BINARY Instance: 1 Index: 0
 z2m:Zwave zwave node 2: changed: 48-1-0:Instance 1: Sensor:false -> true +10s
```

```
2021-10-04 17:17:52.137 Detail, Node002,   Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x02, 0x07, 0x60, 0x0d, 0x01, 0x01, 0x20, 0x01, 0x00, 0xbf
2021-10-04 17:17:52.137 Detail, 
2021-10-04 17:17:52.137 Info, Node002, Received a MultiChannelEncap from node 2, endpoint 1 for Command Class COMMAND_CLASS_BASIC
2021-10-04 17:17:52.138 Info, Node002, Received Basic set from node 2: level=0. Treating it as a Basic report.
2021-10-04 17:17:52.138 Detail, Node002, Queuing (Send) MultiChannel Encapsulated (instance=1): SensorBinaryCmd_Get (Node=2): 0x01, 0x0d, 0x00, 0x13, 0x02, 0x06, 0x60, 0x0d, 0x01, 0x01, 0x30, 0x02, 0x25, 0x11, 0x8e
2021-10-04 17:17:52.138 Detail, 
2021-10-04 17:17:52.138 Info, Node002, Sending (Send) message (Callback ID=0x11, Expected Reply=0x04) - MultiChannel Encapsulated (instance=1): SensorBinaryCmd_Get (Node=2): 0x01, 0x0d, 0x00, 0x13, 0x02, 0x06, 0x60, 0x0d, 0x01, 0x01, 0x30, 0x02, 0x25, 0x11, 0x8e
2021-10-04 17:17:52.138 Info, Node002, Encrypted Flag is 0
[... 564 lines cut ...]
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: main tag 8.4.1
`zwavejs2mqtt` branch: main commit f97288bb9feba478137c049bffcff5463d1a3b34

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_2021-10-04.log](https://github.com/zwave-js/node-zwave-js/files/7279460/zwavejs_2021-10-04.log)

AlCalzone, 2021-11-19:
@darkbasic ping :)

darkbasic, 2021-11-19:
@AlCalzone sorry I completely missed your reply, but I will be able to test this next week

closed 2021-11-30 by zwave-js-assistant[bot]