zwave-js/zwave-js #3523: [bug] Qubino currentValue is overwritten by an intermediate value when dimming the device

vanackej, 2021-10-14:
**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [ ] Docker
- [ ] PKG
- [x] Snap package
- [ ] Manually built (git clone - yarn install - yarn run build )

zwavejs2mqtt: 5.4.5.0ad7537
zwave-js: 8.0.8

### Describe the bug

The bug occurs with Qubino ZMNHDD Flush Dimmer Plus device.

When dimming the light, either using the physical switch or a MQTT command, the current value is first updated to the final target value (99 => 0 if I switch the light off). But then, as this device has a progressive dimming effect, an intermiediate value is reported (anything between the original and the target value), and the final value is never reported again. If I force current value refresh using the GUI, the final value is correctly reported.

### To Reproduce

Steps to reproduce the behavior:

1. Just switch the device from 99 to 0
2. 0 is first reported for currentValue
3. Then any value between 99 and 0 is reported 
4. Force value refresh in the GUI returns the right currentValue (0)

### Expected behavior

The final current value should not be dropped or overriden by an intermediate value.

### Additional context

In the attached logs, I noticed a strange 'Scheduled poll canceled because value was updated' log. I think it can be related to the issue : as the device report an intermediate value, the final value polling is never done ?

[zwavejs2mqtt-store.zip](https://github.com/zwave-js/zwavejs2mqtt/files/7025664/zwavejs2mqtt-store.zip)

DaveCo1701, 2021-10-29:
Something similar seems to be happening with the new GE/Jasco 3 Speed Fan controller (55258 / ZW4002). Some people on the HA forum have been [chatting about it](https://community.home-assistant.io/t/ge-enbrighten-fan-controller-incorrectly-reporting-on-off-state/334036). A solution to this problem might also be helpful beyond the Qubino devices. Log attached for reference. I've also reached out to Jasco to see if they have anything to say about it; will update here if I hear anything.

Don't mean to hijack, if preferred I can open a separate issue for the GE/Jasco device.

[gefanlog.txt](https://github.com/zwave-js/node-zwave-js/files/7445280/gefanlog.txt)

darkbasic, 2021-10-30:
By the way I can confirm that my Qubino Flush Dimmers w/ firmware 50.2 don't suffer from this problem, while my Qubino Flush Shutter DC fw 7.3 still suffer from this problem.

PR #4569 (2022-05-05): fix: cancel verification poll only when expected value is received
With this PR, we no longer blindly cancel value verification polls when an unsolicited report is received. Instead, for each scheduled poll, an expected value can/will be saved and value updates only cancel the poll when the value was changed to this expected value (assuming one is set).

fixes: #3523
fixes: #4453
closed 2022-05-05 by AlCalzone