zwave-js/zwave-js #3543: [question] HomeSeer WD200+ report state inconsistent

i8beef, 2021-10-17:
Greetings! I have most everything working now with one issue... I have three HomeSeer WD200+ dimmers that are no reporting state consistently.

They SEEM to work correctly immediately after inclusion / interview (changing states reports state change back immediately). Then as soon as I operate the switch manually, things change. Past that point, any manual manipulation sends state changes perfectly fine. But any commands sent to change state (via targetValue) no longer receive an update on currentValue..

I've had issues with HomeSeer being... selective... of how they interpret spec before, but any ideas here on work arounds?

i8beef, 2021-10-17:
LOL, its a shame really, they are nice switches, but yeah, HomeSeer hasn't been very interested in acknowledging these do things weird... I suspect it has something to do with "admitting we made a mistake that requires a firmware update" and "charging for a tool to do firmware updates" being in conflict 😄 

I have had open issues with them, as have others, for over a year with no response unfortunately... I gather a lot of people who choose HomeSeer switches choose HomeSeer software which has a work around for this oddity.

AlCalzone, 2021-10-18:
Are you able to test a custom Docker image with the changes? https://zwave-js.github.io/zwavejs2mqtt/#/development/custom-docker
Use `master` for zwavejs2mqtt, `supervision-stuff` for `node-zwave-js`.

i8beef, 2021-10-18:
Sure, I've just never done this... I assume I just run that Docker.contrib script with a name like 'zwavejstest123', stop my running instance, and then do I need to copy my volume so as not to mess up the current one, or do I just repoint it to the same volume?

AlCalzone, 2021-10-18:
@blhoward2 @robertsLando can you help?

i8beef, 2021-10-19:
Ok, took a little doing to get it to build a node docker image on a Pi's limited resources, but I got it running and it seems to work with this... I reinterviewed and tried my same tests and it seems to give me status updates with S2 security with this. Do you need me to capture more logs or anything? If you need to cleanup that PR and have me test again or anything let me know, I'm happy to do it.

AlCalzone, 2021-10-19:
I'd be interested in seeing a log where it works - just to be sure I didn't break something else.

i8beef, 2021-10-21:
Just FYI, HomeSeer support has "escalated to the developers" along with a reminded that the HS-WD200 switches have been discontinued and I should consider upgrading to the new 300 series switches. Thats probably a light indication that they aren't interested in fixing their firmware for these unfortunately...

closed 2021-10-21 by AlCalzone