zwave-js/zwave-js #3544: [feat] Enable Priority Route functionality

Malkierian, 2021-10-18:
**Is your feature request related to a problem? Please describe.**
Heal Network doesn't make the nodes take the shortest route they actually can, in fact I've tried several and they have set very inefficient routes (my controller is positioned such that nothing more than 2 hops should be necessary with my current setup except for one node, yet it refuses to update node connections to reflect this).

**Describe the solution you'd like**
The ability to set Application Priority Routes to effectively minimize the number of hops.

**Describe alternatives you've considered**
Currently there are no alternatives

**Additional context**
https://i.imgur.com/mKcKaZi.png
This is my current network map. However, from north to south, the order of node placement is actually
Front Door Light/Mailbox Door -> R Bedroom Light -> Kitchen Extender -> Scene Controller -> Controller -> Garden Extender -> Quail Light Dimmer. This was after I moved the controller from my room to a central location and healed the network twice. Garden Extender should be connected directly to the Controller and the Mailbox Door should be connected to the Front Door Light. This network map doesn't make any sense.

AlCalzone, 2021-10-18:
First things first: I'm not against implementing that functionality.

> This network map doesn't make any sense.

Just to be clear: This map **only** indicates neighbors, not which routes will be taken.

I'm not yet sure why but I've seen tons of examples where neighbors are also missing from that map, so don't put too much weight into it. Maybe the routes haven't settled yet - there's a ton of magic happening we don't have influence over. Unless you have actual connectivity issues and/or have verified the incorrect routes with a Zniffer, I'd avoid premature optimization.

I do have a theory you could help me test: Which stick are you using? Which firmware (should be visible in the driver log on startup, "serial API version")?

AlCalzone, 2021-10-18:
https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs - make sure to set the loglevel to debug

AlCalzone, 2021-10-18:
Yep, 7.15 seems correct. Did you include all devices on that version or did you do any firmware updates meanwhile?

Silabs seem to have fixed something very recently in 7.16, which could explain weird neighbor information, but I'm still trying to get a bigger picture.

Malkierian, 2021-10-18:
Yeah, the stick was brand new like 2 weeks ago, so everything has been added since then.

AlCalzone, 2021-10-18:
Ok that could support my theory:
Firmware 7.16 fixes a bug where inclusion is done with a too high TX power. Normally the power is reduced, so nodes at the brink of connectivity don't include on direct range if they can't communicate reliably and are instead forced to mesh. Now if they were barely included but later don't see their neighbors, this could explain the weird neighbor situation you're having.

That doesn't help you though 😅

PR #5410 (2023-02-06): feat: add support for managing priority (return) routes
This PR adds support for managing priority routes to nodes, between nodes and to the controller. These can be used in scenarios where the automatic Z-Wave routing resolution produces subpar results.

fixes: #3544
closed 2023-02-06 by AlCalzone