zwave-js/zwave-js #3621: Missing device configuration: Minoston Outdoor Dimmer Plug

brisketfingers, 2021-10-29:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Minoston Outdoor Dimmer Plug

### Manufacturer ID

0x0312

### Product Type

0xff07

### Product ID

0xff03

### Firmware Version

1.0

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4254?selectedFrequencyId=2

### Do you have a manual?

https://products.z-wavealliance.org/products/4254/pics

AlCalzone, 2021-10-30:
@zwave-js-bot import config 4254 from zwa

AlCalzone, 2021-10-30:
The bot has created a starter template, but that usually needs to be cleaned up and aligned to our style guide.
If you'd like to help, you can create a PR on the branch in the bot's PR.

AlCalzone, 2021-10-30:
Ok never mind. We do have a file, it just needs another fingerprint added.

PR #3855 (2021-12-30): feat(config): add many new devices from Z-Wave Alliance DB
Closes #3841
Closes #3834
Closes #3802
Closes #3804
Closes #3621
Closes #3609
closed 2021-12-30 by AlCalzone