zwave-js/zwave-js #3624: S2 bootstrapping times out when PIN entry takes "too long"

kpine, 2021-10-31:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

When including with S2, if the PIN is not entered within a ~10 second time interval, the S2 bootstrapping will fail after a long timeout. Anecdotally, reports from HA users and elsewhere seem to indicate that this failure is common across a variety of devices. The solution is usually "enter the PIN as fast as possible". I think the same problem occurs if you take to long to select the security classes, but I didn't investigate that one much.

Out of curiosity, I repeated the inclusion with the same device using the Silicon Labs PC-Controller software and had quite a different experience. Even after waiting 2 minutes to input the PIN, it successfully includes the device with S2. I've repeated this successfully multiple times waiting 1+ minutes.

Steps to reproduce the behavior:
  1. Include a device with Default inclusion strategy
  2. Select the default security class grant
  3. Wait for PIN prompt
  4. After PIN prompt is shown for 1 or more minutes, enter the correct pin
  5. Watch as zwave-js times out and fails the bootstrapping

Repeat the above with PC Controller, and it succeeds.

### Device information

I used this device for my testing, but it seems to be a problem for many devices and many users.

Manufacturer: HomeSeer Technologies
Model name: HS-FC200+

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 8.6.1
`zwavejs2mqtt` branch: 5.10.1

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

Attached are zniffer traces for the inclusion process. I'll email the S2 keys.

For PC Controller, included is a zniffer trace and the pc-controller trace file.
For Z-Wave JS, a zniffer trace and driver debug logs.

[trace.zip](https://github.com/zwave-js/node-zwave-js/files/7449631/trace.zip)

One thing I noticed in the pc controller trace is that after the PIN is prompted, it seems to periodically "ping" the node every ~10 seconds with an S2 encapsulated message. Z-Wave JS does not do this, it's just silent until the bootstrapping times out and fails.

[image]

kpine, 2021-10-31:
I sent the keys via email, however none of them let me decrypt these messages. I assume these are using the temporary key.

kpine, 2021-10-31:
Despite my total luck of understanding of this topic, at the risk of looking silly I'm going to suggest that zwave-js is not properly replying to the Public Key Report from the node. In the PC Controller trace, the controller sends the Report milliseconds after receiving the report from the node. In Z-Wave JS, the report is not sent until the PIN is entered. The anecdotally observed ~8-12 seconds time allowance seems to follow a 10-second timeout specification.

[image]

According to the diagram, timeout TB3 starts after the node sends its public key report.

[image]

Meanwhile, in zwave-js, no Report until 1 minute later, about the time I entered the PIN. I'm assuming that since the node doesn't get the report within 8-12 seconds, it cancels its bootstrapping.

[image]

AlCalzone, 2021-11-01:
I believe you have just solved the mystery of failed S2 inclusion.

PR #3637 (2021-11-01): fix: send S2 public key immediately, ignore KexSet until DSK verified
With this PR, the controller's public key is sent to the included node immediately instead of waiting for DSK verification so the node doesn't abort the bootstrapping silently.
Until the DSK is verified, the repeated KexSet from the node are now ignored.

fixes: #3624
closed 2021-11-01 by AlCalzone