zwave-js/zwave-js #3641: [bug] Interview not completed for Fibaro motion sensor FGMS001 (missing alarm sensor menu)

davidwaze, 2021-11-02:
**Checklist:**

- [X] I am **not** using Home Assistant. **Or:** a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog and my problem is **not** mentioned there.

Build/Run method

- [ ] Docker
- [X] PKG
- [ ] Snap package
- [ ] Manually built (git clone - yarn install - yarn run build )

### Version

zwavejs2mqtt : 5.10.1
zwavejs : 8.6.1

### Describe the bug

Hello,

I had to reinclude one of my fibaro motion sensor but i'm facing problem one menu is missing : 

Working one i have vs new interviewed one (ive also tried to factory reset the device) : 

Binary Sensor => OK
Multilevel Sensor => OK
Configuration => OK
Notification => OK
Manufacturer Specific => OK
Battery => OK
Wake Up => OK
Version => OK
Alarm Sensor => not existing (i have another menu which is called "Basic")

[image]
[image]
[image]

### To Reproduce

New interview

Please find log (Node is 201, unsecured)
[zwavejs_2021-11-01.log](https://github.com/zwave-js/zwavejs2mqtt/files/7452441/zwavejs_2021-11-01.log)

Thanks

AlCalzone, 2021-11-02:
The device supports Notification CC, which also reports events for the Tamper Alarm. Therefore Alarm Sensor CC is hidden in favor of the Notifications.

davidwaze, 2021-11-02:
Hello,

OK sorry for the disturbance, as all my others fibaro motion sensor were working like this, i didn't noticed that this information was located also in another class.

Working as expected ;-)

Thanks

closed 2021-11-02 by davidwaze