zwave-js/zwave-js #3646: Missing device configuration: Haozee 5-in-1 Multisensor

Botched1, 2021-11-03:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Haozee 5-in-1 Multisensor

### Manufacturer ID

0x0258

### Product Type

0x0020

### Product ID

0x0718

### Firmware Version

2.23

### Is the device listed on the Z-Wave Alliance website?

No

### Do you have a manual?

This is exactly the exact same device as:

https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0258/nas-pd07z.json

https://devices.zwave-js.io/?jumpTo=0x0258:0x0010:0x0720:0.0

It is just a rebranded/relabelled version, and needs the type/id added so the correct driver is used.

AlCalzone, 2021-11-03:
The parameters in the manual you linked don't match the config file you linked.

Botched1, 2021-11-03:
> The parameters in the manual you linked don't match the config file you linked.

I just went through them - and you are right. Curious. Maybe there is a firmware difference between the two models after all, I was assuming they were the same just re-branded as they are physically identical...

The parameters in the linked manual definitely work for the Haozee model, as I wrote the driver for that device for Hubitat and fully tested all parameters.

AlCalzone, 2021-11-03:
Then we need a new config file for it. If you like, you can help with that to speed things up, see https://zwave-js.github.io/node-zwave-js/#/config-files/contributing-files

Botched1, 2021-11-03:
Will do! That is certainly within my capability. Thanks!

Botched1, 2021-11-03:
> Then we need a new config file for it. If you like, you can help with that to speed things up, see https://zwave-js.github.io/node-zwave-js/#/config-files/contributing-files

Tried to do a PR for this. Hopefully did it correctly (I'm a novice).

https://github.com/zwave-js/node-zwave-js/pull/3655

PR #3655 (2021-11-03): feat(config): add NAS-PD07ZU1
Add new device configuration file for NAS-PD07ZU1

fixes: #3646
closed 2021-11-03 by AlCalzone