zwave-js/zwave-js #3649: [bug] No configuration options on Haozee / Neocool cam Multisensor (but it exists on other devices)

Botched1, 2021-11-03:
### Version

**Checklist:**

- [x ] I am **not** using Home Assistant. **Or:** a developer has told me to come here.
- [x ] I have checked the troubleshooting section and my problem is **not** described there.
- [x ] I have read the changelog and my problem is **not** mentioned there.

Build/Run method

- [x ] Docker

Zwavejs2Mqtt version: 5.10.1
Z-Wave JS version: 8.6.1

### Describe the bug
I have a Haozee (aka Neo Cool Cam) 5-in-1 Zwave 700 Multisensor. It works as expected, but there is no configuration section un der the device details to change configuration settings.

### To Reproduce
Paired device successfully with no security.

Open zwavejs2mqtt interface, and expand the device details, and there is no configuration section at all.

### Expected behavior
There should be a configuration section to change device configuration.

### Additional context
Screenshot:
[image]

[image]

AlCalzone, 2021-11-03:
Duplicate of #3646

AlCalzone, 2021-11-03:
Let's keep the discussion in the other issue.

closed 2021-11-03 by AlCalzone