zwave-js/zwave-js #3657: First SIGINT does not cause process to exit

bchabrier, 2021-11-03:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

`SIGINT` is caught in `zwave-js` by a handler that destroys the driver, but does not call `process.exit`.

What do you observe?

The first ^C does not kill the process. Looking at the traces, it does the cleanup on the driver, but does not exit. You need to press ^C a second time to have the process really killed.

This can be an issue when the process is a subprocess (e.g. of a shell command). Indeed, the first ^C will kill the parent process, and the zwave process will continue as a zombie.

What did you expect to happen?

The handler should also call `process.exit()`, so that the process exits on the first ^C.

Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 8.4.1
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

16:52:37.665 DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                      ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                        ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                       ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                      ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                      ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
2021-11-03T15:52:37.657Z DRIVER
16:52:37.702 DRIVER   version 8.4.1
2021-11-03T15:52:37.702Z DRIVER   version 8.4.1
16:52:37.711 DRIVER
2021-11-03T15:52:37.711Z DRIVER
16:52:37.718 DRIVER   starting driver...
2021-11-03T15:52:37.718Z DRIVER   starting driver...
16:52:37.847 DRIVER   opening serial port /dev/ttyACM0
2021-11-03T15:52:37.847Z DRIVER   opening serial port /dev/ttyACM0
16:52:43.431 DRIVER   serial port opened
2021-11-03T15:52:43.431Z DRIVER   serial port opened
16:52:43.444 SERIAL » [NAK]                                                                   (0x15)
2021-11-03T15:52:43.443Z SERIAL » [NAK] (0x15)
16:52:45.170 DRIVER   loading configuration...
2021-11-03T15:52:45.169Z DRIVER   loading configuration...
16:52:45.263 CONFIG   version 8.4.1
2021-11-03T15:52:45.262Z CONFIG   version 8.4.1
=================  ^C typed here ====================================
16:52:52.200 DRIVER   destroying driver instance...
2021-11-03T15:52:52.199Z DRIVER   destroying driver instance...
================== process it no exited ================================

AlCalzone, 2021-11-03:
I'll need some steps to reproduce. If I run the driver using `yarn ts test/run.ts` in the repository, the first CTRL+C does destroy the driver and exit the process:
[image]

bchabrier, 2021-11-03:
Here is short program to reproduce:
```
pi@raspberrypi ~/domoja/modules/openzwave $ cat test-zwave.ts
import { Driver } from "zwave-js";

const driver = new Driver('/dev/ttyACM0', {
    logConfig: {
        enabled: true,
        level: "silly",
    }
});
driver.on("error", (err) => {console.error(err)});
driver.start().catch(reason => {console.log(reason)});

console.log('Starting loop...');
setInterval(() => {console.log('looping')}, 1000);
```
When you run it, the first ^C destroys the driver, the second ^C is needed to stop the program.

```
pi@raspberrypi ~/domoja/modules/openzwave $ ts-node test-zwave.ts
21:15:35.553 DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                      ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                        ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                       ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                      ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                      ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
21:15:35.583 DRIVER   version 8.4.1
21:15:35.587 DRIVER
21:15:35.589 DRIVER   starting driver...
21:15:35.686 DRIVER   opening serial port /dev/ttyACM0
Starting loop...
21:15:35.767 DRIVER   serial port opened
21:15:35.774 SERIAL » [NAK]                                                                   (0x15)
looping
21:15:37.296 DRIVER   loading configuration...
21:15:37.314 CONFIG   version 8.4.1
looping
looping
...
...
looping
looping
looping
looping
^C21:16:19.779 DRIVER   destroying driver instance...
looping
looping
looping
looping
looping
looping
^C
pi@raspberrypi ~/domoja/modules/openzwave $ 
```
Note that if the driver is not started correctly or has error, then the handler on `SIGINT` is not active, hence the first ^C will indeed kill the process.

Hope it helps!

PR #3760 (2021-11-21): fix: don't handle SIGINT in the driver, let applications take care of it
Handling SIGINT in the driver just causes problems and possibly unintended behavior. By letting the application take care of shutting down, we can avoid this.

fixes: #3657
closed 2021-11-21 by AlCalzone