zwave-js/zwave-js #3662: The command class Application Status is not implemented (ZW0303) - Remotec ZXT-600

geirsf, 2021-11-04:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Any setpoint-command

What do you observe?
- DRIVER Dropping message because it could not be deserialized: The command class Application Status is not implemented (ZW0303), and then 
- CNTRLR [Node 019] Timed out while waiting for a response from the node (ZW0201)

What did you expect to happen?
The setpoint value to be updated. 

Steps to reproduce the behavior:
  Any set point command to Remotec ZXT-600

### Device information

Manufacturer: Remotec
Model name: ZXT-600
Node ID in your network: 19

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:8.6.1
`zwavejs2mqtt` branch: 5.10.1

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

zwavejs2mqtt : 5.9.0
zwave-js 8.5.0

### Attach Driver Logfile

[zwavejs2mqtt_2021-11-04 (1).log](https://github.com/zwave-js/node-zwave-js/files/7476825/zwavejs2mqtt_2021-11-04.1.log)

geirsf, 2021-11-04:
Thanx for you feedback.  Is it possible, since the ZXT-600 it's an IR device, that the shorter refreshing time conflicts with  The RF Protection State?

From the Z-Wave Application Command Class Specification (SDS13781 V15)

The RF Protection State field used to set the RF protection state of the device. In the case where a
device set into a RF Protection State which instructs the device not to answer to a “normal operation”
Command, the device MUST return the Application Rejected Request Command (Status = 0) of the
Application Status Command Class. Refer to the Application Status Command Class.

AlCalzone, 2021-11-04:
Protection CC is something different. Also, the device sends a "busy, wait 2 seconds" command, not request rejected.

AlCalzone, 2021-11-04:
I'll just revert the verification delay for thermostats to 5 seconds, that shouldn't hurt.

geirsf, 2021-11-04:
Ah, great, hopefully it will do the trick.

PR #3665 (2021-11-05): fix(cc): increase delay before verifying thermostat values after set
fixes: #3662
closed 2021-11-05 by AlCalzone