zwave-js/zwave-js #3666: No unsolicited status updates from Qubino Flush 1 Relays ZMNHAA

Mr-Hell, 2021-11-05:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I’m not receiving any unsolicited updates from my Qubino Flush 1 Relays (ZMNHAA), no energy reports or status updates when manually operating a switch.

(Similar behaviour as in https://github.com/zwave-js/node-zwave-js/issues/2012)

### Device information

Manufacturer: Qubino 
Model name: Flush 1 Relay (ZMNHAA), Z-Wave chip firmware versions 1.12, Z-Wave protocol version 3.28
Node ID in your network: <3, 4, 7, 9, 10, 22, 24, 28, 29, 30, 41, 48, 49, 51, 58, 62, 67, 71>

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

Home Assistant Community Add-on: Z-Wave JS to MQTT ver. 0.28.0
(zwavejs2mqtt ver. 5.10.1, Z-wave JS ver. 8.6.1).

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Migrated from Fibaro Home Center 2 (where all Qubino devices worked just fine) to running Home Assistant on a NUC with an AEON Labs Gen5 USB Z‐Stick Controller.

### Attach Driver Logfile

[zwavejs_2021-11-05.log](https://github.com/zwave-js/node-zwave-js/files/7486404/zwavejs_2021-11-05.log)

AlCalzone, 2021-11-05:
Ok at least Node 3 seems to be contradicting itself:
```
2021-11-05T13:26:34.186Z CNTRLR « [Node 003] node info received
                                  supported CCs:
[...]
                                  · Association
[...]
```
and later:
```
2021-11-05T13:26:37.717Z DRIVER « [Node 003] [REQ] [ApplicationCommand]
                                  └─[VersionCCCommandClassReport]
                                      CC:      Association
                                      version: 0
2021-11-05T13:26:37.718Z CNTRLR   [Node 003]   does NOT support CC Association (0x85)
```

This means you're not getting a lifeline association automatically set up and therefore no automatic reporting. We can override that with a compat flag in the config file.

AlCalzone, 2021-11-05:
@Mr-Hell please try this config file: https://raw.githubusercontent.com/zwave-js/node-zwave-js/90b9e676c2a66bd76de381c9143386c106089d61/packages/config/config/devices/0x0159/zmnhaa.json

https://zwave-js.github.io/zwavejs2mqtt/#/device-support/custom-device-files describes how to test a custom device file.

After re-interviewing the lifeline association should be created automatically.

Mr-Hell, 2021-11-05:
> @Mr-Hell please try this config file: https://raw.githubusercontent.com/zwave-js/node-zwave-js/90b9e676c2a66bd76de381c9143386c106089d61/packages/config/config/devices/0x0159/zmnhaa.json [...]

I tried adding the custom config file, but received this error: 

2021-11-05T20:03:43.198Z DRIVER   loading configuration...
2021-11-05T20:03:43.204Z CONFIG   Using external configuration dir /data/db
2021-11-05T20:03:43.205Z CONFIG   version 8.6.1
2021-11-05T20:03:43.834Z CONFIG   Error parsing config file zmnhaa.json: Could not parse config file /data/store
                                  /config/templates/master_template.json: ENOENT: no such file or directory, ope
                                  n '/data/store/config/templates/master_template.json'
                                  Import stack: 
                                    in #state_after_power_failure_prev_off
                                    in /data/store/config/zmnhaa.json (ZW0340)
2021-11-05T20:03:43.871Z DRIVER   beginning interview...

AlCalzone, 2021-11-06:
Ahh damn.... Take [this file](https://raw.githubusercontent.com/zwave-js/node-zwave-js/master/packages/config/config/devices/templates/master_template.json) and put it in a subdirectory `templates` in the same location as the test config.

Mr-Hell, 2021-11-06:
Thanks, this worked a bit better, but the lifeline association was not created automatically when I re-interviewed a node.
However, it's now possible to manually add associations. Previously, there were no groups available to choose from when trying to add an association.
After I manually added the lifeline association, the binary sensor and energy meter seems to work correctly!

AlCalzone, 2021-11-06:
Ok can you share a log of the re-interview?

Mr-Hell, 2021-11-06:
Yep, I re-interview nodes 28 & 3 + manually added the lifeline association, see attached log.
[zwavejs_2021-11-06.log](https://github.com/zwave-js/node-zwave-js/files/7491329/zwavejs_2021-11-06.log)

I started doing the same to the rest of my ZMNHAA nodes, but now they seem to flood the network with energy reports...

AlCalzone, 2021-11-09:
Ok I think the auto-assignment didn't work because of a bug I fixed in 8.7.3
https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md#873-2021-11-03

About the energy reports, that's usually an issue of device configuration (report intervals and thresholds that are set too low).

closed 2021-11-09 by AlCalzone

Mr-Hell, 2021-11-09:
Ok, thanks. I will check this when I update ZwaveJS.

I've already adjusted the reporting thresholds for the energy reports. The values fluctuated so much that I had to increase the threshold substantially to avoid excessive reporting.

Many thanks for the quick resolution of my problem, you're doing a fantastic job! Keep up the good work!