zwave-js/zwave-js #3667: interview for steinel lamps rarely reaches the status complete

HubertusH, 2021-11-05:
### Version

**Checklist:**

- [X] I am **not** using Home Assistant. **Or:** a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog and my problem is **not** mentioned there.

Build/Run method

- [ ] Docker
- [ ] PKG
- [X] Snap package
- [ ] Manually built (git clone - yarn install - yarn run build )

Zwavejs2Mqtt version: 5.10.0.3538a6e
Z-Wave JS version: 8.6.0

### Describe the bug

hello all, 

I have three lamps with motion sensors from the company Steinel (XLED Home 2).
During the interview the command class: "Scene_Actuator_Configuration" is queried and a list of 255 parameters with two values each is generated.

[image]
...
[image]

According to the documentation of the luminaire (attached), however, only a few values are mentioned.

Can you check if the command-class reading is correct?
Thanl you und BR
Hubertus

### Additional context
[10322_Z-WAVE_module_Specification_05_2017.pdf](https://github.com/zwave-js/zwavejs2mqtt/files/7486409/10322_Z-WAVE_module_Specification_05_2017.pdf)

robertsLando, 2021-11-05:
Please attach zwavejs logs

AlCalzone, 2021-11-05:
> During the interview the command class: "Scene_Actuator_Configuration" is queried and a list of 255 parameters with two values each is generated.

This is intended. The Scene Actuator Configuration CC lets you configure the values that are sent for each possible scene (1-255). These values aren't actually queried, but just created for you to set.

I'm not sure what this has to do with the issue title though.

HubertusH, 2021-11-08:
@AlCalzone : sorry - i thought maybe this had something to do with the fact that the interview takes so long or is often not completed.
@robertsLando i attach the logfiles , the interesting node is 118
[zwavejs_2021-11-08.zip](https://github.com/zwave-js/node-zwave-js/files/7494926/zwavejs_2021-11-08.zip)
[zwavejs2mqtt_2021-11-08.zip](https://github.com/zwave-js/node-zwave-js/files/7494927/zwavejs2mqtt_2021-11-08.zip)

AlCalzone, 2021-11-08:
Your network is very busy while the interview is being done, e.g. Node 72 reports several measurements a couple of times. Node 82 and 90 have their setpoint changed but do not respond, delaying the interview each time. Node 11, 24 and 54 get polled a bunch of times.

At some point I don't see any more outgoing messages to node 118, but my suspicion is that the message queue for "normal" operation is never empty, so the interview never gets its turn. Try to disable all polling you have set up and retry the interview then.

HubertusH, 2021-11-08:
Thank you for pointing out the node 72. Is a "home energy meter" which of course does not need to send its readings so often. Otherwise I have only ONE polling at the room thermostats to query the set temperature every 5 minutes.

The nodes 82, 90 are heating thermostats with a wakeup of 120 sec - so they can not answer in between. But I have eight more of these devices.

For the nodes 11,24 and 54 I have to invest more time - maybe this will be cleared now when the network traffic decreases in general.

BR
Hubertus

AlCalzone, 2021-11-08:
> The nodes 82, 90 are heating thermostats with a wakeup of 120 sec - so they can not answer in between. But I have eight more of these devices.

I think they may either be excerbating the problem or be affected by it too. They are these weird thermostats (Danfoss LC-13 or one of its OEM variants?) that require to be polled when they wake up or end up with error E5 on the display. Most of these queries expire before they can be sent because the nodes are asleep again.

Node 54 is definitely getting polled:
```
2021-11-08T07:08:52.076Z CNTRLR » [Node 054] querying Binary Switch state...
(queue is busy for another 9 seconds)
2021-11-08T07:09:01.415Z DRIVER » [Node 054] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      60
                                  └─[BinarySwitchCCGet]
...
2021-11-08T07:09:01.578Z CNTRLR » [Node 054] querying current sensor reading...
2021-11-08T07:09:01.751Z DRIVER » [Node 054] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      62
                                  └─[MultilevelSensorCCGet]
...
2021-11-08T07:09:02.021Z CNTRLR » [Node 054] querying meter value (type = Electric, scale = kWh)...
2021-11-08T07:09:02.232Z DRIVER » [Node 054] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      64
                                  └─[MeterCCGet]
                                      scale: kWh
...
2021-11-08T07:09:12.699Z CNTRLR   [Node 054] Timed out while waiting for a response from the node (ZW0201)
2021-11-08T07:09:12.699Z CNTRLR » [Node 054] querying meter value (type = Electric, scale = W)...
2021-11-08T07:09:13.312Z DRIVER » [Node 054] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      68
                                  └─[MeterCCGet]
                                      scale: W
...
2021-11-08T07:10:50.133Z CNTRLR » [Node 054] querying Binary Switch state...
...etc
```
this goes on every minute or so.

HubertusH, 2021-11-09:
> I think they may either be excerbating the problem or be affected by it too. They are these weird thermostats (Danfoss LC-13 or one of its OEM variants?) that require to be polled when they wake up [...]

Yes you are right - these are the thermostats from Danfoss LC-13 or Develo. Now I know where the error message "E5" comes from - thanks for that - must restart my thermostats more often (battery out / clean) so that they work again.

With the Node 54 I am sure that no polling - at least I have not set up - exists. However, a MultilevelSensorCCGet is queried here, which also exists in the configuration, but which the device does not have. Is a WallPlug from Fibaro. I have performed a new interview here and now the configuration is correct again. Possibly now some peace returns in the network.

Really thank you that you have shown me what I have to pay more attention. 

**Now I can try the interview of the Steinel lamps again - will report.**

Best regards
Hubertus

AlCalzone, 2021-11-09:
> However, a MultilevelSensorCCGet is queried here, which also exists in the configuration, but which the device does not have

I think you're confusing some things here. Configuration has nothing to do with the supported CCs. The node does respond to the queries:
```
2021-11-08T07:09:02.016Z DRIVER « [Node 054] [REQ] [ApplicationCommand]
                                  └─[MultilevelSensorCCReport]
                                      type:  Power
                                      scale: Watt
                                      value: 0.5
```
And since zwave-js doesn't do automatic queries outside the interview (and those for LC13), they must come from an application somewhere. Are you sure you don't have setup [polling](https://zwave-js.github.io/zwavejs2mqtt/#/device-support/polling) in zwavejs2mqtt?

> Could the massive network traffic also be the reason why the interview gives nonsensical values for some devices? 

no

> Properties like temperature that the device does not support at all

no

>  - or switches that only have a currentValue but no targetValue?

maybe, if the interview timed out here but the switch reported on its own later.

HubertusH, 2021-11-09:
The one&only polling i setup:
[image]

Back to node 054 - i have 16 of this devices in the network. the most of them are ok - but some devices report after the interview CCs they did'n exists:

[image]
[image]
that it is confusing. Most time this parameter disappear after a reInterview.

AlCalzone, 2021-11-09:
This is a known issue that cannot be fixed for some devices. Starting in v8.5.0, these are filtered out for devices where it is possible to determine which sensors they expose (https://github.com/zwave-js/node-zwave-js/pull/3507), but for others it simply isn't possible without hardcoding the capabilities.

HubertusH, 2021-11-09:
ok - good to know!

the interview of the Steinel lamps are running well!
so wie can close this issue - and by the way you helped me with the performance problems. Thanks

closed 2021-11-09 by AlCalzone

HubertusH, 2021-11-09:
sorry that is already embarrassing :-( two of the lamps are ok - the third produces thousands (75.000 in one hour) of messages: 

2021-11-09T14:20:30.275Z CNTRLR [Node 094] treating BasicCC::Set as a report [Endpoint 2]
2021-11-09 15:20:30.277 INFO ZWAVE: Node 94: value updated: 32-2-currentValue 0 => 0

HubertusH, 2021-11-09:
fantastic - now she's quiet. So many things to consider. Thank you for your patience