zwave-js/zwave-js #3668: Update Aeotec Aerq files (again)

blhoward2, 2021-11-05:
A new fw has been release for the ZWA009 further fixing delay issues. We need to change our fw number cutoffs for the 1.11 file. 

https://help.aeotec.com/support/solutions/articles/6000253486-update-zwa009-aerq-sensor-v12-01

AlCalzone, 2021-11-05:
What's your preference? Expand the range to 11.0 - 255.255 or add a new variant with changed defaults? I'd prefer the former.

blhoward2, 2021-11-05:
My only concern is that they’re likely done updating the ZW009, and they will keep adding to the ZW039 and who knows if they will stick within the 2.0 range. I can ask their plans?

blhoward2, 2021-11-06:
From Aeotec:

ZWA009 are V1.00 to 1.01 and 11.00 to later. (Anything listed as 11 or later are custom versions).

ZWA039 is V2.00 - V9.00 not really sure but this would be the best range i would give here. Realistically, i believe the range will be between V2.00 to V4.00, the main version number will only increase if there is some huge change such as SDK version used.

PR #3673 (2021-11-08): fix(config): change firmware range for Aeotec aërQ v1
Closes #3668
closed 2021-11-08 by AlCalzone