zwave-js/zwave-js #3680: Unable to add ztrm2fx to network.

aizerin, 2021-11-08:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I'm migrating from Fibarho HCLite2 to Z-Wave JS with Aeotec Z-stick 7. I have problems with HeatIt Ztrm2fx thermostats. I tried to add them to new network and it has weird behavior. I observe multiple scenarios:

- Thermostat is not added to network with error shown on thermostat - This is most common.
- Thermostat shows error but it appears in control center but is not able to retrieve node info.
- Thermostat shows error and then shows included. After some time it appears in control center with node info and it works as expected.

Weird is that I'm not able to replicate options 2 and 3 anymore. So I can attach only logs for option 1.

```
2021-11-08T13:02:55.704Z CNTRLR   Starting inclusion process with strategy Insecure...
2021-11-08T13:02:55.712Z SERIAL » 0x0105004ac10a7b                                                     (7 bytes)
2021-11-08T13:02:55.713Z DRIVER » [REQ] [AddNodeToNetwork]
                                    payload: 0xc10a
2021-11-08T13:02:55.719Z SERIAL « [ACK]                                                                   (0x06)
2021-11-08T13:02:55.723Z SERIAL « 0x0107004a0a010000b9                                                 (9 bytes)
2021-11-08T13:02:55.724Z SERIAL » [ACK]                                                                   (0x06)
2021-11-08T13:02:55.726Z DRIVER « [REQ] [AddNodeToNetwork]
                                    payload: 0x0a010000
2021-11-08T13:02:55.728Z DRIVER   handling request AddNodeToNetwork (74)
2021-11-08T13:02:55.728Z DRIVER     1 handler registered!
2021-11-08T13:02:55.729Z DRIVER     invoking handler #0
2021-11-08T13:02:55.729Z CNTRLR   handling add node request (status = Ready)
2021-11-08T13:02:55.729Z CNTRLR     the controller is now ready to add nodes
2021-11-08T13:02:55.731Z DRIVER       the message was handled
2021-11-08T13:03:02.601Z SERIAL « 0x0107004a0a020000ba                                                 (9 bytes)
2021-11-08T13:03:02.607Z SERIAL » [ACK]                                                                   (0x06)
2021-11-08T13:03:02.613Z DRIVER « [REQ] [AddNodeToNetwork]
                                    payload: 0x0a020000
2021-11-08T13:03:02.615Z DRIVER   handling request AddNodeToNetwork (74)
2021-11-08T13:03:02.616Z DRIVER     1 handler registered!
2021-11-08T13:03:02.616Z DRIVER     invoking handler #0
2021-11-08T13:03:02.616Z CNTRLR   handling add node request (status = NodeFound)
2021-11-08T13:03:02.709Z SERIAL « 0x0107004a0a070000bf                                                 (9 bytes)
2021-11-08T13:03:02.710Z SERIAL » [ACK]                                                                   (0x06)
2021-11-08T13:03:02.712Z DRIVER « [REQ] [AddNodeToNetwork]
                                    payload: 0x0a070000
2021-11-08T13:03:02.714Z DRIVER   handling request AddNodeToNetwork (74)
2021-11-08T13:03:02.715Z DRIVER     1 handler registered!
2021-11-08T13:03:02.715Z DRIVER     invoking handler #0
2021-11-08T13:03:02.715Z CNTRLR   handling add node request (status = Failed)
2021-11-08T13:03:02.717Z CNTRLR     starting the inclusion failed
2021-11-08T13:03:02.718Z CNTRLR   stopping inclusion process...
2021-11-08T13:03:02.731Z SERIAL » 0x0105004ac50b7e                                                     (7 bytes)
2021-11-08T13:03:02.731Z DRIVER » [REQ] [AddNodeToNetwork]
                                    payload: 0xc50b
2021-11-08T13:03:02.743Z SERIAL « [ACK]                                                                   (0x06)
2021-11-08T13:03:02.747Z SERIAL « 0x0107004a0b060000bf                                                 (9 bytes)
2021-11-08T13:03:02.749Z SERIAL » [ACK]                                                                   (0x06)
2021-11-08T13:03:02.749Z DRIVER       the message was handled
2021-11-08T13:03:02.751Z DRIVER « [REQ] [AddNodeToNetwork]
                                    payload: 0x0b060000
2021-11-08T13:03:02.753Z DRIVER   handling request AddNodeToNetwork (74)
2021-11-08T13:03:02.754Z DRIVER     1 handler registered!
2021-11-08T13:03:02.754Z DRIVER     invoking handler #0
2021-11-08T13:03:02.754Z CNTRLR   handling add node request (status = Done)
2021-11-08T13:03:02.755Z CNTRLR   done called for 0
2021-11-08T13:03:02.755Z CNTRLR   the inclusion process was stopped
```

### Device information

Thermostat:
Manufacturer: Heat It
Model name: ZTRM2FX
Node ID in your network: NA

Controller:
Manufacturer: Aeotec
Model name: Z-Stick 7 (ZWA010-C)
Z-wave SDK: 7.16.3

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Currently, I'm using Z-Wave JS to MQTT 0.28.0 addon.

version:
zwavejs2mqtt: 5.10.1
zwave-js: 8.6.1
home id: 4210540239
home hex: 0xfaf7becf

I also tried 0.1.46 Z-wave JS plugin in HA, which has zwave-js on 8.7.5.

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

It is working with Fibaro Home Center lite2

### Attach Driver Logfile

[zwavejs_2021-11-08.log](https://github.com/zwave-js/node-zwave-js/files/7497343/zwavejs_2021-11-08.log)

AlCalzone, 2021-11-08:
> handling add node request (status = Failed)

Seems that either the device or the stick is in a weird state. Try the following:
- stop inclusion, start exclusion, stop exclusion, start inclusion again
- factory reset the device before attempting the inclusion
- restart host and/or pull the stick and put it back

aizerin, 2021-11-08:
I tried it without success. I also factory reset the stick. 
Currently, I try to use PC Controller 5 from Simplicity Studio, and inclusion is done correctly with this software. I can see everything ing PC Controller about the node. But when I put stick back to zwave-js then the node stuck on gathering protocol info. There arent much information in logs. I found only this related to node

```
2021-11-08T19:17:50.066Z CNTRLR   [Node 019] Beginning interview - last completed stage: ProtocolInfo
```

And there are some new messages about that driver is dropping something.

```
2021-11-08T19:17:46.742Z DRIVER   starting driver...
2021-11-08T19:17:46.752Z DRIVER   opening serial port /dev/ttyUSB0
2021-11-08T19:17:46.776Z DRIVER   serial port opened
2021-11-08T19:17:46.777Z SERIAL » [NAK]                                                                   (0x15)
2021-11-08T19:17:46.786Z DRIVER   loading configuration...
2021-11-08T19:17:46.787Z SERIAL « 0x011300a800010a0a600d02003105012200c600bb40                        (21 bytes)
2021-11-08T19:17:46.791Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-11-08T19:17:46.791Z SERIAL » [ACK]                                                                   (0x06)
2021-11-08T19:17:46.797Z SERIAL « 0x010f00a814010e063105012200bc00d13f                                (17 bytes)
2021-11-08T19:17:46.799Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-11-08T19:17:46.800Z SERIAL » [ACK]                                                                   (0x06)
2021-11-08T19:17:46.807Z CONFIG   Using external configuration dir /data/db
2021-11-08T19:17:46.807Z CONFIG   version 8.6.2-20211027
2021-11-08T19:17:47.496Z SERIAL « 0x011300a800010a0a600d02003105012200c600bb40                        (21 bytes)
2021-11-08T19:17:47.498Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-11-08T19:17:47.498Z SERIAL » [ACK]                                                                   (0x06)
2021-11-08T19:17:47.504Z SERIAL « 0x011300a800010a0a600d02003105012200c600b64d                        (21 bytes)
2021-11-08T19:17:47.506Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-11-08T19:17:47.506Z SERIAL » [ACK]                                                                   (0x06)
2021-11-08T19:17:47.512Z SERIAL « 0x011300a800010a0a600d02003105012200c600bb40                        (21 bytes)
2021-11-08T19:17:47.513Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-11-08T19:17:47.513Z SERIAL » [ACK]                                                                   (0x06)
2021-11-08T19:17:47.521Z SERIAL « 0x010f00a814010e063105012200bc00d33d                                (17 bytes)
2021-11-08T19:17:47.522Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-11-08T19:17:47.522Z SERIAL » [ACK]                                                                   (0x06)
2021-11-08T19:17:47.529Z SERIAL « 0x010f00a814010e063105012200bc00d03e                                (17 bytes)
2021-11-08T19:17:47.530Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-11-08T19:17:47.530Z SERIAL » [ACK]                                                                   (0x06)
2021-11-08T19:17:47.536Z SERIAL « 0x011300a800010a0a600d02003105012200c600bb40                        (21 bytes)
2021-11-08T19:17:47.537Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-11-08T19:17:47.538Z SERIAL » [ACK]                                                                   (0x06)
2021-11-08T19:17:47.544Z SERIAL « 0x011300a800010a0a600d02003105012200c600b843                        (21 bytes)
2021-11-08T19:17:47.545Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-11-08T19:17:47.546Z SERIAL » [ACK]                                                                   (0x06)
2021-11-08T19:17:47.556Z SERIAL « 0x011300a800010a0a600d02003105012200c600bb40                        (21 bytes)
2021-11-08T19:17:47.558Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-11-08T19:17:47.558Z SERIAL » [ACK]                                                                   (0x06)
2021-11-08T19:17:47.566Z SERIAL « 0x011300a800010a0a600d02003105012200c600a55e                        (21 bytes)
2021-11-08T19:17:47.568Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-11-08T19:17:47.568Z SERIAL » [ACK]                                                                   (0x06)
2021-11-08T19:17:47.575Z SERIAL « 0x011300a800010a0a600d03003105012200b400a52d                        (21 bytes)
2021-11-08T19:17:47.577Z DRIVER   Dropping message because the driver is not ready to handle it yet.
2021-11-08T19:17:47.577Z SERIAL » [ACK]                                                                   (0x06)
2021-11-08T19:17:47.584Z SERIAL « 0x011300a800010a0a600d03003105012200b400a52d                        (21 bytes)
2021-11-08T19:17:47.586Z DRIVER   Dropping message because the driver is not ready to handle it yet.
```

AlCalzone, 2021-11-08:
Please post a complete log file of a startup and re-interview

aizerin, 2021-11-08:
Here are the log files:

[zwavejs_2021-11-08 (3).log](https://github.com/zwave-js/node-zwave-js/files/7500234/zwavejs_2021-11-08.3.log)

It is node 19.

Debug info from node:

```json
{
  "id": 19,
  "name": "",
  "loc": "",
  "values": [
    {
      "id": "19-114-0-manufacturerId",
      "nodeId": 19,
      "commandClass": 114,
      "commandClassName": "Manufacturer Specific",
      "endpoint": 0,
      "property": "manufacturerId",
      "propertyName": "manufacturerId",
      "type": "number",
      "readable": true,
      "writeable": false,
      "label": "Manufacturer ID",
      "stateless": false,
      "min": 0,
      "max": 65535,
      "list": false
    },
    {
      "id": "19-114-0-productType",
      "nodeId": 19,
      "commandClass": 114,
      "commandClassName": "Manufacturer Specific",
      "endpoint": 0,
      "property": "productType",
      "propertyName": "productType",
      "type": "number",
      "readable": true,
      "writeable": false,
      "label": "Product type",
      "stateless": false,
      "min": 0,
      "max": 65535,
      "list": false
    },
    {
      "id": "19-114-0-productId",
      "nodeId": 19,
      "commandClass": 114,
      "commandClassName": "Manufacturer Specific",
      "endpoint": 0,
      "property": "productId",
      "propertyName": "productId",
      "type": "number",
      "readable": true,
      "writeable": false,
      "label": "Product ID",
      "stateless": false,
      "min": 0,
      "max": 65535,
      "list": false
    },
    {
      "id": "19-64-0-mode",
      "nodeId": 19,
      "commandClass": 64,
      "commandClassName": "Thermostat Mode",
[... 213 lines cut ...]
  "isListening": true,
  "isFrequentListening": false,
  "isRouting": true,
  "keepAwake": false,
  "maxDataRate": 100000,
  "deviceClass": {
    "basic": 4,
    "generic": 8,
    "specific": 6
  },
  "deviceId": "undefined-undefined-undefined",
  "status": "Alive",
  "interviewStage": "Complete",
  "statistics": {
    "commandsTX": 2,
    "commandsRX": 1,
    "commandsDroppedRX": 29,
    "commandsDroppedTX": 0,
    "timeoutResponse": 12
  },
  "lastActive": 1636406706095,
  "isBatteryPowered": false,
  "_name": "NodeID_19"
}
```

blhoward2, 2021-11-09:
Is your stick plugged directly into the machine, or in a usb extension cable. If it's not, it should be on an extension cable to minimize interference. 

PC controller isn't doing a full interview when you include so it's much less network traffic.

AlCalzone, 2021-11-09:
@aizerin The node seems to be included securely, but the driver attempts to interview it insecurely because it thinks it was granted no security class. Not sure how that happened.

Are you able to edit the cache files (`<homeid>.json`) by hand? If so, stop zwave-js and open the file in an editor.
There should be a section like this in the file:
```jsonc
// ...
    "nodes": {
        "19": {
            "id": 19,
            // delete from here...
            "securityClasses": {
                "S2_AccessControl": false,
                "S2_Authenticated": false,
                "S2_Unauthenticated": false,
                "S0_Legacy": false
            },
            // to here...
```
Delete the highlighted part and save, then start zwave-js again.

If not, you'll have to re-include the node again. Please make a log this time so we see where it goes wrong.

> I tried it without success. I also factory reset the stick.

_"It doesn't work and I did something else too"_ doesn't really help when troubleshooting.

aizerin, 2021-11-10:
It seems that there is some kind of problem with the controller and these particular devices. I tried to use extension cable as @blhoward2 suggested. And if I put the stick in close distance to the device then it will include very slowly but inclusion is ok. And then it is working normally but still slowly. And this is happening only with the ztrm2fx, other devices are working normally.

I found some old razzberry gen5 module at home, so I tried to include these thermostats and it works fast and without any problems. So I think that the problem is not related to z-wave js anymore. Ztrm2fx are discontinued devices anyway, they already have some design problems and I'm not much happy owner :). 

So I think that we can close it. Thank you guys very much for your time and for work you have done with the zwave-js. I can finally migrate from super slow HC2 lite.

aizerin, 2021-11-11:
I need to reopen this issue because the problem is still there. I switched to a different board and everything seemed to be ok.  After some included devices inclusion is just stopped working and I was not able to include anything more. I have only 9 thermostats and a few switches, nothing big. So, I tried to clear the razzbery board and include only thermostats from MCO Home to see if it will work without ztrm2fx. But I was able to include only first one and then again inclusion stopped to work :/ 

But when I use original software z-way then inclusion is done almost instantly and I'm able to add anything. So it cannot be related to the controller. Also, I'm able to interview devices in z-wave js and use them normally.

I attached the log from second try, with first device added and inclusion process.

[zwavejs_2021-11-11.log](https://github.com/zwave-js/node-zwave-js/files/7523264/zwavejs_2021-11-11.log)

AlCalzone, 2021-11-12:
> But when I use original software z-way then inclusion is done almost instantly and I'm able to add anything

I'm not sure what I can tell you. In the log I see the inclusion gets started and then stopped after 30 seconds without any inclusion related command being received in the meantime.

If it works in z-way it must be doing something different. To know what, I'll need to see some wireless logs from a Zniffer or serial logs with the communication between z-way and the stick.

closed 2021-11-30 by zwave-js-assistant[bot]

jkaberg, 2025-01-12:
@AlCalzone I also have these devices and after upgrading my Aeotec Gen5 (ZW090) to FW 1.2 and also migrating all my 27 devices from unencrypted to S2 authenticated these ztrm2fx devices all become dead within 0-3 days after pairing (no other device on my network does this)

You'll probably want some logs I presume? They will probably become lengthy as it takes a while...

Here's an picture of my second round where 3 out of 6 have gone dead since yesterday, and I'm fairly certain the others will follow in the coming day(s)
[image]

AlCalzone, 2025-01-12:
Yes, but please open a new issue [here](https://github.com/zwave-js/node-zwave-js/discussions/categories/request-support-investigate-issue)