zwave-js/zwave-js #3690: [bug] Fail to remove failed node

fgonza2, 2021-11-10:
### Version

**Checklist:**

- [x ] I am **not** using Home Assistant. **Or:** a developer has told me to come here.
- [ x] I have checked the troubleshooting section and my problem is **not** described there.
- [ x] I have read the changelog and my problem is **not** mentioned there.

Build/Run method

- [x ] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - yarn install - yarn run build )

Zwavejs2Mqtt version: 5.11.0
Z-Wave JS version: 8.7.5

### Describe the bug

Fail to remove failed node

### To Reproduce

Steps to reproduce the behavior:

1. Go to control panel
2. Select failed node
3. CLick advanced
4. Select remove failed node

Logs reports the following:
```
2021-11-09 21:07:10.319 INFO ZWAVE: Calling api removeFailedNode with args: [ 51, [length]: 1 ]
2021-11-09 21:07:18.537 INFO ZWAVE: Failed to send the message after 3 attempts (ZW0202) removeFailedNode undefined
```

### Expected behavior

Node to be removed

### Additional context

Here is the node.json dump

`{
  "id": 51,
  "name": "",
  "loc": "",
  "values": [],
  "groups": [],
  "neighbors": [],
  "ready": false,
  "available": true,
  "hassDevices": {},
  "failed": true,
  "inited": false,
  "hexId": "0xXXXX-0xXXXX-0xXXXX",
  "dbLink": "https://devices.zwave-js.io/?jumpTo=0xXXXX:0xXXXX:0xXXXX:0.0",
  "productLabel": "Unknown product 0xXXXX",
  "productDescription": "0xXXXX",
  "manufacturer": "Unknown manufacturer 0xXXXX",
  "protocolVersion": 3,
  "nodeType": 1,
  "endpointsCount": 0,
  "endpointIndizes": [],
  "isSecure": "unknown",
  "supportsSecurity": false,
  "supportsBeaming": true,
  "isControllerNode": false,
  "isListening": false,
  "isFrequentListening": false,
  "isRouting": true,
  "keepAwake": false,
  "maxDataRate": 100000,
  "deviceClass": {
    "basic": 4,
    "generic": 24,
    "specific": 0
  },
  "deviceId": "undefined-undefined-undefined",
  "status": "Asleep",
  "interviewStage": "ProtocolInfo",
  "statistics": {
    "commandsTX": 0,
    "commandsRX": 0,
    "commandsDroppedRX": 0,
    "commandsDroppedTX": 0,
    "timeoutResponse": 0
  },
  "_name": "NodeID_51"`

AlCalzone, 2021-11-10:
> Failed to send the message after 3 attempts

Looks like there is a problem communicating with the stick. Try re-plugging or soft-resetting it.

fgonza2, 2021-11-10:
i did the soft reset, nothing different from the logs, (attached file, here is the excerpt for the failed node removal)

```
2021-11-10 06:49:29.649 DEBUG SOCKET: Event ZWAVE_API emitted to bnkprHVZc8Y1q8VfAAAd
2021-11-10 06:49:29.650 INFO ZWAVE: Calling api removeFailedNode with args: [ 51, [length]: 1 ]
2021-11-10 06:49:37.786 INFO ZWAVE: Failed to send the message after 3 attempts (ZW0202) removeFailedNode undefined
```
It did a soft reset, i tried twice.  The stick is working fine, the other 36 devices and all other functions are working as expected.     Could it be some incompatibility with this stick in particula
[zwavejs2mqtt_2021-11-10.log](https://github.com/zwave-js/node-zwave-js/files/7513597/zwavejs2mqtt_2021-11-10.log)
r?

AlCalzone, 2021-11-10:
Sorry, wrong log. Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs) where you attempt to remove a node, loglevel `debug` or `silly` and attach it here as a file.

fgonza2, 2021-11-10:
Driver log attached

[zwavejs2mqtt_2021-11-10.log](https://github.com/zwave-js/node-zwave-js/files/7514932/zwavejs2mqtt_2021-11-10.log)

AlCalzone, 2021-11-10:
That's still not a driver log.

fgonza2, 2021-11-10:
what do you suggest

AlCalzone, 2021-11-10:
I posted a link to the documentation above.

fgonza2, 2021-11-16:
I did everything on the link to enable driver logging and restarted.    it is a 2.5M file.   At the end this is what i see, let me know if this looks like driver log now?

`
2021-11-15 18:49:04.531 INFO ZWAVE: Node 47 is alive
2021-11-15 18:49:04.555 DEBUG GATEWAY: Subscribing to updates of 47-37-0-targetValue
2021-11-15 18:49:04.540 INFO ZWAVE: Node 47: value added 47-37-0-currentValue => false
2021-11-15 18:49:04.547 INFO ZWAVE: Node 47: value added 47-37-0-targetValue => undefined
2021-11-15 18:49:04.556 INFO ZWAVE: Node 47: value added 47-112-0-0 => 2
2021-11-15 18:49:04.564 INFO ZWAVE: Node 47: value added 47-112-0-1 => 1
2021-11-15 18:49:04.566 INFO ZWAVE: Node 47: value added 47-114-0-manufacturerId => 335
2021-11-15 18:49:04.568 INFO ZWAVE: Node 47: value added 47-114-0-productType => 8201
2021-11-15 18:49:04.570 INFO ZWAVE: Node 47: value added 47-114-0-productId => 2307
2021-11-15 18:49:04.571 INFO ZWAVE: Node 47: value added 47-134-0-libraryType => 6
2021-11-15 18:49:04.573 INFO ZWAVE: Node 47: value added 47-134-0-protocolVersion => 3.42
2021-11-15 18:49:04.574 INFO ZWAVE: Node 47: value added 47-134-0-firmwareVersions => 14.3
2021-11-15 18:49:04.597 DEBUG ZWAVE: Node 47: mapping absence of CC 128 (Battery) to node property 'isBatteryPowered
2021-11-15 18:49:04.598 INFO ZWAVE: Node 47 ready: Nortek Security & Control LLC - WA105DBZ (Siren and Strobe)
2021-11-15 18:49:04.562 DEBUG GATEWAY: Subscribing to updates of 47-112-0-0
2021-11-15 18:49:04.565 DEBUG GATEWAY: Subscribing to updates of 47-112-0-1
2021-11-15 18:49:04.595 DEBUG GATEWAY: Publishing discovery: {
  type: 'switch',
  object_id: 'switch',
  discovery_payload: {
    payload_off: false,
    payload_on: true,
    value_template: '{{ value_json.value }}',
    command_topic: 'zwave/nodeID_47/37/0/targetValue/set',
    state_topic: 'zwave/nodeID_47/37/0/currentValue',
    device: {
      identifiers: [ 'zwavejs2mqtt_0xd0eea65e_node47', [length]: 1 ],
      manufacturer: 'Nortek Security & Control LLC',
      model: 'Siren and Strobe (WA105DBZ)',
      name: 'nodeID_47',
      sw_version: '14.3'
    },
    name: 'nodeID_47_switch',
    unique_id: 'zwavejs2mqtt_0xd0eea65e_47-37-0-currentValue'
  },
  discoveryTopic: 'switch/nodeID_47/switch/config',
  values: [ '37-0-currentValue', '37-0-targetValue', [length]: 2 ],
  persistent: false,
  ignoreDiscovery: false
}
2021-11-15 18:49:17.780 DEBUG SOCKET: Event ZWAVE_API emitted to LQWFIqoBXJJ7nw9-AAAD
2021-11-15 18:49:17.787 INFO ZWAVE: Calling api removeFailedNode with args: [ 51, [length]: 1 ]
2021-11-15 18:49:27.931 INFO ZWAVE: Node 4: value updated: 50-0-value-66049 0 => 0
2021-11-15 18:49:28.032 INFO ZWAVE: Node 30: value updated: 37-0-currentValue true => false
2021-11-15 18:49:32.872 INFO ZWAVE: Timeout while waiting for an ACK from the controller (ZW0200) removeFailedNode undefined
2021-11-15 18:49:36.231 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:36.265 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:36.856 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:36.923 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:36.975 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:37.007 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:37.048 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:37.125 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:37.359 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:37.715 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:37.755 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:37.849 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:37.933 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:37.969 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:38.272 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:38.376 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:38.466 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:38.533 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:38.662 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:38.680 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:38.733 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:38.804 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:39.596 INFO ZWAVE: Node 30: value updated: 37-0-currentValue false => false
2021-11-15 18:49:57.692 INFO ZWAVE: Node 4: value updated: 50-0-value-66049 0 => 0
2021-11-15 18:50:16.248 DEBUG SOCKET: Event ZWAVE_API emitted to LQWFIqoBXJJ7nw9-AAAD
2021-11-15 18:50:16.250 INFO ZWAVE: Calling api getProvisioningEntries with args: [ [length]: 0 ]
2021-11-15 18:50:16.253 INFO ZWAVE: Success zwave api call getProvisioningEntries [ [length]: 0 ]
2021-11-15 18:50:21.158 INFO APP: GET /api/settings 200 162.827 ms - 771613
2021-11-15 18:50:21.206 INFO APP: GET /api/auth-enabled 304 1.740 ms - -
2021-11-15 18:50:27.682 INFO ZWAVE: Node 4: value updated: 50-0-value-66049 0 => 0
2021-11-15 18:50:33.684 INFO GATEWAY: Closing Gateway...
2021-11-15 18:50:33.685 INFO GATEWAY: Driver is CLOSED
2021-11-15 18:50:33.728 INFO ZWAVE: Client closed
`

AlCalzone, 2021-11-16:
Still the wrong file. I can tell from the filename:
[image]

fgonza2, 2021-11-16:
Hopefully is this ?
[zwavejs_2021-11-15.log](https://github.com/zwave-js/node-zwave-js/files/7548379/zwavejs_2021-11-15.log)

AlCalzone, 2021-11-16:
Thats better. In that log I see a single attempt at removing a failed node which fails because the stick is unresponsive at that moment:
```
2021-11-16T02:49:29.653Z CNTRLR   No response from controller after 1/3 attempts. Scheduling next try in 100 ms.
```
(this goes on for a while)

I can't say for sure why that is happening, but I think the stick is getting overloaded by Node 4 which is spamming like crazy:
```
2021-11-16T02:48:59.924Z DRIVER « [Node 004] [REQ] [ApplicationCommand]
                                  └─[MeterCCReport]
                                      type:       Electric
                                      scale:      W
                                      rate type:  Consumed
                                      value:      0
                                      time delta: 0 seconds
2021-11-16T02:48:59.952Z SERIAL « 0x0112000400040a32022174000000000000cf004d                          (20 bytes)
2021-11-16T02:48:59.953Z CNTRLR   [Node 004] [~] [Meter] value[66049]: 0 => 0                       [Endpoint 0]
2021-11-16T02:48:59.954Z SERIAL » [ACK]                                                                   (0x06)
2021-11-16T02:48:59.955Z DRIVER « [Node 004] [REQ] [ApplicationCommand]
                                  └─[MeterCCReport]
                                      type:       Electric
                                      scale:      W
                                      rate type:  Consumed
                                      value:      0
                                      time delta: 0 seconds
2021-11-16T02:48:59.989Z SERIAL « 0x0112000400040a32022174000000000000cf004d                          (20 bytes)
2021-11-16T02:48:59.990Z CNTRLR   [Node 004] [~] [Meter] value[66049]: 0 => 0                       [Endpoint 0]
2021-11-16T02:48:59.991Z SERIAL » [ACK]                                                                   (0x06)
2021-11-16T02:48:59.992Z DRIVER « [Node 004] [REQ] [ApplicationCommand]
                                  └─[MeterCCReport]
                                      type:       Electric
                                      scale:      W
                                      rate type:  Consumed
                                      value:      0
                                      time delta: 0 seconds
...
2021-11-16T02:49:00.980Z DRIVER « [Node 004] [REQ] [ApplicationCommand]
                                  └─[MeterCCReport]
                                      type:       Electric
                                      scale:      W
                                      rate type:  Consumed
                                      value:      0
                                      time delta: 0 seconds
2021-11-16T02:49:01.063Z SERIAL « 0x0112000400040a32022174000000000000d00052                          (20 bytes)
2021-11-16T02:49:01.070Z CNTRLR   [Node 004] [~] [Meter] value[66049]: 0 => 0                       [Endpoint 0]
2021-11-16T02:49:01.078Z SERIAL » [ACK]                                                                   (0x06)
2021-11-16T02:49:01.082Z DRIVER « [Node 004] [REQ] [ApplicationCommand]
                                  └─[MeterCCReport]
                                      type:       Electric
                                      scale:      W
                                      rate type:  Consumed
                                      value:      0
                                      time delta: 0 seconds
2021-11-16T02:49:01.536Z SERIAL « 0x0112000410040a32022174000000000000c10053                          (20 bytes)
2021-11-16T02:49:01.537Z CNTRLR   [Node 004] [~] [Meter] value[66049]: 0 => 0                       [Endpoint 0]
2021-11-16T02:49:01.545Z SERIAL » [ACK]                                                                   (0x06)
2021-11-16T02:49:01.561Z DRIVER « [Node 004] [REQ] [ApplicationCommand]
                                  └─[MeterCCReport]
                                      type:       Electric
                                      scale:      W
                                      rate type:  Consumed
                                      value:      0
                                      time delta: 0 seconds
2021-11-16T02:49:01.573Z SERIAL « 0x0112000410040a32022174000000000000c10053                          (20 bytes)
2021-11-16T02:49:01.575Z CNTRLR   [Node 004] [~] [Meter] value[66049]: 0 => 0                       [Endpoint 0]
2021-11-16T02:49:01.586Z SERIAL » [ACK]                                                                   (0x06)
2021-11-16T02:49:01.589Z DRIVER « [Node 004] [REQ] [ApplicationCommand]
                                  └─[MeterCCReport]
                                      type:       Electric
                                      scale:      W
[... 94 lines cut ...]
                                  └─[BinarySwitchCCReport]
                                      current value: false
2021-11-16T02:49:37.932Z SERIAL « 0x010b0004001e03250300c6000d                                        (13 bytes)
2021-11-16T02:49:37.933Z CNTRLR   [Node 030] [~] [Binary Switch] currentValue: false => false       [Endpoint 0]
2021-11-16T02:49:37.934Z SERIAL » [ACK]                                                                   (0x06)
2021-11-16T02:49:37.935Z DRIVER « [Node 030] [REQ] [ApplicationCommand]
                                  └─[BinarySwitchCCReport]
                                      current value: false
... etc...
```

---

[... cut ...] problems:

Z-Wave sticks in particular are prone to interference by USB ports, especially by USB3 ports. We recommend putting the stick in a suitable location:

-   on an **USB extension cord** (this works wonders!)
-   away from other USB ports
-   away from metallic surfaces
-   and especially not in the back of a server rack

Once you've done that, perform a network heal to repair any suboptimal routes. Hopefully that stops the nodes from spamming.

Then try again with the remove failed node.

fgonza2, 2021-11-16:
yes, my zwave stick is already on a usb extension cable.    I did the heal network, The node 51 has now dissappeared from the list.

But now all my battery powered nodes are in "healing progress" and some of them say "failed" i try to awake those devices but still in failed state.      Should i just wait or should i do something different to get them out of failed state?

AlCalzone, 2021-11-17:
"failed" just means the healing failed. Can't you abort the healing process in zwavejs2mqtt?

fgonza2, 2021-11-17:
The healing finished and there was a bunch of "node failed" in some battery devices.  But they were still functional, so i restarted zwavejs, and the "node failed" messages dissappeared.  Two nodes wheres showing "unknown" status.    I trigger a re-interview of those two and are now complete/green

closed 2021-12-05 by zwave-js-assistant[bot]