zwave-js/zwave-js #3694: Nodes keep dying, unable to add any zwave light switches.

kwatkins, 2021-11-10:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

YES, BUT a ZWaveJS2MQTT developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Adding/inclusion of any new zWave light switches. 

What do you observe?
The Node is discovered, then fails after the "interview" step. 

What did you expect to happen?
The Nodes/light switches are discovered and usable. 

Steps to reproduce the behavior:
Anytime I'm adding a zWave light switch, GE and zOOZ. Note that this worked before, and have even reset the switches. 

### Device information

GE light switch 
zOOZ  light switch ZW0204

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Aeotec Z-Stick Gen5+ 
Z-wave JS 8.6.1
zwavejs2mqtt 5.10.1

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

It seemed to work before. I did remove a zWave light switch, although did do the heal network after. 

### Attach Driver Logfile

```
2021-11-10 12:42:20.556 INFO ZWAVE: Success zwave api call startInclusion true
2021-11-10T20:42:26.562Z CNTRLR handling add node request (status = NodeFound)
2021-11-10T20:42:26.569Z CNTRLR handling add node request (status = AddingSlave)
2021-11-10T20:42:27.552Z DRIVER the node is unknown or not initialized yet, discarding...
2021-11-10T20:42:28.441Z DRIVER the node is unknown or not initialized yet, discarding...
2021-11-10T20:42:28.684Z DRIVER the node is unknown or not initialized yet, discarding...
2021-11-10T20:42:36.232Z CNTRLR handling add node request (status = ProtocolDone)
2021-11-10T20:42:36.234Z CNTRLR finishing inclusion process...
2021-11-10T20:42:36.347Z CNTRLR The inclusion process was stopped
2021-11-10 12:42:36.348 INFO ZWAVE: Controller status: Inclusion stopped
2021-11-10T20:42:36.353Z CNTRLR finished adding node 10:
basic device class: Routing Slave
generic device class: Binary Switch
specific device class: Binary Power Switch
supported CCs:
· Basic (0x20)
· Binary Switch (0x25)
· All Switch (0x27)
· Z-Wave Plus Info (0x5e)
· Manufacturer Specific (0x72)
· Device Reset Locally (0x5a)
· Powerlevel (0x73)
· Scene Activation (0x2b)
· Scene Actuator Configuration (0x2c)
· Configuration (0x70)
· Version (0x86)
· Notification (0x71)
· Multi Channel (0x60)
· Multi Channel Association (0x8e)
· Association (0x85)
· Association Group Information (0x59)
· Firmware Update Meta Data (0x7a)
· CRC-16 Encapsulation (0x56)
controlled CCs:
2021-11-10T20:42:36.356Z CNTRLR » [Node 010] Assigning SUC return route...
2021-11-10T20:42:36.501Z CNTRLR [Node 010] Configuring Z-Wave+ Lifeline association...
2021-11-10T20:42:40.663Z CNTRLR [Node 010] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
2021-11-10T20:42:45.338Z CNTRLR [Node 010] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
2021-11-10 12:42:49.713 INFO APP: GET /health/zwave 200 2.130 ms - 1873
2021-11-10T20:42:50.030Z CNTRLR [Node 010] Node 10 did not respond after 3 attempts, it is presumed dead
2021-11-10T20:42:50.033Z CNTRLR [Node 010] Failed to configure Z-Wave+ Lifeline association: Failed to send th
e command after 3 attempts (Status NoAck) (ZW0204)
2021-11-10T20:42:50.036Z CNTRLR [Node 010] Beginning interview - last completed stage: None
2021-11-10T20:42:50.036Z CNTRLR [Node 010] new node, doing a full interview...
2021-11-10T20:42:50.037Z CNTRLR » [Node 010] querying protocol info...
2021-11-10T20:42:50.085Z CNTRLR « [Node 010] received response for protocol info:
basic device class: Routing Slave
generic device class: Binary Switch
specific device class: Binary Power Switch
node type: Routing End Node
is always listening: true
is frequent listening: false
can route messages: true
supports security: false
supports beaming: true
maximum data rate: 100000 kbps
protocol version: 3
2021-11-10 12:42:50.087 INFO ZWAVE: Node 10: interview stage PROTOCOLINFO completed
2021-11-10 12:42:50.118 INFO ZWAVE: Node 10: added with security HIGH SECURITY
2021-11-10T20:42:50.126Z CNTRLR [Node 010] Interview stage completed: ProtocolInfo
2021-11-10T20:42:50.127Z CNTRLR » [Node 010] pinging the node...
2021-11-10T20:42:50.364Z CNTRLR [Node 010] The node is now alive.
2021-11-10T20:42:50.365Z CNTRLR [Node 010] Beginning interview - last completed stage: ProtocolInfo
2021-11-10T20:42:50.366Z CNTRLR » [Node 010] querying node info...
2021-11-10 12:42:50.374 INFO ZWAVE: Node 10: has returned from the dead
2021-11-10T20:42:54.256Z CNTRLR [Node 010] Node 10 did not respond after 1 attempts, it is presumed dead
2021-11-10T20:42:54.261Z CNTRLR [Node 010] The node is now dead.
2021-11-10 12:42:54.263 INFO ZWAVE: Node 10 is now dead
2021-11-10T20:42:54.283Z CNTRLR [Node 010] ping failed: Failed to send the command after 1 attempts (Status No
Ack) (ZW0204)
2021-11-10T20:42:54.284Z CNTRLR » [Node 010] querying node info...
2021-11-10T20:42:54.286Z CNTRLR » [Node 010] pinging the node...
2021-11-10T20:42:54.331Z CNTRLR [Node 010] Interview attempt (2/5) failed, node is dead.
2021-11-10 12:42:54.332 ERROR ZWAVE: Interview of node 10 has failed: The node is dead
2021-11-10T20:42:54.361Z DRIVER unexpected response, discarding...
2021-11-10T20:42:55.960Z CNTRLR No response from controller after 1/3 attempts. Scheduling next try in 100 ms.
2021-11-10T20:42:58.531Z DRIVER no handlers registered!
```

AlCalzone, 2021-11-10:
That log doesn't tell much. Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` or `silly` and **attach** it here as a file. No copy&paste please.

That being said, nodes not responding is a connectivity problem. Z-Wave sticks in particular are prone to interference by USB ports, especially by USB3 ports. We recommend putting the stick in a suitable location:

-   on an **USB extension cord** (this works wonders!)
-   away from other USB ports
-   away from metallic surfaces
-   and especially not in the back of a server rack

If you've excluded all this, it's time to look at the mesh. Z-Wave JS doesn't have good tools to measure its quality yet, but often a **network heal** can fix bad routes.

kwatkins, 2021-11-11:
I added a USB extension cord and moved the Aeotec Z-Wave stick per your recommend, and yes, it was in a rack - the front of the rack - but in any case viola, things work. Thanks @AlCalzone

closed 2021-11-11 by kwatkins