zwave-js/zwave-js #3696: [bug] Some devices show "unknown" status and need to be reinterviewed after restart/save

stomko11, 2021-11-11:
### Version

**Checklist:**

- [x] I am **not** using Home Assistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog and my problem is **not** mentioned there.

Build/Run method

- [x] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - yarn install - yarn run build )

Zwavejs2Mqtt version: 5.11.0
Z-Wave JS version: 8.7.5

### Describe the bug

Whenever I restart docker, or within control panel, for instance I changed inclusion duration from 30 to 60 seconds and I hit save, few noted show up with status "unknown" and do not report capabilities. In Home assistant, they are greyed out. I have to re-interview the device to make it work again.
I have about z-wave 50 devices, it is usually the same switches (4 or 5) having this issue

### To Reproduce

Steps to reproduce the behavior:

1. Go to settings, change something (zwave inclusion time for instance) and hit save
2. Go to control panel, all devices' status will go to "unknown", shortly after, majority will report back "OK" status with "last active" date and time corresponding to when I changed value in settings, but few will stay "unknown" and "never". Those are mains powered devices. 
3. As soon as I reinterview the device, everything is fine immediately and I can see values for the device, and opreate it even from HA.

### Expected behavior

There should be no need to reinterview devices randomly

### Additional context

Add any other context about the problem here.

stomko11, 2021-11-11:
Log attached. 1 device was showing "unknown", one "dead". Re-interviewed both.

AlCalzone, 2021-11-11:
After the restart, your stick seems to get into a state where it cannot send, therefore devices are never detected as ready. I hoped that the soft reset would take care of this, but apparently not.

The Aeotec Gen5 seems to be very sensitive, maybe the new firmware update can help here: https://help.aeotec.com/support/solutions/articles/6000252294-z-stick-gen5-v1-02-firmware-update

If not, re-plug the stick next time this happens. Re-interviewing is not necessary.

stomko11, 2021-11-11:
Even if that's the case only for some devices? out of ~50 devices, seems like it is limited number of those that don't "come up".
I never did update FW on the stick, do I need to take it to the PC and update it here, and re-plug back to server running ZwaveJS? Any possibility to lose zwave data? I certainly don't feel like re-adding whole network again :)

AlCalzone, 2021-11-11:
> seems like it is limited number of those that don't "come up"

Probably the permanently listening ones that don't report on their own and cannot be pinged because of the stick's problems? Battery powered devices are assumed to be asleep and when a mains-powered device reports it's marked as alive and ready (if previously interviewed).

> I never did update FW on the stick, do I need to take it to the PC and update it here, and re-plug back to server running ZwaveJS? 

Yes

> Any possibility to lose zwave data? I certainly don't feel like re-adding whole network again :)

In the PC controller software, you can make an NVM backup before the upgrade and restore it if there was a problem

stomko11, 2021-11-12:
I performed upgrade, it does not seem like it helped. 5 wall plugs were showing status "unknown" when I started the dockers.

AlCalzone, 2021-11-12:
Ok then try to pull and re-plug the stick

stomko11, 2021-11-12:
No change either. 2 notes stayed "unknown" and one marked as "dead". As soon as I reinterview them, all is fine.

AlCalzone, 2021-11-12:
> As soon as I reinterview them, all is fine.

🤷🏻‍♂️ Like I said above, this shouldn't be necessary. Sending messages to a dead (non-responsive) device doesn't suddenly make it respond.

Can you make another driver log where you capture a startup that leads to such a situation?

stomko11, 2021-11-12:
Here it is. Just to remind - this does not happen only when I restart the docker. For instance I went to configuration, turned on log to file (level silly) and hit SAVE. I switch to control panel, I observe that all devices' status goes to "unknown" and last active goes to "never" for all devices. Shortly after, they start coming back with status "Alive" and last active of actual date/time.
Except for those that stay unavailable. I have 3 devices that are powered off so I don't count these (ID 8 , 12 and 49)
ID9 is Fibaro double switch, status stays "unknown" and last active "never". In this log, it is the only one.
Before, some that did not come up reported actual state "dead", however I could not perform "heal" action on them, I had to re-interview it to be able to control it.
[zwavejs2mqtt_2021-11-12.log](https://github.com/zwave-js/node-zwave-js/files/7526834/zwavejs2mqtt_2021-11-12.log)

AlCalzone, 2021-11-12:
That's not a driver log

stomko11, 2021-11-12:
How do I get the driver log please?

stomko11, 2021-11-12:
Sorry, I read the docs and attaching correct one. One node had the issue (showing as dead)
[zwavejs_2021-11-12.log](https://github.com/zwave-js/node-zwave-js/files/7527875/zwavejs_2021-11-12.log)

stomko11, 2021-11-12:
One thing I realized - I am on ZwaveJS2MQTT for some time, and I don't think I had that issues before maybe the latest upgrade of ZwaveJS2MQTT?

AlCalzone, 2021-11-12:
This is definitely the controller which is choking on something. After the startup it gets soft reset (restarted) and notifies us that it is back:
```
2021-11-12T13:32:02.374Z CNTRLR   Performing soft reset...
2021-11-12T13:32:02.375Z SERIAL » 0x01030008f4                                                         (5 bytes)
2021-11-12T13:32:02.375Z DRIVER » [REQ] [SoftReset]
2021-11-12T13:32:02.375Z SERIAL « [ACK]                                                                   (0x06)
2021-11-12T13:32:02.376Z CNTRLR   Waiting for the controller to reconnect...
2021-11-12T13:32:03.877Z CNTRLR   Re-opening serial port...
2021-11-12T13:32:04.879Z CNTRLR   Waiting for the Serial API to start...
2021-11-12T13:32:05.180Z SERIAL « 0x010a000a03000102010121de                                          (12 bytes)
2021-11-12T13:32:05.180Z SERIAL » [ACK]                                                                   (0x06)
2021-11-12T13:32:05.180Z DRIVER « [REQ] [SerialAPIStarted]
                                    payload: 0x03000102010121
2021-11-12T13:32:05.181Z CNTRLR   Serial API started
```

Then it successfully pings node 7:
```
2021-11-12T13:32:05.653Z CNTRLR » [Node 007] pinging the node...
2021-11-12T13:32:05.655Z SERIAL » 0x010800130701002501c6                                              (10 bytes)
2021-11-12T13:32:05.655Z DRIVER » [Node 007] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      1
                                  └─[NoOperationCC]
[...]
2021-11-12T13:32:05.877Z SERIAL « [ACK]                                                                   (0x06)
2021-11-12T13:32:05.877Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-11-12T13:32:05.877Z SERIAL » [ACK]                                                                   (0x06)
2021-11-12T13:32:05.878Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-11-12T13:32:05.884Z SERIAL « 0x011800130100000300c97f7f7f7f00000300000000030100003e              (26 bytes)
2021-11-12T13:32:05.884Z SERIAL » [ACK]                                                                   (0x06)
2021-11-12T13:32:05.884Z DRIVER « [REQ] [SendData]
                                    callback id:     1
                                    transmit status: OK
2021-11-12T13:32:05.887Z CNTRLR   [Node 007] The node is alive.
2021-11-12T13:32:05.955Z CNTRLR   [Node 007] The node is ready to be used
2021-11-12T13:32:05.955Z CNTRLR « [Node 007] ping successful
```

And when the driver tries to ping node 8, the controller gets stuck for almost 10 seconds. It doesn't matter that the node is not connected, the communication between the driver and the stick chokes:
```
2021-11-12T13:32:05.886Z DRIVER » [Node 008] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      2
                                  └─[NoOperationCC]
2021-11-12T13:32:05.964Z SERIAL « [ACK]                                                                   (0x06)
2021-11-12T13:32:07.570Z CNTRLR   No response from controller after 1/3 attempts. Scheduling next try in 100 ms.
2021-11-12T13:32:07.670Z SERIAL » 0x010800130801002502ca                                              (10 bytes)
2021-11-12T13:32:07.670Z DRIVER » [Node 008] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      2
                                  └─[NoOperationCC]
2021-11-12T13:32:08.671Z CNTRLR   Failed to execute controller command after 2/3 attempts. Scheduling next try i
                                  n 1100 ms.
2021-11-12T13:32:09.771Z SERIAL » 0x010800130801002502ca                                              (10 bytes)
2021-11-12T13:32:09.771Z DRIVER » [Node 008] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      2
                                  └─[NoOperationCC]
2021-11-12T13:32:10.774Z SERIAL » 0x010800130901002503ca                                              (10 bytes)
2021-11-12T13:32:10.774Z DRIVER » [Node 009] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      3
                                  └─[NoOperationCC]
2021-11-12T13:32:10.774Z CNTRLR   [Node 008] ping failed: Timeout while waiting for an ACK from the controller (
                                  ZW0200)
2021-11-12T13:32:11.776Z CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                                  n 100 ms.
2021-11-12T13:32:11.876Z SERIAL » 0x010800130901002503ca                                              (10 bytes)
[... 19 lines cut ...]
It fails to get a response (NoAck) from Node 10 when it tries to ping that one:
```
2021-11-12T13:32:14.180Z DRIVER » [Node 010] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      5
                                  └─[NoOperationCC]
2021-11-12T13:32:14.181Z CNTRLR   [Node 060] The node is alive.
2021-11-12T13:32:14.252Z CNTRLR   [Node 060] The node is ready to be used
2021-11-12T13:32:14.265Z SERIAL « [ACK]                                                                   (0x06)
[... cut ...]
2021-11-12T13:32:14.525Z SERIAL » [ACK]                                                                   (0x06)
2021-11-12T13:32:14.525Z DRIVER « [REQ] [SendData]
                                    callback id:     5
                                    transmit status: NoAck
```
but I'm not sure if that's an actual connection issue or the controller has still not recovered completely from being stuck.

---

Try disabling soft-reset in the options. That feature was added recently and it seems not well supported by older sticks.

stomko11, 2021-11-12:
I figured that after I posted - so disabled soft reset, yet the results are the same. Each change in config, different handful of devices either showing "dead" or "unknown". Right now it is 2 plugs just sitting there in unknown state, it is about 6 minutes. I will reinterview them now.

AlCalzone, 2021-11-12:
Ok can you show one more log where it happens without soft reset?

stomko11, 2021-11-12:
Here it is. It is node 51 that showed as "dead". Nodes 8, 12 and 49 are off intentionally.
On the second attempt (after I disabled logging and saved, all devices that should come up came up)
[zwavejs_2021-11-12.log](https://github.com/zwave-js/node-zwave-js/files/7529449/zwavejs_2021-11-12.log)

With logging already off, I also attempted ping like you suggested - pinging them. That is almost instant, and brings them back with correct properties and allows them to be controlled from HA.

stomko11, 2021-11-13:
Since pinging the device helps, and is way faster than interview - would there be an option for workaround, for instance ping those nodes that are showing as "unknown" or "dead"? Every time after docker either restarts, or I modify something in settings in ZwaveJS2MQTT. There still would be some nodes that are off on purpose, so I don't know - ignore if ack to ping is not received within 5 seconds? Or something like that

AlCalzone, 2021-11-13:
> would there be an option for workaround, for instance ping those nodes that are showing as "unknown" or "dead"?

Like I tried to explain above, zwave-js pings all listening nodes during startup, but your stick chokes on something and prevents zwave-js from fully doing that. It doesn't look like the soft reset is the problem here. We should try to find out why that happens.

I see in your logs that you're accessing the stick inside docker via `/dev/ttyACM0`. What's the external path you've mapped in from the host? Also `/dev/ttyACM0` or are you using the `serial/by-id` paths?

---

Back to the symptoms you're seeing:

A node that shows as `dead` after a startup was pinged, but the ping did not result in a response from the node.
A node that's `unknown` was likely not pinged because of said problems in the serial communication between zwave-js and the stick.

Node 51 does actually not respond:
```
2021-11-12T17:42:41.750Z DRIVER » [Node 051] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      34
                                  └─[NoOperationCC]
2021-11-12T17:42:41.782Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-11-12T17:42:41.821Z DRIVER « [REQ] [SendData]
                                    callback id:     34
                                    transmit status: NoAck
2021-11-12T17:42:41.822Z CNTRLR   [Node 051] Node 51 did not respond after 1 attempts, it is presumed dead
2021-11-12T17:42:41.824Z CNTRLR   [Node 051] ping failed: Failed to send the command after 1 attempts (Status No
                                  Ack) (ZW0204)
```
but the timeout after which the stick calls back with NoAck seems awfully short (40 ms). Almost seems like the controller has no alternatives to reach that node and gives up after a single attempt. If it comes back after another ping, you can try healing just that one node.

Here's an example from my stick with a node that's dead on purpose:
```
2021-11-13T13:20:39.294Z DRIVER » [Node 002] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ route:            0, 0, 0, 0
                                  │ callback id:      1
                                  └─[NoOperationCC]
2021-11-13T13:20:39.302Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2021-11-13T13:20:44.010Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:     1
                                    transmit status: NoAck
2021-11-13T13:20:44.011Z CNTRLR   [Node 002] Node 2 did not respond after 1 attempts, it is presumed dead
2021-11-13T13:20:44.012Z CNTRLR   [Node 002] The node is dead.
```
As you can see, it takes over 4 seconds until the stick gives up.

> There still would be some nodes that are off on purpose

FYI that's not a good idea. They were likely part of a working route at some point and turning them off can introduce major delays in a network.

stomko11, 2021-11-13:
I am using /dev/ttyACM0 both in docker and on host.
Nodes that are turned off are not hops for any other nodes. Those are switches in well that is turned off for winter.

AlCalzone, 2021-11-13:
Ok then they shouldn't be a problem. Can you try using a `serial/by-id` on the host?

stomko11, 2021-11-13:
It is still the same, random nodes not coming back up. I passed /dev/serial/by-id/usb-xxxxxx and configured the same thing in the settings.

AlCalzone, 2021-11-13:
Logs still full of this shortly after startup?
> No response from controller after 1/3 attempts. Scheduling next try in 100 ms

stomko11, 2021-11-14:
Yes. I also tried to downgrade to 5.9.0, and the behavior is the same there.
I tried twice, once it was node 042 that showed as dead. On next attempt, it was node 009 showing unknown.

> 2021-11-14T12:29:22.009Z CNTRLR » [Node 042] pinging the node...
> 2021-11-14T12:29:44.121Z DRIVER » [Node 042] [REQ] [SendData]
>                                   │ transmit options: 0x25
>                                   │ callback id:      30
>                                   └─[NoOperationCC]
> 2021-11-14T12:29:44.220Z CNTRLR   [Node 042] The node is alive.
> 2021-11-14T12:29:44.254Z CNTRLR   [Node 042] The node is ready to be used
> 2021-11-14T12:29:44.254Z CNTRLR « [Node 042] ping successful
> 2021-11-14T12:30:18.525Z CNTRLR   [Node 042] Embedded device config loaded
> 2021-11-14T12:30:18.680Z CNTRLR » [Node 042] pinging the node...
> 2021-11-14T12:30:39.983Z DRIVER » [Node 042] [REQ] [SendData]
>                                   │ transmit options: 0x25
>                                   │ callback id:      30
>                                   └─[NoOperationCC]
> 2021-11-14T12:30:40.069Z CNTRLR   [Node 042] Node 42 did not respond after 1 attempts, it is presumed dead
> 2021-11-14T12:30:40.069Z CNTRLR   [Node 042] The node is dead.
> 2021-11-14T12:30:40.071Z CNTRLR   [Node 042] ping failed: Failed to send the command after 1 attempts (Status No
>                                   Ack) (ZW0204)
> 								  
> 								  
> 								  
> 2021-11-14T12:36:05.608Z CNTRLR   [Node 009] Embedded device config loaded
> 2021-11-14T12:36:05.845Z CNTRLR » [Node 009] pinging the node...
> 2021-11-14T12:36:10.820Z DRIVER » [Node 009] [REQ] [SendData]
>                                   │ transmit options: 0x25
>                                   │ callback id:      3
>                                   └─[NoOperationCC]
> 2021-11-14T12:36:11.923Z DRIVER » [Node 009] [REQ] [SendData]
>                                   │ transmit options: 0x25
>                                   │ callback id:      3
>                                   └─[NoOperationCC]
> 2021-11-14T12:36:14.024Z DRIVER » [Node 009] [REQ] [SendData]
>                                   │ transmit options: 0x25
>                                   │ callback id:      3
>                                   └─[NoOperationCC]
> 2021-11-14T12:36:14.028Z CNTRLR   [Node 009] ping failed: Failed to send the message after 3 attempts (ZW0202)

AlCalzone, 2021-11-14:
I don't see the quoted part in your log snippet.
Please post the entire logfile

stomko11, 2021-11-14:
It is here, capturing those last 2 starts
[zwavejs_2021-11-14.log](https://github.com/zwave-js/node-zwave-js/files/7533921/zwavejs_2021-11-14.log)

AlCalzone, 2021-11-14:
Looks like the stick reliably chokes when trying to ping node 8. That's the disconnected one, right?
Even if the stick can't reach it, it should be sending a response back to the driver after a couple of milliseconds to indicate that it has started sending. This is very delayed (8 seconds instead of the 130ms it takes for Node 7), which I think is a firmware bug.

Not sure what Aeotec can or will do about it at this point, but I think it's worth contacting support. Feel free to include me in the ticket.

stomko11, 2021-11-15:
OK, at least some progress. I suspect it might be something related to node 12 (this is Heltun HE-RS01), or node 8 (Aeotec ZW078). These 2 were off on purpose.
Node 8 has only node 12 as neighbor and is 3 hops away from controller, and was connected via node 12. Node 12 has roughly about 20 neighbors and is 2 hops away. Most of the devices that were randomly unavailable are neighboring node 12.
Anyhow, I turned those back on. It took significant time to get Heltun up and running, reinterviewing it took ages, literally. During that time while I tried that several times, my other scene contorller was unresponsive. Maybe that's what choked controller somehow?
I tried 3 or 4 times so far, did not have any issues while saving config changes, all nodes came back up as they were supposed to.
At this point, I am not sure if this has something to do with Heltun itself, or Aeotec, or combination of those two.

AlCalzone, 2021-11-15:
> At this point, I am not sure if this has something to do with Heltun itself, or Aeotec, or combination of those two.

Now you've turned the nodes back on, the original problem of the stick is simply not happening I guess. I still think it's worth to contact support about it. If you don't want to, let me know so I can ask my contact at Aeotec.

stomko11, 2021-11-15:
I have opened a case #1465827
I might be posting here, or like you said - if I shall add you to ticket, would that be with email address posted here in your profile?

AlCalzone, 2021-11-15:
You can add the email address in my profile to the ticket.

closed 2021-11-23 by zwave-js-assistant[bot]