zwave-js/zwave-js #3701: Incorrect parameter range for HomeSeer Floodlight Sensor HS-FLS100+

COCottonwood, 2021-11-12:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Unable to input parameter value of zero due to incorrect range in homeseer_template.json file.  Current parameter min to max is 10 to 900, but actual parameter min-max is 0 to 255.  

For HomeSeer HS-FLS100+, Lux_Trigger_Threshold, allowed values are 0, 30-200, 255. Value 0 will NOT activate floodlight. Value 30 represents 10 lux, Value 200 represents 900 lux. Value 255 will activate floodlight regardless of ambient light level

What do you observe?
[image]

What did you expect to happen?

Per the manual (located at - https://products.z-wavealliance.org/ProductManual/File?folder=&filename=MarketCertificationFiles/2947/HS-FLS100-user-manual2b-A501112779R.pdf )

[image]

Parameter file needs to be updated as follows:

	"lux_trigger_threshold": {
		"label": "Light Level Threshold",
		"description": "Allowed values are 0, 30-200, 255. Value 0 will NOT activate floodlight. Value 30 represents 10 lux, Value 200 represents 900 lux. Value 255 will activate floodlight regardless of light level",
		"valueSize": 2,
		"minValue": 0,
		"maxValue": 255,
		"defaultValue": 50,
		"unsigned": true
	},

Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer: HomeSeer
Model name: HS-FLS100+
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:  
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

No Log file needed

AlCalzone, 2021-11-14:
The same template is shared by the [HS-FLS100-G2](https://devices.zwave-js.io/?jumpTo=0x000c:0x0201:0x000c:0.0) which actually uses the range stated there. We'll rather need a second template with the 0..255 range.

COCottonwood, 2021-11-14:
Thank you for looking at this... I figured that the template was mis-applied.  Is there any more information that you need from me?

AlCalzone, 2021-11-14:
No but if you like to help, providing a PR with the changes will speed things up

COCottonwood, 2021-11-14:
So... I'm not a programmer, I'm a Systems engineer (EE degree).  I thought this was a "PR".  I listed the code changes above... (but not sure if that is the correct format to allow the non-contiguous range of 0, 30-200, 255).

AlCalzone, 2021-11-15:
No, this is an issue. A PR ([Pull Request](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/about-pull-requests#about-pull-requests)) means making the changes in your copy of the repo and providing them so they can be merged.

We also have some [documentation](https://zwave-js.github.io/node-zwave-js/#/config-files/contributing-files) on contributing device file changes and predefined environments to make this easier for you.

If this is out of your league, let me know.

COCottonwood, 2021-11-18:
Please go ahead and fix it correctly… I know the technical info, but may not format it right.

Robert Apodaca

closed 2022-02-02 by AlCalzone

COCottonwood, 2022-02-03:
Just updated to 2022.2.0 and the Integrations show my device changed from HS-FLS100+ to a HS-FLS100-G2 which is incorrect.  The Z-Wave JS Config DB still needs to be updated as described in the bug report above.

Issue was NOT fixed by PR #4110.

vinmar1, 2022-02-04:
FYI - Updated to 2022.2.0 mine still shows as HS-FLS100+ but still cannot set correct parameters

blhoward2, 2022-02-04:
You have to reinterview the device to pickup a new device file.

COCottonwood, 2022-02-04:
Thanks.  I did re-interview, and then went to configure but still can't put in "0" for parameter 2.  Parameter 2 also still shows up as "LUX" units, and it should be unit-less values of 0, 30-200, 255.

COCottonwood, 2022-02-04:
Currently on CORE 2022.2.1; SUPERVISOR 2022.01.1; HOME ASSISTANT OS 7.2

COCottonwood, 2022-02-04:
[image]

vinmar1, 2022-02-05:
I'm on Frontend version: 20220203.0 ; core-2022.2.2 ; OS 7.2 (latest updates)

Did a re-interview successfully with the same results as above...

blhoward2, 2022-02-05:
Yeah, the core versions are irrelevant. You need to wait for the addon to pickup the right zwavejs version, then update it, then reinterview.

vinmar1, 2022-02-05:
Got ya... makes perfect sense should have realized that, Thanks!!

COCottonwood, 2022-02-05:
@blhoward2 - Thanks!  Does that happen automatically or can I do re-boot or re-start of something to force it?

blhoward2, 2022-02-05:
I don’t know as none of us run Supervised. I believe that you have to manually check for updates somehow in the Supervisor. Addon releases happen apart from core releases though, so it may have been out a while ago or not out yet. It’s just when they get around to updating.

vinmar1, 2022-02-05:
If you go to the Supervisor screen, then go to the “Add-on Store” tab, on the top right there are 3 dots. Click those, and in the menu that opens, click “Check For Updates” that will trigger an update of all addon repositories but there are none yet.

COCottonwood, 2022-02-05:
Thanks... just checked the ADD-ON STORE for my version of Z-Wave JS (0.1.52) and it is pointing to 8.10.2 and this fix isn't until database 8.11.2 

Z-Wave JS
Current version: 0.1.52
Bump Z-Wave JS to 8.10.2

vinmar1, 2022-02-09:
Just an FYI and a BIG THANKS! I upgraded today and got the new database re-interviewed and re-started all is wonderful! Just want to thank you guys for a great product and the support!!

devWaves, 2023-04-13:
Is this still working for everyone? I am unable to set any parameters on my HS-FLS100+. They always revert to default . I have no issues with setting params on other devices

I am on the latest HA and latest zwave js ui. Running on an RPi 4

[image]

[HS-FLS100-zwavejs.log](https://github.com/zwave-js/node-zwave-js/files/11224732/HS-FLS100-zwavejs.log)

It was suggested on HA forums that the zwave-js device configuration would need to be updated to disable "Supervision with Configuration CC" since other Homeseer devices also require this

https://github.com/zwave-js/node-zwave-js/blob/13c509216bd653abd74d43a6bcd3498367bba20f/packages/config/config/devices/0x000c/hs-wd200.json#L130-L140

I tried to create my own device config and put in store/config with the changes, but it would never load the file. I can create a PR to get the device config updated if that is needed

Thanks

AlCalzone, 2023-04-14:
@devWaves that might be the case. I only have the FLS100-G2 variant and there Supervision seems fixed.

> I tried to create my own device config and put in store/config with the changes, but it would never load the file.

Did you re-interview?

devWaves, 2023-04-14:
@AlCalzone thanks! I was missing the templates folder and I was able to add the custom device config now. I didn't see that mentioned here https://zwave-js.github.io/zwave-js-ui/#/device-support/custom-device-files

Adding the code to disable "Supervision with Configuration CC" as mentioned above works and I am now able to set the parameters without errors

I will open a PR to change the device config. Thanks again