zwave-js/zwave-js #3702: Missing device configuration: Kwikset Home Connect 620

Botched1, 2021-11-12:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Kwikset Home Connect 620

### Manufacturer ID

 0x0090

### Product Type

0x0811

### Product ID

0x13a8

### Firmware Version

7.20

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4184

### Do you have a manual?

https://products.z-wavealliance.org/ProductManual/File?folder=&filename=product_documents/4184/5069369-003.pdf

Botched1, 2021-11-12:
Question as well - 

I get CC notifications that I do not see in Home Assistant (at least not in any way I could figure out). So is there something else that needs to be done to get those events to show up in HA as well?

Lock from keypad button event:
```
2021-11-12 15:01:23.048 INFO ZWAVE: Node 115 CC Notification {
type: 6,
event: 5,
label: 'Access Control',
eventLabel: 'Keypad lock operation',
parameters: undefined
}
```
Unlock with user code #1 from the keypad:
```

2021-11-12 15:01:27.731 INFO ZWAVE: Node 115: value updated: 98-0-currentMode 255 => 0
2021-11-12 15:01:27.732 INFO ZWAVE: Node 115: value updated: 118-0-locked true => false
2021-11-12 15:01:27.732 INFO ZWAVE: Node 115 CC Notification {
type: 6,
event: 6,
label: 'Access Control',
eventLabel: 'Keypad unlock operation',
parameters: { userId: 1 }
}
```

Botched1, 2021-11-12:
NOTE - This device is also known as the "Spectrum Brands 893".

The device is the exact same as the "Spectrum Brands 892", but is zwave 700. Ther parameters appear to be the exact same on the 892 and 893 as well, so adding this product ID to the 892 driver would work.

Botched1, 2021-11-15:
Created PR for this: #3715

AlCalzone, 2021-11-16:
fixed in #3715

closed 2021-11-16 by AlCalzone